espressif/esp-idf #8311: Error initializing partition when already initialized a partition with later address

jonasgeremias, 2022-01-28:
## Environment

- Development Kit: [none]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version: [v4.2.2-dirty]
- Build System: [idf.py]
- Compiler version: [xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0]
- Operating System: [Windows]
- Environment type: [ESP Command Prompt].
- Using an IDE?: [No]
- Power Supply: [3.3V]

## Problem Description

I'm developing an application that uses one partition for logging events and another for settings:
* partition.csv
```
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
nvs,     data, nvs,  0x9000,   16k,
otadata, data, ota,  0xd000,   8K,
factory,app,factory, 0x10000,  0x160000,
ota_0,app,ota_0, 0x200000, 0x160000,
ota_1,app,ota_1, 0x400000, 0x160000,
report, data, nvs, 0x600000,  0x160000,
config, data, nvs, 0x770000,   32k,
``` 
The status of the partitions when the problem occurs. Note that there is already data in the partitions. Without data I can successfully initialize.

* Partition:'config'
   - UsedEntries = (458)
   - FreeEntries = (550)
   - AllEntries = (1008)
   
* Partition:'report'
   - UsedEntries = (15008)
   - FreeEntries = (29344)
   - AllEntries = (44352)

I get a problem when I run:
```
err = nvs_flash_init_partition("config"); // (ESP_OK)
err = nvs_flash_init_partition("report"); // (ESP_ERR_NO_MEM)
```

But it runs fine when I reverse the sequence:
```
err = nvs_flash_init_partition("report"); // (ESP_OK)
err = nvs_flash_init_partition("config"); // (ESP_OK)
```

<details>
<summary>Log error when CONFIG_HEAP_ABORT_WHEN_ALLOCATION_FAILS=y_</summary>

```
Memory allocation failed

Backtrace:0x40087012:0x3ffbddc0 0x400874f5:0x3ffbdde0 0x400d302b:0x3ffbde00 0x400810e3:0x3ffbde20 0x400810fd:0x3ffbde40 0x4008b6fd:0x3ffbde60 0x4016f5f5:0x3ffbde80 0x400edee3:0x3ffbdea0 0x400ee142:0x3ffbdf00 0x400ef1af:0x3ffbdf60 0x400ef222:0x3ffbdf90 0x400ed43f:0x3ffbdfb0 0x400e1102:0x3ffbdfd0 0x400d37ec:0x3ffbe040
0x40087012: panic_abort at C:/esp32/esp-idf/components/esp_system/panic.c:341

0x400874f5: esp_system_abort at C:/esp32/esp-idf/components/esp_system/system_api.c:106

0x400d302b: heap_caps_alloc_failed at C:/esp32/esp-idf/components/heap/heap_caps.c:67

0x400810e3: heap_caps_malloc at C:/esp32/esp-idf/components/heap/heap_caps.c:155

0x400810fd: heap_caps_malloc_default at C:/esp32/esp-idf/components/heap/heap_caps.c:177

0x4008b6fd: malloc at C:/esp32/esp-idf/components/newlib/heap.c:32

[... 302 lines cut ...]

* If I change the function in the file `nvs_page.cpp`, in function `esp_err_t Page::findItem(uint8_t nsIndex, ItemType datatype, const char* key, size_t &itemIndex, Item& item, uint8_t chunkIdx, VerOffset chunkStart)`, in the condition below, the problem gets _solved_: 

   * from:
   ```
      if (nsIndex != NS_ANY && item.nsIndex != nsIndex) {
         continue;
      }
   ``` 
   * to:
   ```
      if (item.nsIndex != nsIndex) {
         continue;
      }
   ```

*  Apparently, if it returns an ESP_OK instead of continuing the loop, the null index causes the function `Storage::populateBlobIndices(TBlobIndexList& blobIdxList)`, in `nvs_storage.cpp`, to try to allocate the entire program heap.

## My questions are:

1 - When the partitions are empty, the partitions  initialization sequence doesn't take effect. When I have some data in the 'report' partition(around 30% used), the device stops booting and I have to change the initialization sequence. Why?

2 - What is the relationship between this issue and the "workaround" I made?

3 - Is my workaround a real solution?

negativekelvin, 2022-01-29:
Are you checking free heap?  The report partition is huge and the ram used for page metadata will be high due to nvs design limitations.  To support large partitions there should be a limited size cache for page metadata.

0xjakob, 2022-01-29:
> 1 - When the partitions are empty, the partitions initialization sequence doesn't take effect. When I have some data in the 'report' partition(around 30% used), the device stops booting and I have to change the initialization sequence. Why?

@jonasgeremias I'm afraid that what @negativekelvin said is true. NVS allocates a certain amount or memory for each partition. The most important memory consumption probably stems from the instances of NVS page. Each one needs 84 bytes alone, with the hashed data much more. A page data structure is allocated in memory for each 4K of flash in the NVS partition. In the case of the `report` partition, this is `(0x160000 / 4K) * 84` = `29568.0` bytes *without* the hashes. If there are items in a page, add another minimum of 128 bytes for that page (If I am right, cc @igrr). Hence, for large pages, the memory on an ESP32 might not be enough.

> 2 - What is the relationship between this issue and the "workaround" I made?

Well, you removed part of most probably necessary code in NVS. The [host unit tests](https://github.com/espressif/esp-idf/tree/master/components/nvs_flash/test_nvs_host) fail with your modification.

> 3 - Is my workaround a real solution?

The unit test `"storage doesn't add duplicates within one page"` fails on an assertion in the code, so I think no.

jonasgeremias, 2022-01-29:
@0xjakob and @negativekelvin, thanks for helping me.

My ateration didn't work properly, I'm very sad.

I have enough heap to initialize all partitions.

The fact is that if I boot in the correct sequence ( 1 - 'report', 2 - 'config' ), 'report' uses ~70k and 'config' uses ~1.2k, leaving a lot of heap for the application.

But the opposite ( 1 - 'config', 2 - 'report'), the 'config' partition uses ~1.2k and 'report' allocates the whole heap.

But why if I initialize the partitions in order, start correctly?

Also, it takes a long time to write data to flash memory when don't have a key already created! In addition to using heap, recording is very slow! I wonder if there isn't a better way to write data to partitions? But let's solve one problem at a time hahahah.

negativekelvin, 2022-01-29:
Heap is not contiguous so you have to check the heap_caps_get_largest_free_block.  If you allocate the config memory first the largest free block probably shrinks below the size needed for report.

0xjakob, 2022-01-30:
@jonasgeremias If you need to store just one large block of data, you can use a raw partition. However, that only works best if the device isn't experiencing sudden power losses.

jonasgeremias, 2022-02-01:
I removed my change from the issue as it had no effect.

I added more data to the 'report' partition and the boot always fails.

Print the return of the `heap_caps_get_largest_free_block` function:
 
```
W (472) NVS: MALLOC_CAP_8BIT=130188
W (472) NVS: MALLOC_CAP_INTERNAL=130188
W (474) NVS: MALLOC_CAP_IRAM_8BIT=0
W (478) NVS: MALLOC_CAP_SPIRAM=0
E (2353) NVS: report initialization: ESP_ERR_NO_MEM
W (2353) NVS: Count: err = (ESP_ERR_NVS_NOT_INITIALIZED)
```

So it might be the memory allocation for the big partition.

My hardware suffers power loss all the time, so I made safeguards to protect old data in case of read and write failure:

but it doesn't make sense for me to have 16MB of memory and not be able to use it.

@negativekelvin, @0xjakob , here is a link to the test project where the issue occurs:

<https://github.com/jonasgeremias/Material-ESP32/tree/master/NVS_MEMORY>

How do I use all the flash memory on the chip? It could be native functions. If not using Heap (SPI only) it would be even better.

Is there a way to decrease the recording time when you have more keys? For me, it should be something like 20ms max.

igrr, 2022-02-01:
@jonasgeremias Unfortunately NVS will not fit your use case if you require fast updates (20ms) and large partition size (>1MB). NVS uses between [128 and 640 bytes of RAM per 4kB page](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#:~:text=The%20minimum%20amount%20of%20extra%20RAM%20usage%20per%20page%20is%20therefore%20128%20bytes%3B%20maximum%20is%20640%20bytes.) to maintain the item cache, which is required to speed up retrieval of values. From this, you can estimate how much RAM NVS will require depending on the partition size.

> How do I use all the flash memory on the chip? It could be native functions. If not using Heap (SPI only) it would be even better.

For lower-level access to Flash, you can use [partition API](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spi_flash.html#partition-table-api). You can define a large partition in the partition table and then use esp_partition_erase and esp_partition_write functions to write data into it.

If you need to write to the same sector of the partition a large number of times, you can also choose to use the [wear levelling API](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/wear-levelling.html). The wear levelling library will take care of shuffling the sectors in Flash to prevent excessive number of erase operations for any given sector.

jonasgeremias, 2022-02-01:
@igrr thanks for your quick response!

I will review these workarounds and try to implement them.

Most applications in embedded systems do not have reliable power.

I think these topics could be more explicit in the documentation. Perhaps for the more experienced this is obvious, but my considerations are:

1 - This could be put as a warning in the IDF compilation as "partition too big, maybe heap is not enough to start NVS type partition".
2 - Consider using an external memory, if you still want to risk the internal flash:
3 - Always record 2 times in flash to ensure that one of them does not corrupt during recording and later reading;
4 - A data partition takes up a lot of HEAP with large partitions, so it might be a solution to create several smaller partitions and use one at a time.
5 - Recording time is too high when creating many keys, it can be up to 1s with the NVS library when in high positions of flash memory.

It would be very interesting to put this in the documentation so that more people can know about it!

We can close this issue for me.

Thank you friends!

igrr, 2022-02-01:
Thank you for your suggestions! We will write a guide to introduce different storage APIs and explain the cases where they are applicable. The points you made are very useful, we will try to include them.

closed 2022-02-01 by jonasgeremias