espressif/esp-idf #8313: WolfSSL esp_tls TLS1.3 configuration issue

klassen9, 2022-01-28:
This can cause problems when trying to run WolfSSL in TLS1.3 only mode. 

https://github.com/espressif/esp-idf/blob/84d11d1c3ed08d4013dd9397bbefc1d75da71215/components/esp-tls/esp_tls_wolfssl.c#L171
https://github.com/espressif/esp-idf/blob/84d11d1c3ed08d4013dd9397bbefc1d75da71215/components/esp-tls/esp_tls_wolfssl.c#L313

AdityaHPatwardhan, 2022-02-14:
Hi @klassen9 Thanks for the issue.
I think using the `wolfSSL23_client_method();` would be appropriate here, as the API is able to downgrade to appropriate version which has been selected by menuconfig. 
I will make the change in the IDF.

klassen9, 2022-02-25:
Isnt the `wolfSSL23_client_method()` funktion used for SSLv2 and SSLv3? I think that funktion is outdated because SSLv2 and SSLv3 are deprecated. 

My proposed change is to add a compiler switch, checking if `WOLFSSL_TLS13` is defined. If not, the `wolfTLSv1_2_xy_method()` is called.

closed 2022-03-08 by espressif-bot