espressif/esp-idf #8314: got an unexpected keyword argument 'required'

nopnop2002, 2022-01-29:
I got ESP-IDF v5.0-dev-1389-g5dcd630444.
But i can't install it on Ubuntu 18.04.6 LTS.

```
$ idf.py
ERROR: /home/nop/.espressif/python_env/idf5.0_py3.6_env/bin/python doesn't exist! Please run the install script or "idf_tools.py install-python-env" in order to create it

ESP-IDF v5.0-dev-1389-g5dcd630444

$ git clone --recursive https://github.com/espressif/esp-idf.git

$ sudo apt-get install git wget flex bison gperf python3 python3-pip python3-setuptools cmake ninja-build ccache libffi-dev libssl-dev dfu-util libusb-1.0-0

$ cd esp-idf

$ ./install.sh esp32
Detecting the Python interpreter
Checking "python" ...
Python 3.6.9
"python" has been detected
Traceback (most recent call last):
  File "/home/nop/esp-idf/tools/install_util.py", line 70, in <module>
    main()
  File "/home/nop/esp-idf/tools/install_util.py", line 55, in main
    subparsers = parser.add_subparsers(dest='action', required=True)
  File "/usr/lib/python3.6/argparse.py", line 1716, in add_subparsers
    action = parsers_class(option_strings=[], **kwargs)
TypeError: __init__() got an unexpected keyword argument 'required'
```

dobairoland, 2022-01-29:
Hi @nopnop2002. I'm sorry for the weird error and a better message will be printed soon. ESP-IDF v5.0 will support Python 3.7 or newer only. The reason is that Python 3.6 is no longer supported by the Python community.

Python 3.8.10 is a good choice. The reason for getting the error is that you are running `install.sh` in a terminal where you have already run `. export.sh` before. Please open a new, clean terminal and run `install.sh` there.

nopnop2002, 2022-01-29:
@dobairoland 

Thank you for your quick response.

After stopping .export.sh, the installation was successful.

```
$ idf.py --version
ESP-IDF v5.0-dev-1389-g5dcd630444
```

nopnop2002, 2022-01-29:
In python3.7, idf.py menuconfig fails.
In python3.8, idf.py menuconfig succeeds.

```
$ python --version
Python 3.7.12

$ cd esp-idf/examples/get-started/hello_world

$ idf.py menuconfig
Executing action: menuconfig
Running ninja in directory /home/nop/esp-idf/examples/get-started/hello_world/build
Executing "ninja menuconfig"...
[0/1] cd /home/nop/esp-idf/examples/get-started/hel.../esp-idf/examples/get-started/hello_world/sdkconfig
WARNING: Menuconfig may fail because of the TERM environment variable is set to "xterm". Please consult the documentation of your terminal to set it up. Some good, proved to been working setups include xterm-256color, screen-256color, rxvt-unicode-256color.
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/nop/.espressif/python_env/idf5.0_py3.7_env/lib/python3.7/site-packages/menuconfig.py", line 194, in <module>
    import curses
  File "/usr/lib/python3.7/curses/__init__.py", line 13, in <module>
    from _curses import *
ModuleNotFoundError: No module named '_curses'
FAILED: CMakeFiles/menuconfig
cd /home/nop/esp-idf/examples/get-started/hello_world/build && /home/nop/.espressif/python_env/idf5.0_py3.7_env/bin/python /home/nop/esp-idf/tools/kconfig_new/prepare_kconfig_files.py --list-separator=semicolon --env-file /home/nop/esp-idf/examples/get-started/hello_world/build/config.env && /home/nop/.espressif/python_env/idf5.0_py3.7_env/bin/python /home/nop/esp-idf/tools/kconfig_new/confgen.py --list-separator=semicolon --kconfig /home/nop/esp-idf/Kconfig --sdkconfig-rename /home/nop/esp-idf/sdkconfig.rename --config /home/nop/esp-idf/examples/get-started/hello_world/sdkconfig --env-file /home/nop/esp-idf/examples/get-started/hello_world/build/config.env --env IDF_TARGET=esp32 --env IDF_ENV_FPGA= --dont-write-deprecated --output config /home/nop/esp-idf/examples/get-started/hello_world/sdkconfig && /home/nop/.espressif/python_env/idf5.0_py3.7_env/bin/python /home/nop/esp-idf/tools/check_term.py && /usr/bin/cmake -E env COMPONENT_KCONFIGS_SOURCE_FILE=/home/nop/esp-idf/examples/get-started/hello_world/build/kconfigs.in COMPONENT_KCONFIGS_PROJBUILD_SOURCE_FILE=/home/nop/esp-idf/examples/get-started/hello_world/build/kconfigs_projbuild.in KCONFIG_CONFIG=/home/nop/esp-idf/examples/get-started/hello_world/sdkconfig IDF_TARGET=esp32 IDF_ENV_FPGA= /home/nop/.espressif/python_env/idf5.0_py3.7_env/bin/python -m menuconfig /home/nop/esp-idf/Kconfig && /home/nop/.espressif/python_env/idf5.0_py3.7_env/bin/python /home/nop/esp-idf/tools/kconfig_new/confgen.py --list-separator=semicolon --kconfig /home/nop/esp-idf/Kconfig --sdkconfig-rename /home/nop/esp-idf/sdkconfig.rename --config /home/nop/esp-idf/examples/get-started/hello_world/sdkconfig --env-file /home/nop/esp-idf/examples/get-started/hello_world/build/config.env --env IDF_TARGET=esp32 --env IDF_ENV_FPGA= --output config /home/nop/esp-idf/examples/get-started/hello_world/sdkconfig
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

dobairoland, 2022-01-29:
> ModuleNotFoundError: No module named '_curses'

The issue is coming from the core of the Python interpreter itself. I guess, you can reproduce by running `python -c "import curses"`.

Probably your `/usr/lib/python3.7` is broken. Either it doesn't support curses (all of them should) or somehow it got corrupted.

nopnop2002, 2022-01-29:
I built python 3.7 from source.

The following libraries are required at build time.

```
$ sudo apt-get install libncurses5 libncurses5-dev libncursesw5
```

idf.py menuconfig is work fine.

closed 2022-01-29 by nopnop2002

dobairoland, 2022-01-29:
BTW, have you tried https://packages.ubuntu.com/bionic-updates/python3.7 from bionic-updates?

nopnop2002, 2022-01-29:
No.

I don't want to destroy the current environment.

I will try it after installing ubuntu18 on another PC.

nopnop2002, 2022-01-30:
@dobairoland 

>have you tried https://packages.ubuntu.com/bionic-updates/python3.7 from bionic-updates?

I installed the new ubuntu18.04.

I confirmed that both installation and memuconfig work with the following procedure.

```
sudo apt install python3.7 -y

sudo update-alternatives --install /usr/bin/python3 python3 /usr/bin/python3.7 1

$ python --version
Python 3.7.5
```

dobairoland, 2022-02-14:
Just for the record and cross referencing, https://github.com/espressif/esp-idf/commit/fb12d03dc38a5f7cb299d84bd4b81b7c0ed38644 should produce a error message about incompatibility with 3.6 and earlier Python versions.

vguttmann, 2023-05-11:
I've encountered this error on windows with VS code, and after installing windows-curses from the SAME TERMINAL (not from CMD.exe as it said in the error message) it started working, for anyone that's stumbling on this during debugging