espressif/esp-idf #8315: How to post an Image to server in ESP32S2?

subhash-coder, 2022-01-29:
Hello,
        I'm using ESP32S2 Dev module for my project, ESP32S2 is connected to OV5640 camera sensor, so what I want to do is capture an Image  from camera sensor and POST it to webserver. I'm able to capture the image, but I need to know how to POST image to web server?

nopnop2002, 2022-01-30:
https://github.com/nopnop2002/esp-idf-http-camera

mahavirj, 2022-02-07:
@subhash-coder 

Please check out https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/file_serving#simple-http-file-server-example for this. This example has POST method usage for uploading files to SPIFFS partition on device.

Please post `how-to` questions on https://esp32.com forum.

closed 2022-02-07 by mahavirj