espressif/esp-idf #8316: idf.py baud is not working

chegewara, 2022-01-29:
I am trying to change monitor baudrate but it is not working either with `-b` nor with `--baud`:
- esptool.py v3.3-dev
- OS: ubuntu 20.04

```
idf.py -b 9600 -p /dev/ttyUSB0 monitor
Executing action: monitor
Running idf_monitor in directory /home/chegewara/upwork/craig_atkins/project
Executing "/home/chegewara/.espressif/python_env/idf5.0_py3.8_env/bin/python /home/chegewara/esp/master/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 /home/chegewara/upwork/craig_atkins/project/build/simple_injection.elf -m '/home/chegewara/.espressif/python_env/idf5.0_py3.8_env/bin/python' '/home/chegewara/esp/master/tools/idf.py' '-b' '9600' '-p' '/dev/ttyUSB0'"...
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```
Looks like it is overridden by monitor command.

Thanks

nopnop2002, 2022-01-30:
-b is Baud rate for flashing.

There is also a Kconfig option for UART baud rate.
see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#console-uart-baudrate
You can change it using 'make menuconfig'.

chegewara, 2022-01-30:
It is not for flashing, or it is not working, because flashing speed is exactly the same with `-b 9600` and without it.
Im pretty sure i would see the difference with flashing speed 9600.

nopnop2002, 2022-01-30:
Change here.
[image]

chegewara, 2022-01-30:
Thanks, but that does not change the fact `-b` option is not working, no matter what it suppose to do.

dobairoland, 2022-02-08:
@chegewara `idf.py monitor -B <baud>` also work. BTW `idf.py -b <baud> monitor` should as well but it is not at the moment. We will patch this. But it is a breaking change so it will be available only in v5.0 and later versions.

closed 2022-02-17 by espressif-bot