espressif/esp-idf #8317: idf.py erase-otadata broken?

chegewara, 2022-01-29:
There is no option to set port and baud, like with all other `idf.py` commands:
```
idf.py erase-otadata --help
Usage: idf.py erase-otadata [OPTIONS]

  Erase otadata partition. Deprecated alias: "erase_otadata".

Options:
  -C, --project-dir PATH         Project directory.
  -D, --define-cache-entry TEXT  Create a cmake cache entry. This option can be used at most once either globally, or for one subcommand.
  --help                         Show this message and exit.
```
Instead we have to create env params:
```
Note: /home/chegewara/.espressif/python_env/idf5.0_py3.8_env/bin/python 
otatool.py will search for a serial port. To specify a port, set the ESPPORT environment variable.
Note: /home/chegewara/.espressif/python_env/idf5.0_py3.8_env/bin/python 
otatool.py will attempt to set baud rate automatically. To specify a baud rate, set the ESPBAUD environment variable.
```

dobairoland, 2022-01-30:
@chegewara `idf.py erase-otadata --help` will show you only the `erase-otadata`'s local options. The global options from `idf.py --help` should work work.

```sh
idf.py -p /dev/ttyUSB1 flash monitor  # port is global and it applies to all subcommand
```

So please try this:
```sh
idf.py -p <port> -b <baud> erase-otadata
```

chegewara, 2022-01-30:
Hi,
if it would work i would not open issue.
Please read the next few lines, it is error printed when i am trying to use `-p PORT`:
```
idf.py -p /dev/ttyUSB0 erase-otadata 
Executing action: erase-otadata
Running ninja in directory /home/chegewara/.../.../project/build
Executing "ninja erase-otadata"...
[0/1] cd /home/chegewara/esp/master/components/app_update && /usr/bin/cmake -D IDF...t/build" -P /home/chegewara/esp/master/components/esptool_py/run_serial_tool.cmake
Note: /home/chegewara/.espressif/python_env/idf5.0_py3.8_env/bin/python /home/chegewara/esp/master/components/app_update/otatool.py will search for a serial port. To specify a port, set the ESPPORT environment variable.
Note: /home/chegewara/.espressif/python_env/idf5.0_py3.8_env/bin/python /home/chegewara/esp/master/components/app_update/otatool.py will attempt to set baud rate automatically. To specify a baud rate, set the ESPBAUD environment variable.
```
```
chegewara@ubuntu:~/.../.../project$ idf.py erase-otadata -p /dev/ttyUSB0  
Usage: idf.py erase-otadata [OPTIONS]
Try 'idf.py erase-otadata --help' for help.

Error: No such option: -p
```

chegewara, 2022-01-30:
Now funny part:
```
chegewara@ubuntu:~/esp/master/examples/get-started/hello_world$ idf.py erase-otadata 
Executing action: erase-otadata
Running ninja in directory /home/chegewara/esp/master/examples/get-started/hello_world/build
Executing "ninja erase-otadata"...
ninja: error: unknown target 'erase-otadata'
ninja failed with exit code 1
```
It is from espressif example, where `build` and `monitor` work perfectly fine.

dobairoland, 2022-01-31:
Ok, thanks @chegewara. We'll look into this why it is not working.