espressif/esp-idf #8318: V4.4 ESP_HID_HOST example not recognising bt keyboard keys being pressed

turgu1, 2022-01-29:
## Environment

- Development Kit:      DOIT ESP32 DEVKIT 
- Kit version: v1
- Module or chip used:  ESP32-WROOM-32
- IDF version: V4.4
- Build System:         idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:  USB

## Problem Description

Using the ESP_HID_HOST example of v4.3.1, I was able to get a Bluetooth keyboard working (Logitech K380). Using the same keyboard with the ESP_HID_HOST example application of v4.4 (no change to the application done), the keyboard seems to be pairing, but every time a key is pressed, the following message is sent instead of the packet content:

```
E (xxxxx) BT_HIDH: Not find report handle: 0 mode: REPORT
```

### Expected Behavior

Packet information related to the key pressed as per v4.3.1 application example output.

### Actual Behavior

```
E (40464) BT_HIDH: Not find report handle: 0 mode: REPORT
```

### Steps to reproduce

1. Compile the ESP_HID_HOST example using idf.py build
2. Upload the code to the ESP32, open monitoring tool (minicom)
3. Pair the keyboard
4. Press some keys on the keyboard

// If possible, attach a picture of your setup/wiring here.

Wiring is only a USB cable from the laptop to the dev board. Nothing else.

### Code to reproduce this issue

The ESP_HID_HOST example without any change

## Debug Logs

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6612
load:0x40078000,len:14780
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (27) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
I (27) boot: compile time 12:42:30
I (27) boot: chip revision: 0
I (30) boot.esp32: SPI Speed      : 40MHz
I (35) boot.esp32: SPI Mode       : DIO
I (39) boot.esp32: SPI Flash Size : 4MB
I (44) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=24990h (149904) map
I (149) esp_image: segment 1: paddr=000349b8 vaddr=3ffbdb60 size=04648h ( 17992) load
I (157) esp_image: segment 2: paddr=00039008 vaddr=40080000 size=07010h ( 28688) load
I (169) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=951e0h (610784) map
I (390) esp_image: segment 4: paddr=000d5208 vaddr=40087010 size=0fcb4h ( 64692) load
I (417) esp_image: segment 5: paddr=000e4ec4 vaddr=50000000 size=00010h (    16) load
I (428) boot: Loaded app from partition at offset 0x10000
I (429) boot: Disabling RNG early entropy source...
I (440) cpu_start: Pro cpu up.
I (440) cpu_start: Starting app cpu, entry point is 0x40081184
0x40081184: esp_ota_get_app_elf_sha256 at /home/turgu1/esp/esp-idf/components/app_update/esp_app_desc.c:100

I (0) cpu_start: App cpu up.
I (457) cpu_start: Pro cpu start user code
I (457) cpu_start: cpu freq: 160000000
I (457) cpu_start: Application information:
I (461) cpu_start: Project name:     esp_hid_host
I (467) cpu_start: App version:      1
I (471) cpu_start: Compile time:     Jan 29 2022 12:42:26
I (477) cpu_start: ELF file SHA256:  dc985cfd1612637f...
I (483) cpu_start: ESP-IDF:          v4.4-dirty
I (489) heap_init: Initializing. RAM available for dynamic allocation:
I (495) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (502) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (508) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (514) heap_init: At 3FFC7F60 len 000180A0 (96 KiB): DRAM
I (520) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (526) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (533) heap_init: At 40096CC4 len 0000933C (36 KiB): IRAM
I (540) spi_flash: detected chip: gd
I (543) spi_flash: flash io: dio
I (549) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (634) ESP_HIDH_DEMO: setting hid gap, mode:3
I (634) BTDM_INIT: BT controller compile version [6a07b06]
I (634) system_api: Base MAC address is not set
I (634) system_api: read default base MAC address from EFUSE
I (644) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1374) ESP_HIDH_DEMO: EVENT: 5
I (1384) ESP_HIDH_DEMO: SCAN...
BLE: 6e:fd:88:aa:de:bb, RSSI: -60, UUID: 0x0000, APPEARANCE: 0x0000, ADDR_TYPE: 'RANDOM'
BLE: 52:73:9a:9b:dc:7b, RSSI: -82, UUID: 0x0000, APPEARANCE: 0x0000, ADDR_TYPE: 'RANDOM'
BLE: 4f:45:4b:2c:ec:a6, RSSI: -88, UUID: 0x0000, APPEARANCE: 0x0000, ADDR_TYPE: 'RANDOM'
BLE: ff:6d:a1:73:53:e2, RSSI: -76, UUID: 0x0000, APPEARANCE: 0x0000, ADDR_TYPE: 'RANDOM'
BT : f4:73:35:5d:da:6f, COD: major: PERIPHERAL, minor: 16, service: 0x001, RSSI: -63
BT : f4:73:35:5d:da:6f, COD: major: PERIPHERAL, minor: 16, service: 0x001, RSSI: -67
BT : f4:73:35:5d:da:6f, COD: major: PERIPHERAL, minor: 16, service: 0x001, RSSI: -65
BT : f4:73:35:5d:da:6f, COD: major: PERIPHERAL, minor: 16, service: 0x001, RSSI: -64
BT : f4:73:35:5d:da:6f, COD: major: PERIPHERAL, minor: 16, service: 0x001, RSSI: -64
BT : f4:73:35:5d:da:6f, COD: major: PERIPHERAL, minor: 16, service: 0x001, RSSI: -65
BT : f4:73:35:5d:da:6f, COD: major: PERIPHERAL, minor: 16, service: 0x001, RSSI: -70
BT : f4:73:35:5d:da:6f, COD: major: PERIPHERAL, minor: 16, service: 0x001, RSSI: -62
BT : f4:73:35:5d:da:6f, BDNAME: Keyboard K380
I (11124) ESP_HIDH_DEMO: SCAN: 1 results
  BT : f4:73:35:5d:da:6f, RSSI: -62, USAGE: GENERIC, COD: PERIPHERAL[KEYBOARD] srv 0x001, UUID16: 0x0000, NAME: Keyboard K380 
W (12044) BT_SDP: process_service_attr_rsp

W (12304) BT_SDP: process_service_attr_rsp

W (12414) BT_APPL: new conn_srvc id:23, app_id:2
I (12444) ESP_HIDH_DEMO: f4:73:35:5d:da:6f OPEN: 
BDA:f4:73:35:5d:da:6f, Status: OK, Connected: YES, Handle: 0, Usage: KEYBOARD
Name: , Manufacturer: , Serial Number: 
PID: 0xb342, VID: 0x046d, VERSION: 0x4201
Report Map Length: 270
    VENDOR  OUTPUT REPORT, ID:  17, Length:  19
    VENDOR   INPUT REPORT, ID:  17, Length:  19
    VENDOR  OUTPUT REPORT, ID:  16, Length:   6
    VENDOR   INPUT REPORT, ID:  16, Length:   6
  CCONTROL   INPUT REPORT, ID: 255, Length:   1
   GENERIC   INPUT REPORT, ID:   4, Length:   1
  CCONTROL   INPUT REPORT, ID:   3, Length:   7
  CCONTROL   INPUT REPORT, ID:   2, Length:   2
  KEYBOARD  OUTPUT   BOOT, ID:   1, Length:   1
  KEYBOARD  OUTPUT REPORT, ID:   1, Length:   1
  KEYBOARD   INPUT   BOOT, ID:   1, Length:   8
  KEYBOARD   INPUT REPORT, ID:   1, Length:   8
E (12594) BT_HIDH: Not find report handle: 0 mode: REPORT
E (12604) BT_HIDH: Not find report handle: 0 mode: REPORT
E (12664) BT_HIDH: Not find report handle: 0 mode: REPORT
I (12664) ESP_HID_GAP: BT GAP MODE_CHG_EVT mode:2
I (13214) ESP_HID_GAP: BT GAP MODE_CHG_EVT mode:0
I (14244) ESP_HID_GAP: BT GAP MODE_CHG_EVT mode:2
E (39534) BT_HIDH: Not find report handle: 0 mode: REPORT
E (39684) BT_HIDH: Not find report handle: 0 mode: REPORT
E (40384) BT_HIDH: Not find report handle: 0 mode: REPORT
E (40464) BT_HIDH: Not find report handle: 0 mode: REPORT
E (40604) BT_HIDH: Not find report handle: 0 mode: REPORT
E (40664) BT_HIDH: Not find report handle: 0 mode: REPORT
E (40764) BT_HIDH: Not find report handle: 0 mode: REPORT
E (40824) BT_HIDH: Not find report handle: 0 mode: REPORT

```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder) -> renamed sdkconfig.txt
- [x] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**) -> zip file
- [ ] coredump (This provides stacks of tasks.)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/7964468/sdkconfig.txt)
[esp_hid_host.zip](https://github.com/espressif/esp-idf/files/7964473/esp_hid_host.zip)

turgu1, 2022-01-29:
Sorry.... seems to have been fixed in a subsequent update to 4.4

closed 2022-01-29 by turgu1

AptemCat, 2022-02-01:
ESP-IDF v4.4 installed just yesterday, same issue.

turgu1, 2022-02-01:
I had the issue at first because an update to v4.4 was not made properly on my computer (network crashed). After reinstalling and doing a git pull, it worked for me. 

The installation was done like this:

```
cd ~/esp
git clone -b release/v4.4 --recursive https://github.com/espressif/esp-idf.git
```

The pull was done like this:

```
cd ~/esp/esp-idf
git pull
git submodule --update --init --recursive
```

AptemCat, 2022-02-01:
I used the [Windows Get Started tools installer](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/windows-setup.html#esp-idf-tools-installer) from the docs.

AptemCat, 2022-02-01:
Will try git installing.

turgu1, 2022-02-01:
I'm on Linux, can't help on Windows. Sorry...

watershade, 2022-03-12:
I also have the same question. The different part is I try to connect with a gampad.  I try it in Debian and Win10. The same question.

baskingcat, 2022-03-15:
Solution(idf v4.4): In file `~/esp/esp-idf/components/esp_hid/src/esp_hidh.c`, line 642
> if (r->value_len == len + 1 && r->report_id == *data && (r->report_type & 1) &&

That `+` is the bug. Change it to a `-`, and problem will be solved.
For keyboard K380, `len` is 9 (including a head which is one byte) while `r->value_len` equals 8 (just data without head). So, it should be a `-` instead of `+`. The [idf master version](https://github.com/espressif/esp-idf/blob/master/components/esp_hid/src/esp_hidh.c) has already fixed this problem.

The error message `BT_HIDH: Not find report handle: 0 mode: REPORT ` comes from (idf v4.4) `~/esp/esp-idf/components/esp_hid/src/bt_hidh.c `line 697, you can check it out if interested.

Btw, `release/v4.4` and `v4.4 stable` are two different versions. This problem exists in `v4.4 stable` not in `release/v4.4`.

watershade, 2022-03-18:
> Solution(idf v4.4): In file `~/esp/esp-idf/components/esp_hid/src/esp_hidh.c`, line 642
>
>
> That `+` is the bug. Change it to a `-`, and problem will be solved. For keyboard K380, `len` is 9 [...]

Thanks, Bro. You description is very clear. I will try to test it later. And let you know.

watershade, 2022-03-19:
> Solution(idf v4.4): In file `~/esp/esp-idf/components/esp_hid/src/esp_hidh.c`, line 642
>
>
> That `+` is the bug. Change it to a `-`, and problem will be solved. For keyboard K380, `len` is 9 [...]

watershade, 2022-03-19:
> Solution(idf v4.4): In file `~/esp/esp-idf/components/esp_hid/src/esp_hidh.c`, line 642
>
>
> That `+` is the bug. Change it to a `-`, and problem will be solved. For keyboard K380, `len` is 9 [...]

Your solution is the right way. After compare v4.4 and the new code in github, this problem have been fixed. 
Thanks again!