espressif/esp-idf #8320: esp32c3 sets the local time but the data is not normal

dreamcmi, 2022-01-30:
## Environment

- Development Kit:   ESP32-C3-DevkitM-1 v1.6
- Module or chip used:  ESP32C3
- IDF version (run ``git describe --tags`` to find it): ESP-IDF v4.4-12-gfb1d00a9d1
- Build System:  idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): riscv32-esp-elf-gcc.exe (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System:  Windows
- Using an IDE?: VSCode
- Power Supply: USB

## Problem Description
I tried to set the local time using mktime and settimeofday two functions, but the converted timestamp has errors.

```c
#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include "sys/time.h"

void app_main(void)
{
    /* set time */
    struct tm t = {0}; 
    struct timeval now = {0};

    /* get time */
    time_t now1 = {0};
    char strftime_buf[64] = {0};
    struct tm timeinfo= {0};

    /* ------------------------- */
    t.tm_year = (2022 - 1900);
    t.tm_mon = 1;
    t.tm_mday = 31;
    t.tm_hour = 17;
    t.tm_min = 50;
    t.tm_sec = 45;

    time_t timeSinceEpoch = mktime(&t);
    printf("timestamp:%d\n",timeSinceEpoch);

    now.tv_sec = timeSinceEpoch;
    settimeofday(&now, NULL);

    /* -------------------------- */
    time(&now1);
    localtime_r(&now1, &timeinfo);
    strftime(strftime_buf, sizeof(strftime_buf), "%c", &timeinfo);
    printf("gettime: %d %d %d %d %d %d\n",timeinfo.tm_year + 1900,timeinfo.tm_mon,timeinfo.tm_mday,timeinfo.tm_hour,timeinfo.tm_min,timeinfo.tm_sec);
}

```

## Debug Logs

```
I (185) cpu_start: Pro cpu up.
I (195) cpu_start: Pro cpu start user code
I (195) cpu_start: cpu freq: 160000000
I (195) cpu_start: Application information:
I (198) cpu_start: Project name:     gpio-intr
I (203) cpu_start: App version:      1
I (207) cpu_start: Compile time:     Jan 29 2022 18:33:56
I (213) cpu_start: ELF file SHA256:  afab2a53f4f5c817...
I (219) cpu_start: ESP-IDF:          v4.4-12-gfb1d00a9d1
I (225) heap_init: Initializing. RAM available for dynamic allocation:
I (232) heap_init: At 3FC8C2C0 len 00033D40 (207 KiB): DRAM
I (239) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (245) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (252) spi_flash: detected chip: winbond
I (257) spi_flash: flash io: dio
I (261) sleep: Configure to isolate all GPIO pins in sleep state
I (267) sleep: Enable automatic switching of GPIO sleep configuration
I (274) cpu_start: Starting scheduler.
timestamp:1646329845
gettime: 2022 2 3 17 50 45
```

chegewara, 2022-01-30:
It all looks good to me.
1. months is counted from 0, so if you set `t.tm_mon = 1;` it is february
2. since february this year has only 28 days and you set `t.tm_mday = 31;` it will be day 3rd of next month, which is march
3. from epochconverter.com your epoch is `GMT: Thursday, 3 March 2022 17:50:45`
4. gain, since month is counted from 0, you should do it like this 
`printf("gettime: %d %d %d %d %d %d\n",timeinfo.tm_year + 1900,timeinfo.tm_mon + 1,`
and everything will be like it suppose to be

dreamcmi, 2022-01-30:
> It all looks good to me.
> 
> 1. month is counted from 0, so if you set `t.tm_mon = 1;` it is february
> 2. since february this year has only 28 days and you set `t.tm_mday = 31;` it will be day 3rd of next month, which is march
> 3. from epochconverter.com your epoch is `GMT: Thursday, 3 March 2022 17:50:45`
> 4. again, since month is counted from 0, you should do it like this
>    `printf("gettime: %d %d %d %d %d %d\n",timeinfo.tm_year + 1900,timeinfo.tm_mon + 1,`
>    and everything will be like it suppose to be
> 
> PS this is C++ function, but can help you to check this out https://www.cplusplus.com/reference/ctime/strftime/

Thanks, I mistakenly thought that the month was counted from 1, which caused the output to be different, now the problem has been solved.

closed 2022-01-30 by dreamcmi

Ventus2ct, 2023-04-13:
Thank you for a useful code! :)