espressif/esp-idf #8321: Echo example doesn´t work with UART 1

Kampi, 2022-01-30:
## Environment

- Development Kit:  ESP32 Wrover Kit VE
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used:  ESP32-WROVER-E
- IDF version (run ``git describe --tags`` to find it): 4.0.1
- Build System:  PlatformIO
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 3.80200.200512 (8.2.0)
- Operating System: Windows
- Using an IDE?: Visual Studio code
- Power Supply:  External

## Problem Description

I use the esp-idf (UART echo example)[https://github.com/espressif/esp-idf/blob/master/examples/peripherals/uart/uart_echo/main/uart_echo_example_main.c] with `IO17` and `IO18` and `UART_NUM_1`.

```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/uart.h"
#include "driver/gpio.h"
#include "sdkconfig.h"
#include "esp_log.h"

static const char *TAG = "UART TEST";

#define INTERFACE UART_NUM_1
#define BUF_SIZE (1024)

static void echo_task(void *arg)
{
    uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity    = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE
    };
    int intr_alloc_flags = 0;

#if CONFIG_UART_ISR_IN_IRAM
    intr_alloc_flags = ESP_INTR_FLAG_IRAM;
#endif

    ESP_ERROR_CHECK(uart_driver_install(INTERFACE, BUF_SIZE * 2, 0, 0, NULL, intr_alloc_flags));
    ESP_ERROR_CHECK(uart_param_config(INTERFACE, &uart_config));
    ESP_ERROR_CHECK(uart_set_pin(INTERFACE, 17, 18, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE));

    char Message[] = "AA";
    uint8_t *data = (uint8_t *) malloc(BUF_SIZE);

    while(1)
    {
        uart_write_bytes(INTERFACE, (const char*)Message, sizeof(Message));
        int len = uart_read_bytes(INTERFACE, data, (BUF_SIZE - 1), 20 / portTICK_RATE_MS);
        if(len)
        {
            data[len] = '\0';
            ESP_LOGI(TAG, "Recv str: %s", (char *) data);
        }

        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

extern "C" void app_main(void)
{
    xTaskCreate(echo_task, "uart_echo_task", 2048, NULL, 10, NULL);
}
```

### Expected Behavior

I have tested the code with `UART_NUM_0` and got the following output.

```
I (70) boot: Chip Revision: 3
I (70) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot: ESP-IDF 4.0.1 2nd stage bootloader
I (40) boot: compile time 23:39:17
I (40) boot: Enabling RNG early entropy source...
I (44) boot: SPI Speed      : 40MHz
I (48) boot: SPI Mode       : DIO
I (52) boot: SPI Flash Size : 4MB
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (75) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (82) boot:  2 app0             OTA app          00 10 00010000 001f0000
I (90) boot:  3 app1             OTA app          00 11 00200000 001f0000
I (97) boot:  4 nvs_keys         NVS keys         01 04 003f0000 00001000
I (105) boot: End of partition table
I (109) boot_comm: chip revision: 3, min. application chip revision: 0
I (116) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x07a18 ( 31256) map
I (136) esp_image: segment 1: paddr=0x00017a40 vaddr=0x3ffbdb60 size=0x02258 (  8792) load
I (140) esp_image: segment 2: paddr=0x00019ca0 vaddr=0x40080000 size=0x06370 ( 25456) load
I (156) esp_image: segment 3: paddr=0x00020018 vaddr=0x400d0018 size=0x179b8 ( 96696) map
I (191) esp_image: segment 4: paddr=0x000379d8 vaddr=0x40086370 size=0x049ec ( 18924) load
I (206) boot: Loaded app from partition at offset 0x10000
I (206) boot: Disabling RNG early entropy source...
I (206) cpu_start: Pro cpu up.
I (210) cpu_start: Application information:
I (215) cpu_start: Project name:     UART
I (220) cpu_start: App version:      1
I (224) cpu_start: Compile time:     Jan 29 2022 23:36:52
I (230) cpu_start: ELF file SHA256:  5d07d2145b2dd17b...
I (236) cpu_start: ESP-IDF:          4.0.1
I (241) cpu_start: Starting app cpu, entry point is 0x40081148
I (0) cpu_start: App cpu up.
I (252) heap_init: Initializing. RAM available for dynamic allocation:
I (258) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (264) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (270) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (277) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (283) heap_init: At 3FFC0650 len 0001F9B0 (126 KiB): DRAM
I (289) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (295) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (302) heap_init: At 4008AD5C len 000152A4 (84 KiB): IRAM
I (308) cpu_start: Pro cpu start user code
I (326) spi_flash: detected chip: generic
I (327) spi_flash: flash io: dio
I (327) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA...
```` 

### Actual Behavior

I run the code with `UART_NUM_1` and got the following output:

```
I (70) boot: Chip Revision: 3
I (70) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot: ESP-IDF 4.0.1 2nd stage bootloader
I (40) boot: compile time 23:39:17
I (40) boot: Enabling RNG early entropy source...
I (44) boot: SPI Speed      : 40MHz
I (48) boot: SPI Mode       : DIO
I (52) boot: SPI Flash Size : 4MB
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (75) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (82) boot:  2 app0             OTA app          00 10 00010000 001f0000
I (90) boot:  3 app1             OTA app          00 11 00200000 001f0000
I (97) boot:  4 nvs_keys         NVS keys         01 04 003f0000 00001000
I (105) boot: End of partition table
I (109) boot_comm: chip revision: 3, min. application chip revision: 0
I (116) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x07a18 ( 31256) map
I (136) esp_image: segment 1: paddr=0x00017a40 vaddr=0x3ffbdb60 size=0x02258 (  8792) load
I (140) esp_image: segment 2: paddr=0x00019ca0 vaddr=0x40080000 size=0x06370 ( 25456) load
I (156) esp_image: segment 3: paddr=0x00020018 vaddr=0x400d0018 size=0x179b8 ( 96696) map
I (191) esp_image: segment 4: paddr=0x000379d8 vaddr=0x40086370 size=0x049ec ( 18924) load
I (206) boot: Loaded app from partition at offset 0x10000
I (206) boot: Disabling RNG early entropy source...
I (206) cpu_start: Pro cpu up.
I (210) cpu_start: Application information:
I (215) cpu_start: Project name:     UART
I (220) cpu_start: App version:      1
I (224) cpu_start: Compile time:     Jan 29 2022 23:36:52
I (230) cpu_start: ELF file SHA256:  9d2640da6db53ebb...
I (236) cpu_start: ESP-IDF:          4.0.1
I (241) cpu_start: Starting app cpu, entry point is 0x40081148
I (0) cpu_start: App cpu up.
I (252) heap_init: Initializing. RAM available for dynamic allocation:
I (258) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (264) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (270) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (277) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (283) heap_init: At 3FFC0650 len 0001F9B0 (126 KiB): DRAM
I (289) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (295) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (302) heap_init: At 4008AD5C len 000152A4 (84 KiB): IRAM
I (308) cpu_start: Pro cpu start user code
I (326) spi_flash: detected chip: generic
I (327) spi_flash: flash io: dio
I (327) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
```

The output is missing.

negativekelvin, 2022-01-30:
Why are you getting the debug output on UART_NUM_1? Did you change it in menuconfig?

Kampi, 2022-01-30:
> Why are you getting the debug output on UART_NUM_1? Did you change it in menuconfig?

The port for the debug output stays the same and I don´t change it in menuconfig. The application doesn't receive the data over `UART_NUM_1` and so I don´t get any output over the debug output.

negativekelvin, 2022-01-30:
Ok I thought AA was echo not log, didn't read.  You can't use gpio17 on wrover, it is not connected.

Kampi, 2022-01-31:
I have changed the pins to `19` and `21` and the example still does not work.

Kampi, 2022-02-02:
Any idea how to fix the problem? I have checked it with `5` and `18` on a Node MCU board too and the problem still exists.

Kampi, 2022-02-02:
I have updated the ESP-IDF to version `4.3.1` and my code is working now. But I need to get it to work with version `4.0.x` because I have to use the Arduino framework too.

closed 2022-04-06 by Kampi