espressif/esp-idf #8323: ESP_HTTP_CLIENT throws error when it successfully completed second GET request

ragavan10, 2022-01-31:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): v4.4
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     [Linux]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

I am working on a small application where the esp32 device fetches X.509 certificates from a server. The esp client may make multiple GET or POST requests ( it depends on the server's response). I am facing an issue where the client throws the following error when the second request has successfully completed. There is a 5s delay between each request.

```
[INFO] [cert-manager] [esp32_est_client.c:572] cert_mgr_init
[INFO] [cert-manager] [esp32_est_client.c:597] Make a request to get CA certs
W (5763) wifi:<ba-add>idx:1 (ifx:0, 58:ef:68:2f:5b:85), tid:0, ssn:0, winSize:64
[DEBUG] [cert-manager] [esp32_est_client.c:80] HTTP_EVENT_ON_CONNECTED
[DEBUG] [cert-manager] [esp32_est_client.c:83] HTTP_EVENT_HEADER_SENT
[DEBUG] [cert-manager] [esp32_est_client.c:86] HTTP_EVENT_ON_HEADER, key=Server, value=nginx
[DEBUG] [cert-manager] [esp32_est_client.c:86] HTTP_EVENT_ON_HEADER, key=Date, value=Mon, 31 Jan 2022 11:57:03 GMT
[DEBUG] [cert-manager] [esp32_est_client.c:86] HTTP_EVENT_ON_HEADER, key=Content-Type, value=text/html; charset=utf-8
[DEBUG] [cert-manager] [esp32_est_client.c:86] HTTP_EVENT_ON_HEADER, key=Content-Length, value=2352
[DEBUG] [cert-manager] [esp32_est_client.c:86] HTTP_EVENT_ON_HEADER, key=Connection, value=close
[DEBUG] [cert-manager] [esp32_est_client.c:86] HTTP_EVENT_ON_HEADER, key=Content-Transfer-Encoding, value=base64
[DEBUG] [cert-manager] [esp32_est_client.c:86] HTTP_EVENT_ON_HEADER, key=Access-Control-Allow-Origin, value=*
[DEBUG] [cert-manager] [esp32_est_client.c:86] HTTP_EVENT_ON_HEADER, key=Strict-Transport-Security, value=max-age=63072000; includeSubDomains; preload
[DEBUG] [cert-manager] [esp32_est_client.c:86] HTTP_EVENT_ON_HEADER, key=X-Frame-Options, value=DENY
[DEBUG] [cert-manager] [esp32_est_client.c:86] HTTP_EVENT_ON_HEADER, key=X-XSS-Protection, value=1; mode=block
[INFO] [cert-manager] [esp32_est_client.c:432] status_code: 200
[INFO] [cert-manager] [esp32_est_client.c:433] Response
-----BEGIN CERTIFICATE-----
MIIDZzCCAw2gAwIBAgIULl3erCLeb5kiEk+Y7wrzLcbOKvkwCgYIKoZIzj0EAwIw
TzELMAkGA1UEBhMCU0cxCzAJBgNVBAcMAlNHMREwDwYDVQQKDAhtaWNyb3NlYzEN
MAsGA1UECwwEdGVzdDERMA8GA1UEAwwIcm9vdC0xMjMwHhcNMjIwMTIyMDQyNDAz
WhcNMzIwMTIwMDQyNDAzWjBVMQswCQYDVQQGEwJTRzELMAkGA1UEBwwCU0cxETAP
BgNVBAoMCG1pY3Jvc2VjMQ0wCwYDVQQLDAR0ZXN0MRcwFQYDVQQDDA5yb290LTEy
MyAtIEludDBZMBMGByqGSM49AgEGCCqGSM49AwEHA0IABCL3+vXz0HFV9mLwj0X2
uLw+GG+9j3dlImTwbbv2iXWO1+JkOyT+TWoZCwQGp0c69Ghtzj7yoybCDiquCSw1
UcyjggG/MIIBuzASBgNVHRMBAf8ECDAGAQH/AgEAMA4GA1UdDwEB/wQEAwIBpjAn
BgNVHSUEIDAeBggrBgEFBQcDAQYIKwYBBQUHAwIGCCsGAQUFBwMJMB0GA1UdDgQW
BBRBgoMrm0KgHdLsD/cH5WDBjOxflzAfBgNVHSMEGDAWgBRqvZP3XssHm6iqlXU7
ewG27lNSujBPBggrBgEFBQcBAQRDMEEwPwYIKwYBBQUHMAGGM2h0dHA6Ly9taWNy
b3NlYy1sY21zOjgwL29jc3AvMS9pbnRlcm1lZGlhdGUtc2lnbmluZzCB2gYDVR0f
BIHSMIHPMIHMoFCgToZMaHR0cDovL21pY3Jvc2VjLWxjbXMvY3JsZC8xLzI2NDcw
Njk0MjI2Mzc1MjYzMjQ4MjI2MDA3NDcwMTA3MDI1MjA5MTcyMDQ3NzQzM6J4pHYw
dDELMAkGA1UEBhMCU0cxCzAJBgNVBAcMAlNHMREwDwYDVQQKDAhtaWNyb3NlYzEN
MAsGA1UECwwEdGVzdDE2MDQGA1UEAwwtcm9vdC0xMjMgLSBSb290IE9DU1AgUmVz
cG9uZGVyIGFuZCBDUkwgU2lnbmVyMAoGCCqGSM49BAMCA0gAMEUCIQDKaYpf1nLI
7Uma6yirbhTSHdb6w7UT/cdhZ/QjWy2OsQIgSKP8Ndr0SbPMawKJNY30Ch41j8of
1QM/GjmRLyd+iEg=
-----END CERTIFICATE-----
-----BEGIN CERTIFICATE-----
MIIDBzCCAq2gAwIBAgIUJFZDRPK6pAWNNvAAxiaj4EasdWYwCgYIKoZIzj0EAwIw
TzELMAkGA1UEBhMCU0cxCzAJBgNVBAcMAlNHMREwDwYDVQQKDAhtaWNyb3NlYzEN
MAsGA1UECwwEdGVzdDERMA8GA1UEAwwIcm9vdC0xMjMwHhcNMjIwMTIyMDQyNDAz
WhcNNDIwMTIyMDQyNDAzWjBPMQswCQYDVQQGEwJTRzELMAkGA1UEBwwCU0cxETAP
BgNVBAoMCG1pY3Jvc2VjMQ0wCwYDVQQLDAR0ZXN0MREwDwYDVQQDDAhyb290LTEy
MzBZMBMGByqGSM49AgEGCCqGSM49AwEHA0IABOBBpuPPNTLI+jG+PwJpX7rfGPx7
R4tE9D4iuEv1RQ2R1A7gQdiZS9vCcLJdWFYY8VlM1A7LNK0bDsCRJO1qsRWjggFl
MIIBYTASBgNVHRMBAf8ECDAGAQH/AgEBMA4GA1UdDwEB/wQEAwICpDAdBgNVHQ4E
FgQUar2T917LB5uoqpV1O3sBtu5TUrowPwYIKwYBBQUHAQEEMzAxMC8GCCsGAQUF
BzABhiNodHRwOi8vbWljcm9zZWMtbGNtczo4MC9vY3NwLzEvcm9vdDCB2gYDVR0f
BIHSMIHPMIHMoFCgToZMaHR0cDovL21pY3Jvc2VjLWxjbXMvY3JsZC8xLzIwNzQ0
NzM5MTgyMTczMjUwMzk3MzU2MjgyMzA3MTExNzI0MzIzMTk5ODA4MDM1OKJ4pHYw
dDELMAkGA1UEBhMCU0cxCzAJBgNVBAcMAlNHMREwDwYDVQQKDAhtaWNyb3NlYzEN
[... 292 lines cut ...]
                log_info(("Response\n%s", response));
            } else {
                log_error(("Failed to read response"));
                return 0;
            }
        }
    }
    esp_http_client_close(client);
    esp_http_client_cleanup(client);
    return status_code;

}
```

// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

I provided in the Problem Description section

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

allard-potma, 2022-02-02:
Try to enable heap poisoning detection to rule this out, the assertion may indicate a corrupt heap.

mahavirj, 2022-02-09:
@ragavan10 

It is possible that this issue could be related #8346.

Could you please try following patch and see if that fixes this issue?

```
diff --git a/examples/protocols/esp_http_client/main/esp_http_client_example.c b/examples/protocols/esp_http_client/main/esp_http_client_example.c
index d400544fd4b..d5dbd1cc8d3 100644
--- a/examples/protocols/esp_http_client/main/esp_http_client_example.c
+++ b/examples/protocols/esp_http_client/main/esp_http_client_example.c
@@ -100,14 +100,14 @@ esp_err_t _http_event_handler(esp_http_client_event_t *evt)
             int mbedtls_err = 0;
             esp_err_t err = esp_tls_get_and_clear_last_error(evt->data, &mbedtls_err, NULL);
             if (err != 0) {
-                if (output_buffer != NULL) {
-                    free(output_buffer);
-                    output_buffer = NULL;
-                }
-                output_len = 0;
                 ESP_LOGI(TAG, "Last esp error code: 0x%x", err);
                 ESP_LOGI(TAG, "Last mbedtls failure: 0x%x", mbedtls_err);
             }
+            if (output_buffer != NULL) {
+                free(output_buffer);
+                output_buffer = NULL;
+            }
+            output_len = 0;
             break;
         case HTTP_EVENT_REDIRECT:
             ESP_LOGD(TAG, "HTTP_EVENT_REDIRECT");
-- 
```

shubhamkulkarni97, 2022-02-10:
Hi @ragavan10,

It is recommended to use esp_http_client_perform if you want to use esp_http_client with event handler.

Please check my previous reply (https://github.com/espressif/esp-idf/issues/6065#issuecomment-723770180) on two ways of using esp_http_client.

hmalpani, 2022-02-18:
Hello @ragavan10,

I am unable to reproduce this issue locally. can you please try the above suggestions and see if they work.
If not, can you provide some more details about the code so that I can reproduce it locally?

mahavirj, 2022-02-23:
Closing, please feel free to re-open if there are any updates.

closed 2022-02-23 by mahavirj