espressif/esp-idf #8324: Excessive continuous power consumption on esp_wifi_start()

bosiden, 2022-01-31:
## Environment

- Development Kit:      [ESP32-Wrover]
- Kit version (custom board)
- Module or chip used:  [ESP32-WROVER-I]
- IDF version 4.3
- Build System:         ESP-IDF v4.3.1
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:    macOS
- Using an IDE?: vsCode
- Power Supply:        USB

## Problem Description

Hi,

I'm using ESP IDF 4, bare bone source code with only wifi initialization set as follow:
CODE: 
`
ESP_ERROR_CHECK(esp_netif_init());
ESP_ERROR_CHECK(esp_event_loop_create_default());
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
ESP_ERROR_CHECK(esp_wifi_init(&cfg));
ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));
ESP_ERROR_CHECK(esp_wifi_set_mode(ESPNOW_WIFI_MODE));
ESP_ERROR_CHECK(esp_wifi_start());
`
Before calling the above wifi initialization code, current consumption is at around ~4 mA, with a display backlight.

As soon as esp_wifi_start is called, power jumps to ~150 mA continuous!
Is this the expected behavior? It seems to be very excessive seeing that no task or anything is running that uses Wifi, its not even connected to any AP. ~150mA continuous without transmitting or receiving any data yet.

This is is even before I initialize ESP NOW. Any help is appreciated.

### Expected Behavior

A fairly more conservative power consumption?  it's understandable that power consumption should peak while transmitting and being connected to an AP for a short period of time, and it should be fairly low when idling or not in use.

### Actual Behavior

 Initialization alone causes over 140mA continuous drain, making it very hard to use on battery powered devices.

### Steps to reproduce

Run a bare bone example program with the above Wifi initialization alone and measure power consumption.

### Code to reproduce this issue

```cpp
ESP_ERROR_CHECK(esp_netif_init());
ESP_ERROR_CHECK(esp_event_loop_create_default());
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
ESP_ERROR_CHECK(esp_wifi_init(&cfg));
ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));
ESP_ERROR_CHECK(esp_wifi_set_mode(ESPNOW_WIFI_MODE));
ESP_ERROR_CHECK(esp_wifi_start());

```

## Debug Logs

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7036
ho 0 tail 12 room 4
load:0x40078000,len:16504
ho 0 tail 12 room 4
load:0x40080400,len:5312
0x40080400: _init at ??:?

entry 0x400806e8
I (31) boot: ESP-IDF v4.3.1-dirty 2nd stage bootloader
I (31) boot: compile time 06:15:44
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) qio_mode: Enabling default flash chip QIO
I (46) boot.esp32: SPI Speed      : 80MHz
I (51) boot.esp32: SPI Mode       : QIO
I (56) boot.esp32: SPI Flash Size : 16MB
I (60) boot: Enabling RNG early entropy source...
I (66) boot: Partition Table:
I (69) boot: ## Label            Usage          Type ST Offset   Length
I (77) boot:  0 nvs              WiFi data        01 02 0000a000 00004000
I (84) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (91) boot:  2 phy_init         RF data          01 01 00010000 00001000
I (99) boot:  3 factory          factory app      00 00 00020000 0050df20
I (106) boot:  4 app0             OTA app          00 10 00530000 0050df20
I (114) boot:  5 app1             OTA app          00 11 00a40000 0050df20
I (122) boot: End of partition table
I (126) boot: Defaulting to factory image
I (130) boot_comm: chip revision: 1, min. application chip revision: 0
I (138) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=4f9b0h (326064) map
I (231) esp_image: segment 1: paddr=0006f9d8 vaddr=3ffbdb60 size=00640h (  1600) load
I (232) esp_image: segment 2: paddr=00070020 vaddr=400d0020 size=f4100h (999680) map
I (498) esp_image: segment 3: paddr=00164128 vaddr=3ffbe1a0 size=048f4h ( 18676) load
I (504) esp_image: segment 4: paddr=00168a24 vaddr=40080000 size=1df84h (122756) load
I (543) esp_image: segment 5: paddr=001869b0 vaddr=50000000 size=00014h (    20) load
I (558) boot: Loaded app from partition at offset 0x20000
I (558) boot: Disabling RNG early entropy source...
I (569) psram: This chip is ESP32-D0WD
I (569) spiram: Found 64MBit SPI RAM device
I (570) spiram: SPI RAM mode: flash 80m sram 80m
I (575) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (582) cpu_start: Pro cpu up.
I (586) cpu_start: Starting app cpu, entry point is 0x400811f0
0x400811f0: call_start_cpu1 at /esp/esp-idf/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (1065) spiram: SPI SRAM memory test OK
I (1073) cpu_start: Pro cpu start user code
I (1073) cpu_start: cpu freq: 240000000
I (1073) cpu_start: Application information:
I (1076) cpu_start: Project name:    
I (1081) cpu_start: App version:      1
I (1086) cpu_start: Compile time:     Jan 31 2022 06:47:59
I (1092) cpu_start: ELF file SHA256:  7ae8ab3215754d44...
I (1098) cpu_start: ESP-IDF:          v4.3.1-dirty
I (1104) heap_init: Initializing. RAM available for dynamic allocation:
I (1111) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1117) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1123) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1129) heap_init: At 3FFCD088 len 00012F78 (75 KiB): DRAM
I (1136) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1142) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1149) heap_init: At 4009DF84 len 0000207C (8 KiB): IRAM
I (1155) spiram: Adding pool of 4079K of external SPI memory to heap allocator
I (1163) spi_flash: detected chip: gd
I (1167) spi_flash: flash io: qio
I (1172) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1182) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
Configuring power ledI (1192) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1232) wifi:wifi driver task: 3ffb6004, prio:23, stack:6656, core=0
I (1232) system_api: Base MAC address is not set
I (1232) system_api: read default base MAC address from EFUSE
I (1242) wifi:wifi firmware version: 88c8747
I (1242) wifi:wifi certification version: v7.0
I (1242) wifi:config NVS flash: enabled
I (1242) wifi:config nano formating: disabled
I (1252) wifi:Init data frame dynamic rx buffer num: 32
I (1252) wifi:Init management frame dynamic rx buffer num: 32
I (1262) wifi:Init management short buffer num: 32
I (1262) wifi:Init static tx buffer num: 16
I (1272) wifi:Init tx cache buffer num: 32
I (1272) wifi:Init static rx buffer size: 1600
I (1272) wifi:Init static rx buffer num: 10
I (1282) wifi:Init dynamic rx buffer num: 32
I (1282) wifi_init: tcpip mbox: 32
I (1292) wifi_init: udp mbox: 6
I (1292) wifi_init: tcp mbox: 6
I (1292) wifi_init: tcp tx win: 5744
I (1302) wifi_init: tcp rx win: 5744
I (1302) wifi_init: tcp mss: 1440
I (1312) wifi_init: WiFi/LWIP prefer SPIRAM
I (1312) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1402) wifi:mode : softAP (f0:08:d1:7e:10:35)
I (1402) wifi:Total power save buffer number: 8
I (1402) wifi:Init max length of beacon: 752/752
I (1402) wifi:Init max length of beacon: 752/752
I (1412) ESPNOW: esp now already deinit!

I (1872) gpio: GPIO[26]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:15991028 
E (1882) gpio: gpio_set_intr_type(159): GPIO interrupt type error
I (1892) gpio: GPIO[33]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:15991028 
E (1902) gpio: gpio_set_intr_type(159): GPIO interrupt type error
I (1912) BUTTON: Registering button input: 26
I (1912) BUTTON: Registering button input: 33
adc1_init...
center is: 1904
I (2122) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2122) StopDisplayAutoOff: .
I (2122) StopDisplayAutoOff: Suspending handle
I (2142) wifi: Wifi Initializing STA mode...
I (2142) wifi_init: tcpip mbox: 32
I (2142) wifi_init: udp mbox: 6
I (2142) wifi_init: tcp mbox: 6
I (2152) wifi_init: tcp tx win: 5744
I (2152) wifi_init: tcp rx win: 5744
I (2152) wifi_init: tcp mss: 1440
I (2162) wifi_init: WiFi/LWIP prefer SPIRAM
I (2162) wifi:mode : sta (f0:08:d1:7e:10:34) + softAP (f0:08:d1:7e:10:35)
I (2172) wifi:enable tsf
I (2172) wifi: wifi_init_sta finished.
I (2172) wifi: WIFI_EVENT_STA_START
```

0xjakob, 2022-02-07:
@bosiden 

> As soon as esp_wifi_start is called, power jumps to ~150 mA continuous!
Is this the expected behavior?

I'm afraid yes. It is not far off from the figures in [Table 15 in section **5.5 RF Power­Consumption Specifications** of the ESP32 datasheet](5.5 RF Power­Consumption Specifications). Note that they mention
> power consumption measurements are taken (...) at the RF
port.
I'm not entirely sure what the RF port is but it doesn't sound like the main power rail on the module or even development board (which may draw current through other components like a USB-bridge, etc.).

WiFi in the current version available on ESP chips is not very power-saving, but that's a design issue of WiFi itself, which is basically optimized for throughput rather than power.

bosiden, 2022-02-08:
> WiFi in the current version available on ESP chips is not very power-saving, but that's a design issue of WiFi itself, which is basically optimized for throughput rather than power.

Its understandable that Wifi would require high power when fully utilized, but for me, the following statement is a bit misleading or not entirely true:

> ESP-NOW enables a direct and low-power control of smart
lights, without the need of a router. This method is power-
efficient and convenient

ESP-NOW is far from being a low-power protocol, especially when talking about battery powered devices.

Enabling ESP-NOW alone should theoretically results in a reduced power draw since it bypasses many Wifi layers? I mean the good old nrf24 operates on the same frequency as ESP-NOW and uses 10x less power for approximately the same range and data rate.

MaxwellAlan, 2022-03-29:
hi @bosiden 
When WiFi start stage, there is a phy calibration in the chip which include tx & rx will cause high power consumption, for ESP32, there is little optimization for power consumption that we can do. But for ESP32C3, when WiFi start, 110mA now, there is a lot opmization that we can do, maybe we can get lower 50mA. Have a try with ESP32C3.

> ESP-NOW is far from being a low-power protocol, especially when talking about battery powered devices.
>
> Enabling ESP-NOW alone should theoretically results in a reduced power draw since it bypasses [...]

As for this, different protocol have much more different power consumption. We can only ensure that using ESP-NOW have less power consumption than usually full TCP/IP & WiFi stack.

What's more, can you provide more details about what scenario or what application your production work in?

Thanks.

jack0c, 2022-06-07:
@bosiden You need consider some factors:
1. DC-DC circuit, yes, current ESP32 series do not include such circuit，will added in some of subsequent espressif chips
2. Wi-Fi performance, TX power and RX sensitivity, it will impact the substrate current of the chips, ESP32 series are with very good performance
3. Soc part will also impact the substrate current，ESP32 has 2 cores work on 240MHz，while ESP32 has one 1 core works on 160MHz
4. Each time when SP32 series chips power up, RF calibration is necessary, however, we can reduce the time to do the RF calibration
5. ESP-NOW is a protocol which reduce the time to connect to AP, get IP address and setup TCP connection, it mainly reduce the time to wake up
6. The application need design how to use ESP-NOW and deepsleep, a good example is a push button which can be used to control the light

closed 2023-02-21 by Alvin1Zhang