espressif/esp-idf #8325: Support MBEDTLS_DYNAMIC_BUFFER in conjunction with DTLS support

hennejg, 2022-01-31:
**Is your feature request related to a problem? Please describe.**

Commit c91afab332a9fcf071cb478b25f8f8e03d0afcf1 disables the MBEDTLS_DYNAMIC_BUFFER  configuration option when MBEDTLS_SSL_PROTO_DTLS is y. This is unfortunate, as we would like to use this combination. 

**Describe the solution you'd like**

The default mbedTLS buffer sizes are way too large particularly for DTLS data. Being able to use that combination of options would be great.

**Describe alternatives you've considered**

We might be willing to implement the missing pieces ourselves and contribute the solution, however, a few pointers and comments on the rationale behind the change would be helpful and more than welcome.

lhespress, 2022-03-11:
@hennejg Thanks for the suggestion, we will try to discuss it and tell you when have any update.

LJYSP, 2022-05-05:
[0001-feat-mbedtls-support-dtls-with-mbedtls-dynamic-memor.txt](https://github.com/espressif/esp-idf/files/8628568/0001-feat-mbedtls-support-dtls-with-mbedtls-dynamic-memor.txt)

@hennejg please rename the file to xxx.patch, and  apply the patch to IDF master branch. Please test with this feature and welcome to feedback to us.

AxelLin, 2022-05-08:
@LJYSP 
Got below build warning with your patch: (apply patch to v5.0-dev-2586-ga82e6e63d9)
/home/axel/esp/esp-idf/components/mbedtls/port/dynamic/esp_mbedtls_dynamic_impl.c: In function 'esp_mbedtls_add_rx_buffer':
/home/axel/esp/esp-idf/components/mbedtls/port/dynamic/esp_mbedtls_dynamic_impl.c:421:39: error: 'in_left' may be used uninitialized in this function [-Werror=maybe-uninitialized]
         ssl->MBEDTLS_PRIVATE(in_left) = in_left;
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
cc1: all warnings being treated as errors
[123/1219] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedtls.dir/ssl_tls.c.obj

LJYSP, 2022-05-09:
@AxelLin  Just need to initialize the in_left to 0:
```
//Line： 348
    size_t in_msglen, in_left = 0;
```

AxelLin, 2022-06-20:
@hennejg 
Does this patch https://github.com/espressif/esp-idf/issues/8325#issuecomment-1118134508 work for your?

hennejg, 2022-06-22:
> @hennejg Does this patch [#8325 (comment)](https://github.com/espressif/esp-idf/issues/8325#issuecomment-1118134508) work for your?

@LJYSP, @AxelLin sorry for taking so song to get back to you. I back-ported the patch to "my" version of 4.4, since I am currently stuck on a slightly older version which does not yet use the `MBEDTLS_PRIVATE` macro. The patch seems to work fine for me. Thank you very much for you support!

AxelLin, 2023-12-17:
> [0001-feat-mbedtls-support-dtls-with-mbedtls-dynamic-memor.txt](https://github.com/espressif/esp-idf/files/8628568/0001-feat-mbedtls-support-dtls-with-mbedtls-dynamic-memor.txt)
>
> @hennejg please [...]

@LJYSP 
Looks like your patch is not yet merged? Why?

AxelLin, 2024-01-12:
Hi @mahavirj  @LJYSP 

MBEDTLS_DYNAMIC_BUFFER is disabled when MBEDTLS_SSL_PROTO_DTLS is set since 2b1402890de0ec26fb382ca8a19dc4a9f1ff1853 .
Just wondering if someone can help to pick up this fix?