espressif/esp-idf #8328: esp_ble_gap_set_device_name does not update BLE device name

krupis, 2022-02-01:
I am working on the BLE and used BLE_security_server example from esp-idf and modified the code to my needs. One of the requirements of the product that I am working on is to be able to dynamically create/change BLE device name. esp_ble_gap_set_device_name() function is used to update the ble device name.

```
case ESP_GATTS_REG_EVT:
        {
            char default_ble_uuid[23];
            unsigned char mac_base[6];
            esp_efuse_mac_get_default(mac_base);
            char temp_buf1[3];
            char temp_buf2[3];
            char temp_buf3[3];
            char temp_buf4[3];
            char temp_buf5[3];
            char temp_buf6[3];
            sprintf(temp_buf1, "%02x", mac_base[0]);
            sprintf(temp_buf2, "%02x", mac_base[1]);
            sprintf(temp_buf3, "%02x", mac_base[2]);
            sprintf(temp_buf4, "%02x", mac_base[3]);
            sprintf(temp_buf5, "%02x", mac_base[4]);
            sprintf(temp_buf6, "%02x", mac_base[5]);

            strcpy(default_ble_uuid,"BLE");
            strcat(default_ble_uuid,"custom_id");
            strcat(default_ble_uuid,temp_buf1);
            strcat(default_ble_uuid,temp_buf2);
            strcat(default_ble_uuid,temp_buf3);
            strcat(default_ble_uuid,temp_buf4);
            strcat(default_ble_uuid,temp_buf5);
            strcat(default_ble_uuid,temp_buf6);

            printf("BLE device name = %s \n",default_ble_uuid);    
            esp_ble_gap_set_device_name(default_ble_uuid);
            //generate a resolvable random address

            esp_ble_gap_config_local_privacy(true);
            esp_ble_gatts_create_attr_tab(heart_rate_gatt_db, gatts_if,
                                      ELSTAT_IDX_NB, ELSTAT_SVC_INST_ID);

            break;
        }
```

My BLE structures are set to following (if they are relevant):
```
static esp_ble_adv_data_t elstat_adv_config = {
    .set_scan_rsp = false,
    .include_txpower = true,
    .min_interval = 0x0006, //slave connection min interval, Time = min_interval * 1.25 msec
    .max_interval = 0x0011, //slave connection max interval, Time = max_interval * 1.25 msec
    .appearance = 0x00,
    .manufacturer_len = 0, 
    .p_manufacturer_data =  NULL, 
    .service_data_len = 0,
    .p_service_data = NULL,
    .service_uuid_len = sizeof(sec_service_uuid),
    .p_service_uuid = sec_service_uuid,
    .flag = (ESP_BLE_ADV_FLAG_GEN_DISC | ESP_BLE_ADV_FLAG_BREDR_NOT_SPT),
};

static esp_ble_adv_params_t elstat_adv_params = {
    .adv_int_min        = 0x100,
    .adv_int_max        = 0x100,
    .adv_type           = ADV_TYPE_IND,
    .own_addr_type      = BLE_ADDR_TYPE_PUBLIC,
    .channel_map        = ADV_CHNL_ALL,
    .adv_filter_policy = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY,
};

```

Everytime the program starts, during the BLE initialization the ESP_GATTS_REG_EVT will be triggered where I calculate the BLE device name. It will consist of "BLE" prefix + custom client id + mac address.

"BLE"<custom_id><mac_address>.

I have tried changing the custom ID and I can see that the BLE device name does not update after I reset the device. I also tried to set the name to something simple such as:
```
            char test_ble_name[20] = "TEST1";
            esp_ble_gap_set_device_name(test_ble_name);
```
But it still shows the old BLE device name... I have tried to restart the device multiple times, unpair it with my phone multiple times and eventually it updated. This is not good for us as we are going to have multiple of those devices deployed and being able to update the BLE device name on first reset is crucial. I hope I can get some clarification regarding this.

Weijian-Espressif, 2022-03-30:
@krupis  please set `.include_name = true,` in adv data struct, and retry.

Alvin1Zhang, 2022-04-28:
@krupis Thanks for reporting, would you please help share if any updates for the issue? Thanks.

ricmoo, 2022-05-12:
I'm having the same problem; the initial sketch I uploaded set the name and now I can't get it to change...

ricmoo, 2022-05-12:
More confusing, the write bytes look like they are going to the hci, I've added some login deeper inside the stack:

```
ble_hs_hci_cmd_send: 17 02 01 06 03 03 12 18 08 09 46 69 72 65 66 6c 79 02 0a 03 03 19 c4 03 18 d0 fc 3f b7 11 fc 3f
ble_hs_hci_cmd_send: 30 00 60 00 00 00 00 00 00 00 00 00 00 07 00
ble_hs_hci_cmd_send: 01
```

That has "Firefly" in there `46 69 72 65 66 6c 79`, but on the phone and laptop using the OS and LightBlue, shows up as "ble_kbdhid", which the initial sketch I flashed on it.

krupis, 2022-05-12:
Hey. There is a more reliable way of setting a BLE device name but it is a little bit more complex than just using esp_ble_gap_set_device_name.

Have a look at my BLE setup code. The BLE advertisment name is constructed using raw bytes advertisment. In order to achieve this, you will need to lookup how RAW advertisment BLE packet is constructed. Try to read up on:
https://docs.silabs.com/bluetooth/latest/general/adv-and-scanning/bluetooth-adv-data-basics
or simmilar material.
uint8_t raw_adv_data2[50];
`void BLE_setup()
{
    unsigned char mac_base[6];
    esp_read_mac(mac_base, ESP_MAC_BT);
    
    char temp_buf1[3];
    char temp_buf2[3];
    char temp_buf3[3];
    char temp_buf4[3];
    char temp_buf5[3];
    char temp_buf6[3];
    sprintf(temp_buf1, "%02X", mac_base[0]);
    sprintf(temp_buf2, "%02X", mac_base[1]);
    sprintf(temp_buf3, "%02X", mac_base[2]);
    sprintf(temp_buf4, "%02X", mac_base[3]);
    sprintf(temp_buf5, "%02X", mac_base[4]);
    sprintf(temp_buf6, "%02X", mac_base[5]);

    esp_err_t ret;
   
    raw_adv_data2[0] = 0x02;
    raw_adv_data2[1] = 0x01;
    raw_adv_data2[2] = 0x06;

    raw_adv_data2[3] = 0x02;
    raw_adv_data2[4] = 0x0a;
    raw_adv_data2[5] = 0xeb;

    raw_adv_data2[6] = 0x16; // length

    raw_adv_data2[7] = 0x09;

    raw_adv_data2[8] = 0x41;// asci "A"
    raw_adv_data2[9] = 0x41; // asci "A"
    raw_adv_data2[10] = 0x41; // asci "A"
    raw_adv_data2[11] = 0x41; // asci "A"
    
    for(int i = 0; i <3;i++){
        raw_adv_data2[12+i] = device_info.client_id[i];
    }
    raw_adv_data2[15] = 0x50; // 0x50
    raw_adv_data2[16] = 0x33; // 0x33

    raw_adv_data2[17] = temp_buf1[0];
    raw_adv_data2[18] = temp_buf1[1];

    raw_adv_data2[19] = temp_buf2[0];
    raw_adv_data2[20] = temp_buf2[1];

    raw_adv_data2[21] = temp_buf3[0];
    raw_adv_data2[22] = temp_buf3[1];
    
    raw_adv_data2[23] = temp_buf4[0];
    raw_adv_data2[24] = temp_buf4[1];

    raw_adv_data2[25] = temp_buf5[0];
    raw_adv_data2[26] = temp_buf5[1];

    raw_adv_data2[27] = temp_buf6[0];
    raw_adv_data2[28] = temp_buf6[1];

    ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT));
    //ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_BLE));

    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    ret = esp_bt_controller_init(&bt_cfg);
    if (ret) {
        ESP_LOGE(GATTS_TABLE_TAG, "%s init controller failed: %s", __func__, esp_err_to_name(ret));
        return;
    }
    ret = esp_bt_controller_enable(ESP_BT_MODE_BLE);
    if (ret) {
        ESP_LOGE(GATTS_TABLE_TAG, "%s enable controller failed: %s", __func__, esp_err_to_name(ret));
        return;
    }

    ESP_LOGI(GATTS_TABLE_TAG, "%s init bluetooth", __func__);
    ret = esp_bluedroid_init();
    if (ret) {
        ESP_LOGE(GATTS_TABLE_TAG, "%s init bluetooth failed: %s", __func__, esp_err_to_name(ret));
        return;
    }
    ret = esp_bluedroid_enable();
    if (ret) {
        ESP_LOGE(GATTS_TABLE_TAG, "%s enable bluetooth failed: %s", __func__, esp_err_to_name(ret));
        return;
    }
    /*
    const uint8_t* point = esp_bt_dev_get_address();
    for (int i = 0; i < 6; i++) {
        char str[3];
        sprintf(str, "%02X", (int)point[i]);
        DebugPrint(str);
        if (i < 5){
            DebugPrint(":");
        }
    }
    */

    ret = esp_ble_gatts_register_callback(gatts_event_handler);
    if (ret){
        ESP_LOGE(GATTS_TABLE_TAG, "gatts register error, error code = %x", ret);
        return;
    }
    ret = esp_ble_gap_register_callback(gap_event_handler);
    if (ret){
        ESP_LOGE(GATTS_TABLE_TAG, "gap register error, error code = %x", ret);
        return;
    }

    ret = esp_ble_gatts_app_register(ELSTAT_CUSTOM_APP_ID);
    if (ret){
        ESP_LOGE(GATTS_TABLE_TAG, "gatts app register error, error code = %x", ret);
        return;
    }

    /* set the security iocap & auth_req & key size & init key response key parameters to the stack*/
    esp_ble_auth_req_t auth_req = ESP_LE_AUTH_REQ_SC_MITM_BOND;     //bonding with peer device after authentication
    esp_ble_io_cap_t iocap = ESP_IO_CAP_OUT;           //set the IO capability to No output No input
    uint8_t key_size = 16;      //the key size should be 7~16 bytes
    uint8_t init_key = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
    uint8_t rsp_key = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
    //set static passkey
    //uint32_t passkey = 123456;
    uint8_t auth_option = ESP_BLE_ONLY_ACCEPT_SPECIFIED_AUTH_DISABLE;
    uint8_t oob_support = ESP_BLE_OOB_DISABLE;
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_STATIC_PASSKEY, &BLE_passkey, sizeof(uint32_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_AUTHEN_REQ_MODE, &auth_req, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_IOCAP_MODE, &iocap, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_MAX_KEY_SIZE, &key_size, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_ONLY_ACCEPT_SPECIFIED_SEC_AUTH, &auth_option, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_OOB_SUPPORT, &oob_support, sizeof(uint8_t));
    /* If your BLE device acts as a Slave, the init_key means you hope which types of key of the master should distribute to you,
    and the response key means which key you can distribute to the master;
    If your BLE device acts as a master, the response key means you hope which types of key of the slave should distribute to you,
    and the init key means which key you can distribute to the slave. */
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_INIT_KEY, &init_key, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_RSP_KEY, &rsp_key, sizeof(uint8_t));

    

    /* Just show how to clear all the bonded devices
     * Delay 30s, clear all the bonded devices
     *
     * vTaskDelay(30000 / portTICK_PERIOD_MS);
     * remove_all_bonded_devices();
     */

}`

ricmoo, 2022-05-12:
The actual bytes sent over the HCI are correct though (manually parsing the bytes). Is there any difference preparing the ra bytes gains you? I think it should be the same (it's that same as I'd have prepared by hand, except the last field in a different order than I'd have put it)...

krupis, 2022-05-12:
Im afraid I wont be able to help you here as this is beyond my knowledge. Hopefully someone will clarify that from the esp-idf

ricmoo, 2022-05-12:
I think the “problem” is entirely in the OS caching for security/UX purposes.

By using the `esp_base_mac_addr_set` to give it a new BLE address, name changes work.

Thanks for your time though. :)

esp-cjh, 2023-02-21:
The device name in advertising is included in advertising data or scan response data. So if you want to change the device name when advertising, you don't need to reboot or stop advertising. Just follow the steps below:
```
/* set device name */
esp_ble_gap_set_device_name("device name");
/* if adv data include name */
esp_ble_gap_config_adv_data(&adv_data);
/* or if scan rsp data include name */
esp_ble_gap_config_adv_data(&scan_rsp_data);
```

closed 2023-05-24 by Alvin1Zhang