espressif/esp-idf #8329: warning: unused variable 'guard'

ammaree, 2022-02-01:
For the last couple of months (< 12) I have had a consistent message warning of an unused variable 'guard' in the spirit_flash/flash_ops.c component.

```
/Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/spi_flash/flash_ops.c: In function 'spi_flash_erase_sector':
/Users/andremaree/Dropbox/devs/ws/z-sdk/esp-idf/components/spi_flash/flash_ops.c:284:36: warning: unused variable 'guard' [-Wunused-variable]
     const spi_flash_guard_funcs_t *guard = spi_flash_guard_get();
                                                            ^~~~~
```

I have attached the sdkconfig file used. Any suggestion to remove this warning?

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/7977263/sdkconfig.txt)

kayzen9, 2022-02-01:
Try disabling `CONFIG_SPI_FLASH_DANGEROUS_WRITE_ALLOWED=y`

ammaree, 2022-02-01:
Thanks for the advice but we need the config setting. We support dynamically changing/fixing the partition table for devices in the field, upgrading old FW without a LittleFS partition, to add the partition with new firmware.

Any other suggestions?

mahavirj, 2022-02-03:
@ammaree 

Please try following patch:

```
diff --git a/components/spi_flash/flash_ops.c b/components/spi_flash/flash_ops.c
index f40925edb27..0b6bb8e3f6d 100644
--- a/components/spi_flash/flash_ops.c
+++ b/components/spi_flash/flash_ops.c
@@ -126,7 +126,8 @@ const DRAM_ATTR spi_flash_guard_funcs_t g_flash_guard_no_os_ops = {
    bootloader, partition table, or running application region.
 */
 #if CONFIG_SPI_FLASH_DANGEROUS_WRITE_ALLOWED
-#define CHECK_WRITE_ADDRESS(ADDR, SIZE)
+// Following helps in masking "unused variable" warning
+#define CHECK_WRITE_ADDRESS(ADDR, SIZE) ({(void) guard;})
 #else /* FAILS or ABORTS */
 #define CHECK_WRITE_ADDRESS(ADDR, SIZE) do {                            \
         if (guard && guard->is_safe_write_address && !guard->is_safe_write_address(ADDR, SIZE)) {                       \
```

closed 2022-02-09 by espressif-bot