espressif/esp-idf #8330: Missing documentation about how to properly use ESP-NETIF LWIP abstraction

crisvdn, 2022-02-01:
## Environment

- Development Kit:      Custom PCB with mentioned chip
- Module or chip used:  ESP32-WROOM-32E
- IDF version (run ``git describe --tags`` to find it): v4.4 Stable
- Build System:         CMake with idf.py
- Compiler version: xtensa-esp32-elf-gcc 8.4.0
- Operating System:  Windows
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: Yes: CLion

## Problem Description

ESP-IDF Uses LWIP for handling network interfaces(netif objects). In our project, we are using WiFi, Ethernet and Cellular(PPPoS) network interfaces. We want to switch accordingly. esp_netif abstraction of LWIP has some functionality to set a default network interface. That function specific is
 ```cpp
 static void esp_netif_set_default_netif(esp_netif_t *esp_netif)
{
    if (esp_netif->is_ppp_netif) {
        esp_netif_ppp_set_default_netif(esp_netif->netif_handle);
    } else {
        netif_set_default(esp_netif->netif_handle);
    }
}
```
The documentation says it must be called from an lwip thread. Unfortunately it is unclear how this is called and how this should be used, hence I think some documentation is missing. Correct me if I'm wrong.

At the moment we are switching interfaces by using the base layer of lwip, and using the following code.
```cpp
netif* tmp;
NETIF_FOREACH(tmp){
// do stuff, switch interfaces etc.
}
```

We would like to use the abstraction layer of ESP_NETIF.

david-cermak, 2022-02-02:
Hi @crisvdn 

This ESP-NETIF API `esp_netif_set_default_netif()` is only meant to be used internally. This is called automatically whenever any interface is set *up* or *down*, so the default interface is chosen from those, that are up, with the highest `route_prio`. In that sense, it should be enough to configure the `route_prio` when initializing an interface.
Do you need to implement some additional logic when setting default netif? Could you please explain a bit what is your use case?

crisvdn, 2022-02-02:
Hi @david-cermak , thanks for your quick response.
Ah, as far as I understood it was an API function call and we could use it ourselves. That clears it up that its linked to the `route_prio`.

We have multiple use-cases regarding the network interfaces and a single development issue that has to be resolved with it.
Our interface combinations are primarily the following:
- Ethernet + Wifi
- Ethernet + Cellular.

Unfortunately our PHY(kzs8863) is not supported as of yet and a basic custom implementation of the driver is written. But it will always be "available/link up", even though an Ethernet cable is not connected. The `ETHERNET_EVENT_START` and `ETHERNET_EVENT_STOP` are not triggered.
I assume Ethernet has the highest priority by default, but this is something I would have to check. We do have proper IP_EVENTS we can listen to, so then we trigger the `netif_set_default()` function for Ethernet.

As of now, I want to switch manually to WiFi as a default interface when a connection to an access point is made.
Also manually switching to Cellular when the PPPoS connection is made and an IP is assigned. But we want to switch to Ethernet when an Ethernet cable is plugged in, no matter if WiFi or Cellular has been set up.

I dont think changing route_prio values dynamically is the solution here. So a manual function call would be the best solution in my eyes. Let me know what you think.

david-cermak, 2022-02-02:
@crisvdn Thanks for explaining!
Yes, I agree that for this use case it would be more convenient to add the `esp_netif_set_default_netif()` to public API of ESP-NETIF, so it could be called manually (and at the same time, optionally disable the `route_prio` functionality).
I will propose this API update to ESP-NETIF and share it here, soon.

closed 2022-03-12 by espressif-bot