espressif/esp-idf #8333: esp_wifi_stop does not release PM_LOCK for light sleep

shadowpho, 2022-02-02:
## Environment

- Development Kit:      Custom PCB (sch available at github but not relevant)
- Module or chip used:  ESP32-S2-Wrover
- IDF version: v4.3.2 and v4.4
- Build System:         idf.py
- Compiler version: crosstool-NG esp-2021r2-patch2 8.4.0 
- Operating System:     Windows
- Environment type: PowerShell
- Using an IDE?:  Yes (VSCode)
- Power Supply:         USB

## Problem Description
When calling esp_wifi_stop it does not seem to release the pm_lock and therefore the chip does not go into automatic light sleep.  I have the full code available at [My Github Repo](https://github.com/shadowpho/ESP32-Sensor-Software). 

My code calls start_wifi every 60s to transmit for ~1s and then stop_wifi.  You can see the code snippet I have below.  After stop_wifi it does a little more cleanup and then calls vTaskDelay (70). I have both PM_ENABLE and TICKLESS_SUPPORT configured.  I expect it to go to sleep but instead I still see ~10mA power. 

I added test call to esp_pm_dump_locks to see what's holding up the sleep and apparently ADC2 and Wifi is still holding locks that prevents tickless mode. I suspect the Wifi does not release the lock. 

I tested both IDF4.4 and 4.3.2.  I suspect not many people tried the tickless mode yet. 

### Expected Behavior
Wifi releases PM_LOCK and system goes into low power mode (<1mA) between messages

### Actual Behavior
Wifi keeps lock and system stays up at 10mA -- the base line does not change at all. 
[image]

### Code to reproduce this issue
```cpp
void app_main()
{
   ...
   setup_wifi();
   start_wifi();
   send wifi message
   stop_wifi();
   ...
   esp_pm_config_esp32s2_t t;
   esp_pm_get_configuration(&t);
   printf("max%i, abp%i, tkless%i\n",t.max_freq_mhz,t.min_freq_mhz, t.light_sleep_enable);
   esp_pm_dump_locks(stdout);
   vTaskDelay(700 / portTICK_PERIOD_MS); //Idle time
}

void start_wifi(void)
{
   wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
   ESP_ERROR_CHECK(esp_wifi_init(&cfg));
   ...
   ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
   ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config));
   ESP_ERROR_CHECK(esp_wifi_start());
}
void stop_wifi()
{
    ESP_ERROR_CHECK(esp_wifi_disconnect());
    ESP_ERROR_CHECK(esp_wifi_stop());
    ESP_ERROR_CHECK(esp_wifi_deinit());
}
```

## Debug Logs

```
max240, abp40, tkless0

Lock stats:
adc2              APB_FREQ_MAX    0    0
wifi              APB_FREQ_MAX    0    0
rtos0             CPU_FREQ_MAX    0    1
```

## ELF file and SDKConfig

[Elf_And_SDKCONFIG.zip](https://github.com/espressif/esp-idf/files/7984363/Elf_And_SDKCONFIG.zip)

shadowpho, 2022-02-02:
Wanted to add more details:

- My board has an external 32KHz that it uses for accurate timing.
- All the code that can goes into external flash.  Wifi functions, FreeRTOS, go into external flash to preserve internal memory.
- Error reproduces on battery power, USB power and accurate power supply. 

Is there a way to bypass the lock?  I essentially tear down Wifi and rebuild it every time so if I could just delete the lock that would be good enough for now.

zhangyanjiaoesp, 2022-03-29:
we will look into this issue as soon as possible.

shadowpho, 2022-04-03:
Thank you! Let me know if you need more information.

jack0c, 2022-04-27:
@shadowpho I think the problem is not related with esp_wifi_stop.
According the log, it show adc2 and wifi do not keep the pm_lock, but rtos0 require the pm_lock.
When you call `esp_pm_dump_locks(stdout);`,  the context is in a task, which will always keep the pm_lock.

About the 10mA current, it means the system didn't go into light sleep.
If you want to debug this problem, the `esp_pm_dump_locks(stdout);` should be called at beginning  of `vApplicationSleep`.

xiaojie76, 2022-04-27:
@shadowpho  According the log : max240, abp40, tkless0,  it seems that light sleep is not enabled. Do you use` esp_pm_configure `to configure `.light_sleep_enable = true` ?

lennyJL, 2022-04-27:
Hi @shadowpho. To enable light sleep, you can refer code in `esp-idf/examples/wifi/power_save`. And you can refer https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/power_management.html for more details.

shadowpho, 2022-04-28:
> Hi @shadowpho. To enable light sleep, you can refer code in `esp-idf/examples/wifi/power_save`. And you can refer https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/power_management.html for more details.

Ahhh I see what you mean.  Ok, I checked the IDLE_TASK in menuconfig and thought that was all I needed to enable it. 

I added the following code to my startup code:
`    {
        esp_pm_config_esp32s2_t sleep_cfg = {0};
        sleep_cfg.light_sleep_enable = true;
        sleep_cfg.max_freq_mhz = 240;
        sleep_cfg.min_freq_mhz = 80;
        assert(ESP_OK == esp_pm_configure((const void*)&sleep_cfg));
    }
`
However, now when I do a   `vTaskDelay(300 / portTICK_PERIOD_MS);`  the chip stays asleep and does not wake up. This is unexpected as it works if I just do 
` esp_sleep_enable_timer_wakeup(1000000);
 esp_light_sleep_start();
`

Does the vTaskDelay not do enable_timer_wakeup?

xiaojie76, 2022-04-28:
@shadowpho The normal process is like this:
```
vTaskDelay(300 / portTICK_PERIOD_MS);
    prvIdleTask
       vApplicationSleep
           esp_sleep_enable_timer_wakeup(); // sleep time should be slightly less than 300ms
           esp_light_sleep_start();
```
You can check if a reasonable sleep time is set in `vApplicationSleep`.

shadowpho, 2022-04-29:
> @shadowpho The normal process is like this:
>
> ```
> vTaskDelay(300 / portTICK_PERIOD_MS);
>     prvIdleTask
>        vApplicationSleep
>            esp_sleep_enable_timer_wakeup(); // sleep time should be [...]

Thank you! That is very helpful information. I will continue debugging on my side.

jack0c, 2022-06-15:
@shadowpho any updates?

shadowpho, 2022-07-08:
> @shadowpho <https://github.com/shadowpho> are you still doing this
> project?

Yes, my apologies! I had some issues when updating to the latest toolchain
so I am sorting that out first. Will provide an update after that!

closed 2022-10-17 by Alvin1Zhang