espressif/esp-idf #8334: requirements.txt missing

franz-ms-muc, 2022-02-02:
using current "master" the requirements.txt is missing. 

the eclipse plugin searches for it ...

franz-ms-muc, 2022-02-02:
pasting the 
https://github.com/espressif/esp-idf/blob/9fc95216dad82dbaa93c5f277c0be54b0775947d/requirements.txt
into the esp-idf folder seems to work.

dobairoland, 2022-02-02:
`requirement.txt` is no longer used in the master branch.

PTAL @kolipakakondal

kolipakakondal, 2022-02-02:
Hi @franz-ms-muc Thanks for bringing this up. Just realized it was used as part of the esp-idf folder validation. Fix will be provided as part of the idf-eclipse-plugin v2.4.1 release.

Workaround: Having a file with requirements.txt(not concerned about content) in the esp-idf root directory would avoid this error.

kolipakakondal, 2022-02-02:
Hi @dobairoland Could you transfer this issue to [eclipse project](https://github.com/espressif/idf-eclipse-plugin/)?