espressif/esp-idf #8335: BLE server, remote address does not match mobile address

jpbaptista30, 2022-02-02:
Dear all.

II tried to test the ble gatt_server example and connected from an Android phone.

Then I noticed that the remote address doesn't match my mobile.

My mobile address is 2C:D0:66:52:B7:8B and esp32 shows me 66:42:b5:59:43:a5

`
case ESP_GATTS_CONNECT_EVT:

     const uint8_t *point = esp_bt_dev_get_address();
        for (int i = 0; i < 6; i++) {
            char str[3];
            sprintf(str, "%02X", (int)point[i]);
            printf(str);
            if (i < 5){
                printf(":");
            }
        }
        ESP_LOGI(GATTS_TAG, "CONNECT_EVT, conn_id %d, remote %02x:%02x:%02x:%02x:%02x:%02x:",
                 param->connect.conn_id,
                 param->connect.remote_bda[0], param->connect.remote_bda[1], param->connect.remote_bda[2],
                 param->connect.remote_bda[3], param->connect.remote_bda[4], param->connect.remote_bda[5]);
        
        gl_profile_tab[PROFILE_B_APP_ID].conn_id = param->connect.conn_id;
        break;`

negativekelvin, 2022-02-02:
https://www.novelbits.io/bluetooth-address-privacy-ble/

jpbaptista30, 2022-02-02:
Hi @negativekelvin 

I tried it with my computer and the peer address is fine. Is it possible that the bd address on the cell phone is secret or modified?

negativekelvin, 2022-02-02:
Yes it is very common especially for mobile devices to use random addresses for privacy, details are described in the link above

jpbaptista30, 2022-02-03:
Hi @negativekelvin  
I tested the bt_spp_acceptor (classic_bt) and in this example the mobile bd addr that esp32 receives is already correct.
Will you know what the problem will be?

negativekelvin, 2022-02-03:
Bluetooth classic profiles do not support address privacy

jpbaptista30, 2022-02-03:
@negativekelvin my esp32 is a gatt_server and my mobile phone is a client.
Is it possible to read the real mac address of the mobile on the ble gatt server?

negativekelvin, 2022-02-03:
No, I don't think so. Why do you need it?

jpbaptista30, 2022-02-03:
@negativekelvin 

I try to explain my application more.

In my app it will only connect with cell phones, but it can connect with one or more cell phones. The device uses some restrictions. In my application there are two types of user, admin users and normal users.

Admin users will be able to read and write to some resources, but normal users will not. So I need to check if that user is admin. I save the admin user's address in memory and I want to check their address when I get a read request. Do you suggest something to implement this feature?

negativekelvin, 2022-02-03:
You have to use ble security and bonding, at successful bonding you will get an identity address that uniquely identifies the device.

jpbaptista30, 2022-02-03:
@negativekelvin 
ok, but on my system it is possible to have more than 32 devices connected. Over time I will need to remove devices from the bound list.
I need the real mac address for validate the user.

jpbaptista30, 2022-02-03:
Is it possible use the irk key to decoding the mac address?

negativekelvin, 2022-02-03:
No.  You might be able to maintain your own database of identity address and IRK.  But maybe you can consider a different authentication mechanism like tokens or cmac.

jpbaptista30, 2022-02-03:
@negativekelvin 
For me it is impossible to use database. Is it possible in the android app to send a private mac address?
The problem is that my app doesn't use internet.

I really need to identify the actual mac address of the client.

negativekelvin, 2022-02-03:
I don't think it is possible to force an Android device to use a fixed ble address.  It doesn't seem secure anyway since address could be spoofed.  Have you tried increasing the size of the bond list in bluedroid or nimble?  How many devices do you need to support?

jpbaptista30, 2022-02-04:
@negativekelvin 

I've thought about saving to the title list, but I need to save up to 500 devices.
Is it possible to use the server in dual mode to receive the real address?
Why if my computer is connected to ESP BlE server, the real bd address is read?

negativekelvin, 2022-02-04:
Your computer chooses to not use address privacy.  Even if you have the real address you can't correlate it with the random BLE address.  I think you need a new strategy.

jpbaptista30, 2022-02-04:
Hi @negativekelvin.
OK. I'm already thinking of a new strategy. I'm currently thinking of using an ssp server, but is it possible to connect multiple clients at the same time?

Weijian-Espressif, 2022-03-30:
@jpbaptista30  In order to protect privacy, the mobile terminal uses a resolvable random address It will change every time the Bluetooth of the mobile phone is restarted

jpbaptista30, 2022-03-30:
@Weijian-Espressif 
Thanks for your answer. Is it possible to get the static address of the mobile phone?

Weijian-Espressif, 2022-04-05:
@jpbaptista30  After binding, you can get the identity address of the peer device. For the mobile phone, its identity address is the public address.

closed 2022-04-28 by Alvin1Zhang