espressif/esp-idf #8336: Crash in xEventGroupSetBits after a few hours of running

maBarabas, 2022-02-02:
## Environment

- Development Kit:     none
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.3.1-dirty (we have updated nimble to https://github.com/espressif/esp-nimble/commit/94afe2711752ba08bef93780be1b9e5143a87159, there are no other changes)
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:        external 5V

## Problem Description

After a few hours of running, a crash happens due to null pointer dereference in FreeRTOS code. We have reproduced this on multiple hardware units and multiple times on each unit.

We are currently running the test with latest esp-idf, and trying to create a minimal project for reproduction.

Our firmware is setting event bits every second to update time information on a screen. Before the crash, everything works as expected.

It looks like there might be a race condition where the task is already unblocked from a timeout when `xEventGroupSetBits` is trying to unblock it?

This looks similar to https://github.com/espressif/esp-idf/issues/242. The [commit message](https://github.com/espressif/esp-idf/commit/01c4a980ee05b5eddefd9d01a2bbdecde9944ce7) for the fix says the race condition is less likely to happen. Does this mean that the issue is still present?

### Expected Behaviour

No crashes.

### Actual Behaviour

See description above.

### Steps to reproduce

1. Start firmware
2. After a while, the crash happens

### Code to reproduce this issue

```cpp
// This is how we set the bits:
// The DISPLAY_INFO bits and change value are provided by a UART peripheral connected to the system
    EventBits_t bits = 0;
    if(change & DISPLAY_INFO_MASK_NEW_INDEX)
    {
        bits |= CD_NEW_INDEX;
    }
    if(change & DISPLAY_INFO_MASK_NEW_TRACK)
    {
        bits |= CD_NEW_TRACK;
    }
    if(change & DISPLAY_INFO_MASK_TIME_CHANGED)
    {
        bits |= CD_NEW_TIME;
    }

    xEventGroupSetBits(xCdEventGroup, bits);

// This is how we are waiting for them in a different task:
    xEventGroupValue = xEventGroupWaitBits( xCdEventGroup,
                                            CD_NEW_INDEX |
                                            CD_NEW_TRACK |
                                            CD_NEW_TIME |
                                            CD_STOP_OPEN_CLOSE_PRESSED |
                                            CD_STATE_CHANGED,
                                            pdTRUE,
                                            pdFALSE,
                                            20 / portTICK_PERIOD_MS);
```

## Debug Logs

backtrace: https://gist.github.com/maBarabas/c0d683adfdaf6c8769a0b310e78c93a8

maBarabas, 2022-02-28:
I've ran a test with the timeout on `xEventGroupWaitBits` set to 0, and this has caused to issue to go away. Seems likely that there is a race condition between the timeout and the call to `xEventGroupSetBits`.

Haven't managed to set up a devkit to reproduce the issue yet.

SpComb, 2022-07-02:
I am debugging a very similar crash in https://github.com/qmsk/esp/issues/23 with esp-idf v4.4 on an ESP-32, where a task running on core 0 is calling `xEventGroupWaitBits()` with a 10ms timeout (1 tick), and another task running on core 1 is calling `xEventGroupSetBits()` at a high frequency (~60Hz). This occasionally causes a `xEventGroupSetBits()` -> `vTaskRemoveFromUnorderedEventList()` -> `uxListRemove()` panic, and occasionally just locks up the ESP-32 without triggering a watchdog reset (!).

Looking at the https://github.com/espressif/esp-idf/issues/242 bug and https://github.com/espressif/esp-idf/commit/01c4a980ee05b5eddefd9d01a2bbdecde9944ce7 fix, this code change only addressed the `xTaskRemoveFromEventList()` used by the queue implementation, and not the `vTaskRemoveFromUnorderedEventList()` used by the eventgroup implementation.

To me it looks like the `xEventGroupSetBits()` / `xEventGroupWaitBits()` implementation would suffer from the same kind of race condition, as the functions traverse the `xEventListItem` -> `xTasksWaitingForBits`  list items with the per-EventGroup `eventGroupMux`, but the timer interrupt -> `xTaskIncrementTick()` manipulates the `xEventListItem` with the `xTaskQueueMutex` held? This would explain this race condition, with the timer interrupt on core0 removing the waiting task's `xEventListItem` from the EventGroup's `xTasksWaitingForBits` list at the same time as the core1 task calling `xEventGroupSetBits()` is traversing the same list, causing it to call `vTaskRemoveFromUnorderedEventList()` on the same list item that was already removed by the timer interrupt?

SpComb, 2022-07-02:
https://gist.github.com/SpComb/909d20847645aa22361cf4d3e8bc3700 here is a simple core0 `xEventGroupSetBits()`  -> core1 `xEventGroupWaitBits()` repro case, which reliably either crashes or locks up

SpComb, 2022-07-02:
Here's a PoC patch that adds a similar "check for empty list with `xTaskQueueMutex` held" workaround for the timeout SMP race -> empty list case as https://github.com/espressif/esp-idf/commit/01c4a980ee05b5eddefd9d01a2bbdecde9944ce7 to `vTaskRemoveFromUnorderedEventList()`.

It fixes the repro case linked above, but I'm not sure if the early return discarding the `xItemValue` breaks the `xEventGroupWaitBits()` logic and would lead to other misbehavior in the race condition case? The early return shouldn't have any effect on the normal behavior outside of the race condition, though :thinking: 

```
diff --git a/components/freertos/tasks.c b/components/freertos/tasks.c
index 40a07839e7..86a3778019 100644
--- a/components/freertos/tasks.c
+++ b/components/freertos/tasks.c
@@ -3741,6 +3741,12 @@ void vTaskRemoveFromUnorderedEventList( ListItem_t * pxEventListItem,
 
     taskENTER_CRITICAL();
 
+    if ( listLIST_ITEM_CONTAINER( pxEventListItem ) == NULL ) {
+      // XXX: https://github.com/espressif/esp-idf/issues/8336
+      taskEXIT_CRITICAL();
+      return;
+    }
+
     /* Store the new item value in the event list. */
     listSET_LIST_ITEM_VALUE( pxEventListItem, xItemValue | taskEVENT_LIST_ITEM_VALUE_IN_USE );
```

SpComb, 2022-07-02:
> but I'm not sure if the early return discarding the `xItemValue` breaks the `xEventGroupWaitBits()` logic and would lead to other misbehavior in the race condition case?

Yes, when `xEventGroupSetBits()` calls `vTaskRemoveFromUnorderedEventList()` and the event bits are ignored in the race condition case:

https://github.com/espressif/esp-idf/blob/v4.4.1/components/freertos/event_groups.c#L647-L665

then it will still clear those bits from `pxEventBits->uxEventBits`:

https://github.com/espressif/esp-idf/blob/v4.4.1/components/freertos/event_groups.c#L673-L675

causing those bits to get lost in the case condition case - the task calling `xEventGroupWaitBits()` will get a timeout return value, and the bits will not show up in the next call to `xEventGroupWaitBits()` :confused:

SpComb, 2022-07-02:
Updated the repro at https://gist.github.com/SpComb/909d20847645aa22361cf4d3e8bc3700 to catch the lost bits, and confirm that the simplistic PoC patch above causes the output to stall at approximately `sync=14537`.

I have an improved patch that also fixes `xEventGroupSetBits()` to not clear the bits in the race condition case, and with this change I'm fairly confident that the event groups now behave correctly in this SMP case. However, the patch requires changes to the `vTaskRemoveFromUnorderedEventList()`  -> `xTaskRemoveFromUnorderedEventList()` return value such that `xEventGroupSetBits()` can omit the `uxBitsToClear` if the task was not actually unblocked.  I'm hesitant to submit a PR as I'm not sure if such an invasive change to the FreeRTOS APIs is a good idea?

Dazza0, 2022-07-04:
@SpComb Thanks for the issue report. We've noticed this issue internally as well and have a fix already merged in our internal master (currently waiting sync with GitHub).

This issue affects any FreeRTOS API that manipulates tasks/event lists manually (without going through the `vTask...OnEventList()` API), namely `xEventGroupSetBits()` and `vEventGroupDelete()`. The fix we went with differs quite a bit from https://github.com/espressif/esp-idf/pull/9271. We simply make sure that `xEventGroupSetBits()` and `vEventGroupDelete()` take the task list locks in `tasks.c` before accessing the tasks/event list.

In case you want to test our fix right now, I've attached the contents of the patch based on v4.4 below. Simply copy the contents into a `.patch` file and `git am the_file.patch` to apply it. I'll keep this issue open until the v4.4. backport has been merged.

```patch
From 554fed0fc3ab604bc8532fc51e6ef06bf1f47941 Mon Sep 17 00:00:00 2001
From: Darian Leung <darian@espressif.com>
Date: Mon, 4 Jul 2022 20:28:31 +0800
Subject: [PATCH] freertos: Fix event group task list race condition

FreeRTOS synchronization primitives (e.g., queues, eventgroups) use various event lists (i.e., task lists) to track what
tasks are blocked on a current primitive. Usually these event lists are accessed via one of the event lists functions
(such as vTask[PlaceOn|RemoveFrom]UnorderedEventList()), which in turn ensure that the global task list spinlock
(xTaskQueueMutex) is taken when accessing these lists.

However, some functions in event_groups.c manually traverse their event lists. Thus if a tick interrupt occurs on
another core during traversal and that tick interrupt unblocks a task on the event list being traversed, the event list
will be corrupted.

This commit modifies the following event_groups.c functions so that they take the global task list lock before
traversing their event list.

- xEventGroupSetBits()
- vEventGroupDelete()
---
 components/freertos/event_groups.c          | 16 ++++++++++++++++
 components/freertos/include/freertos/task.h | 19 +++++++++++++++++++
 components/freertos/tasks.c                 | 14 ++++++++++++++
 3 files changed, 49 insertions(+)

diff --git a/components/freertos/event_groups.c b/components/freertos/event_groups.c
index f7f38c4607..906fa0a110 100644
--- a/components/freertos/event_groups.c
+++ b/components/freertos/event_groups.c
@@ -600,6 +600,10 @@ EventBits_t xEventGroupSetBits( EventGroupHandle_t xEventGroup,
     pxListEnd = listGET_END_MARKER( pxList ); /*lint !e826 !e740 !e9087 The mini list structure is used as the list end to save RAM.  This is checked and valid. */
 #ifdef ESP_PLATFORM // IDF-3755
     taskENTER_CRITICAL();
+    /* The critical section above only takes the event groups spinlock. However, we are about to traverse a task list.
+     * Thus we need call the function below to take the task list spinlock located in tasks.c. Not doing so will risk
+     * the task list's being changed while be are traversing it. */
+    vTaskTakeEventListLock();
 #else
     vTaskSuspendAll();
 #endif // ESP_PLATFORM
@@ -675,6 +679,8 @@ EventBits_t xEventGroupSetBits( EventGroupHandle_t xEventGroup,
         pxEventBits->uxEventBits &= ~uxBitsToClear;
     }
 #ifdef ESP_PLATFORM // IDF-3755
+    /* Release the previously held task list spinlock, then release the event group spinlock. */
+    vTaskReleaseEventListLock();
     taskEXIT_CRITICAL();
 #else
     ( void ) xTaskResumeAll();
@@ -693,6 +699,12 @@ void vEventGroupDelete( EventGroupHandle_t xEventGroup )
 
     // IDF-3755
     taskENTER_CRITICAL();
+#ifdef ESP_PLATFORM
+    /* The critical section above only takes the event groups spinlock. However, we are about to traverse a task list.
+     * Thus we need call the function below to take the task list spinlock located in tasks.c. Not doing so will risk
+     * the task list's being changed while be are traversing it. */
+    vTaskTakeEventListLock();
+#endif
     {
         while( listCURRENT_LIST_LENGTH( pxTasksWaitingForBits ) > ( UBaseType_t ) 0 )
         {
@@ -702,6 +714,10 @@ void vEventGroupDelete( EventGroupHandle_t xEventGroup )
[... 44 lines cut ...]
@@ -3733,6 +3733,20 @@ BaseType_t xTaskRemoveFromEventList( const List_t * const pxEventList )
 }
 /*-----------------------------------------------------------*/
 
+#ifdef ESP_PLATFORM
+void vTaskTakeEventListLock( void )
+{
+    /* We call the tasks.c critical section macro to take xTaskQueueMutex */
+    taskENTER_CRITICAL();
+}
+
+void vTaskReleaseEventListLock( void )
+{
+    /* We call the tasks.c critical section macro to release xTaskQueueMutex */
+    taskEXIT_CRITICAL();
+}
+#endif // ESP_PLATFORM
+
 void vTaskRemoveFromUnorderedEventList( ListItem_t * pxEventListItem,
                                         const TickType_t xItemValue )
 {
-- 
2.25.1

```

AxelLin, 2022-07-05:
> ```diff
> diff --git a/components/freertos/event_groups.c b/components/freertos/event_groups.c
> index f7f38c4607..906fa0a110 100644
> --- a/components/freertos/event_groups.c
> +++ b/components/freertos/event_groups.c
> @@ -600,6 +600,10 @@ EventBits_t xEventGroupSetBits( EventGroupHandle_t xEventGroup,
>      pxListEnd = listGET_END_MARKER( pxList ); /*lint !e826 !e740 !e9087 The mini list structure is used as the list end to save RAM.  This is checked and valid. */
>  #ifdef ESP_PLATFORM // IDF-3755
>      taskENTER_CRITICAL();
> +    /* The critical section above only takes the event groups spinlock. However, we are about to traverse a task list.
> +     * Thus we need call the function below to take the task list spinlock located in tasks.c. Not doing so will risk
> +     * the task list's being changed while be are traversing it. */
> +    vTaskTakeEventListLock();
>  #else
>      vTaskSuspendAll();
>  #endif // ESP_PLATFORM

I don't quite understand your changes.
What is the difference of the first  taskENTER_CRITICAL(); call v.s. the vTaskTakeEventListLock() call (which just calls taskENTER_CRITICAL();?

(Both are within #ifdef ESP_PLATFORM guard).

Dazza0, 2022-07-05:
@AxelLin The `#define taskENTER_CRITICAL()` is different for each FreeRTOS source file.

Dazza0, 2022-07-07:
Master fix 1625a3aae210b605686fb202d3b66fa6a91486af

AxelLin, 2022-07-15:
> I'll keep this issue open until the v4.4. backport has been merged.

@Dazza0
This issue was reported against v4.3.1, I think v4.3 branch also needs fix?

Dazza0, 2022-07-19:
@AxelLin This should be backported all the way to v4.1 (i.e., oldest version that is still in its maintenance period) as this bug is quite severe.

maBarabas, 2022-07-20:
@Dazza0 Do you have a time estimate of when this fix will be released for 4.4 or 4.3?

Dazza0, 2022-07-27:
@maBarabas Backport for v4.3 has already been merged internally and awaiting GitHub sync, backport for v4.4 is still pending internal review. So I guess maybe a week or two.

Dazza0, 2022-08-11:
Seems like all backports have been merged, commits listed below:

- Master/v5.0: 1625a3aae210b605686fb202d3b66fa6a91486af
- v4.4: d465316f0bd4257fb63093716a92bf8ca07999f1
- v4.3: ad39eaaeb06837185ade90f4df9e1a84dc64743c

Closing this issue.

closed 2022-08-11 by Dazza0

AxelLin, 2022-08-12:
@Dazza0 
https://github.com/espressif/esp-idf/issues/8336#issuecomment-1188543771
I'm curious if this fix is not required for v4.2 and v4.1?

Dazza0, 2022-08-12:
@AxelLin 

v4.2 : 5e15bafd26374981b4ad06e545d0c9ae65217ea0
v4.1 is still waiting to be merged internally.

maBarabas, 2022-08-12:
@Dazza0 When can we expect a new release to be tagged from the v4.4 branch? I would like to update our firmware with the fixes ASAP, but don't want to use the untagged release branch.

It looks like the recent v4.4.2 release does not fix this race.

igrr, 2022-08-12:
@maBarabas Release v4.4.2 was tagged just recently. Currently we are working on v4.2.4 bugfix release, and will consider if we need to do v4.4.3 or v4.3.4 afterwards. I'm sorry, we can't give a concrete timeline for the v4.4.3 release yet.