espressif/esp-idf #8340: RTC GPIO deep sleep ESP32S3

hardyacht, 2022-02-03:
I wrote this here, because our offical contact William  檀军华 is on holiday and referred me here. We are a big (quantities) of espressif.
This topic is important, we are about to start mass production. 

We want to use 3 Pins to wakeup from deep sleep, Pin 8 going to 1, Pin 20 and 21 going to 0.  

There is no API to do that. But what is worse is that the datasheet does not contain the information about ext1 and ext0 source as wakeup. the chapter about RTC_CNTL_EXT_WAKEUP is completely missing in 0.3: https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf

For S3 there is no config like : SOC_GPIO_SUPPORT_DEEPSLEEP_WAKEUP
I cannot use use esp_sleep_enable_ext1_wakeup, because the mode is set once and we have inverted pins.
I cannot use esp_sleep_enable_ext0_wakeup, it overwrite the pin is selected. 

I have already looked deep into hal, at a level that users should not use, but find no solution.

ginkgm, 2022-02-14:
Hi @hardyacht ,

For your case, you may need to use EXT0 and EXT1 together, with different triggering level.

I have tried that on master, both ext0 and ext1 is working properly. (However, the logic of EXT1 seems to be ANY_HIGH or ANY_LOW, unlike the enum name, I'm checking this).

You may need to properly configure the internal pullup/downs if you don't have external ones to make the pins stable during deepsleep. See https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html#external-wakeup-ext0 for

Alvin1Zhang, 2022-05-16:
Thanks for reporting, sorry for slow turnaround, fix is available https://github.com/espressif/esp-idf/commit/5a2f672d5aa60ed6286bf232afed804d2f4c1c67, feel free to reopen.

closed 2022-05-16 by Alvin1Zhang

hardyacht, 2022-05-16:
@Alvin1Zhang in which official release will this fix be included?

Alvin1Zhang, 2022-05-17:
Thanks for reporting and the notes, the fix will be included in the forthcoming 5.0, and will back port to release/4.3 and release/4.4, thanks.

hardyacht, 2022-05-17:
Is there a time when a new tagged release is planned for the 4.4?