espressif/esp-idf #8344: Internal DAC audio broke in v4.4

cfint, 2022-02-03:
## Environment

- Development Kit:      none
- Kit version: none
- Module or chip used:  ESP32-WROOM-32
- IDF: v5.0-dev-1446-ga470ae224d
- Build System:         idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         external 5V

## Problem Description

I2S internal DAC produces garbled sound with A2DP-Sink example. I've bisected the problem to commit f7f8c9c11fa7caeb51b3d728897f69b81fde79cc. The prior commit (2f1247e1c4dd8bb9401d6b1256d34dd7319fbe5c) does not compile. The most recent working commit on the `v4.4` branch is 12c76b4c5c5d374f5d40eb6ad8badb38a5ca6e59.

The problem still exists on on current master (a470ae224d2479cbb1019a0b69490a1249e6a562)

### Expected Behavior

Analog sound playback through I2S internal DAC.

### Actual Behavior

Internal DAC doesn't play correct sound.

### Steps to reproduce

1. Install A2DP-Sink example on an ESP32 board.
2. Pair an iPhone bluetooth with the ESP32.
3. Play music on iPhone through Bluetooth audio.
4. Hear garbled audio.

negativekelvin, 2022-02-03:
https://github.com/espressif/esp-idf/pull/8327

cfint, 2022-02-06:
> #8327

That patch works with [pschatzmann/ESP32-A2DP](https://github.com/pschatzmann/ESP32-A2DP). The A2DP-Sink example is kind of working. It has much louder audio and it's full of static.

L-KAYA, 2022-02-17:
> > #8327
> 
> That patch works with [pschatzmann/ESP32-A2DP](https://github.com/pschatzmann/ESP32-A2DP). The A2DP-Sink example is kind of working. It has much louder audio and it's full of static.

@cfint That may be caused by the input signed value are output in unsigned value, try the following change to see if it can help:
in `examples/bluetooth/bluedroid/classic_bt/a2dp_sink/main/bt_app_av.c` line 403, replace the function `bt_app_a2d_data_cb` with
```c
void bt_app_a2d_data_cb(const uint8_t *data, uint32_t len)
{
    uint8_t *recv_data = malloc(len);
    for (int i = 0; i < len; i++) {
        recv_data[i] = data[i] + 128;
    }
    write_ringbuf(recv_data, len);
    free(recv_data);

    /* log the number every 100 packets */
    if (++s_pkt_cnt % 100 == 0) {
        ESP_LOGI(BT_AV_TAG, "Audio packet count: %u", s_pkt_cnt);
    }
}
```

karawin, 2022-02-21:
Working with patch https://github.com/espressif/esp-idf/pull/8327/files

closed 2022-03-03 by espressif-bot

dzanis, 2022-06-03:
This fix broke the I2S ADC, had to revert the old code back