espressif/esp-idf #8346: Issue on esp_http_client_example

luanfrj, 2022-02-04:
Hi,

I was doing some tests with esp_http_client and used the _http_event_handler as show in esp_http_client_example.

I think the buffer must be freed always in HTTP_EVENT_DISCONNECTED, not only if mbedtls errors happens.
In my tests if I don't free buffer on disconnect event heap is corrupted after a sequence of HTTP requests.

ESP-YJM, 2022-02-07:
@luanfrj Can you specify which buffer needs to be freed, preferably by posting the code.

luanfrj, 2022-02-07:
Hi @ESP-YJM ,
The buffer is **output_buffer**. In the file **esp_http_client_example.c**, this code snippet, starting from line 98, could be changed from:
```c
        case HTTP_EVENT_DISCONNECTED:
            ESP_LOGI(TAG, "HTTP_EVENT_DISCONNECTED");
            int mbedtls_err = 0;
            esp_err_t err = esp_tls_get_and_clear_last_error(evt->data, &mbedtls_err, NULL);
            if (err != 0) {
                if (output_buffer != NULL) {
                    free(output_buffer);
                    output_buffer = NULL;
                }
                output_len = 0;
                ESP_LOGI(TAG, "Last esp error code: 0x%x", err);
                ESP_LOGI(TAG, "Last mbedtls failure: 0x%x", mbedtls_err);
            }
            break;
```
To:
```c
        case HTTP_EVENT_DISCONNECTED:
            ESP_LOGI(TAG, "HTTP_EVENT_DISCONNECTED");
            int mbedtls_err = 0;
            esp_err_t err = esp_tls_get_and_clear_last_error(evt->data, &mbedtls_err, NULL);
            if (err != 0) {
                output_len = 0;
                ESP_LOGI(TAG, "Last esp error code: 0x%x", err);
                ESP_LOGI(TAG, "Last mbedtls failure: 0x%x", mbedtls_err);
            }
            if (output_buffer != NULL) {
                free(output_buffer);
                output_buffer = NULL;
            }
            break;
```
In this case, when disconnect event happens it woud free the **output_buffer** even when mbedtls return no errors.

ESP-YJM, 2022-02-08:
@luanfrj Yes, you are right. Thanks for reporting it.

luanfrj, 2022-02-08:
Thanks for the answer @ESP-YJM.
If possible, I can fork and send a pull request with the fix.

Regards,
Luan

closed 2022-02-10 by espressif-bot