espressif/esp-idf #8349: How do I check, is there any issue with direct use of "struct netif" instead of "struct esp_netif_t " for ethernet implementation?

enthunilu, 2022-02-04:
## Environment
Development Kit: ESP32-Ethernet kit_A_V1.1
Kit version
Module or chip used: ESP32
IDF version :v4.2.1
Build System: Make|CMake|idf.py
Compiler version : xtensa-esp32-elf-gcc (crosstool NCG esp-2020r3)8.4.0
Operating System: Windows
(Windows only) environment type: Command Prompt
Using an IDE?: Yes (Eclipse)
Power Supply: USB

## Problem Description

Currently I am using ESP32 ethernet kit. On which I have added ethernet support. And I am able to ping device. 
In my code I am using
struct netif     lwip_netif; 
instead of 
esp_netif_t    *eth_netif;  // This is used in the ethernet example code available in esp-idf 

So, can anyone please suggest will it impact on the ethernet functionality if I am not using esp_netif_t? 
Due to some limitation I can not use example code as it is. So, wanted to know limitation if any on this approach.

Thanks!!

david-cermak, 2022-02-07:
Hi @enthunilu 

It is perfectly okay to interface lwip directly when adding new network interfaces. It's only considered a bit more advanced use-case, as you'd have *know what you're doing* (follow exactly the lwip guidelines, take care of thread-safety issues,...)
I would generally recommend using esp-netif component for these reasons:
* Predefined interfaces (IDF provides some default configs for the most widely used interfaces, like WiFi, Ethernet, PPPoS, SLIP)
* Access to some lwip APIs should be thread safe
* Test coverage (The default esp-netif configs are actively tested by Espressif)

david-cermak, 2022-02-15:
@enthunilu Any other concern of question?

enthunilu, 2022-02-16:
Hi @david-cermak,

Thank you for your quick support. I was checking for thread safety part which took some time.
I understood it now. So, no other concern.
Thanks again!!

Closing this ticket now.

closed 2022-02-16 by enthunilu