espressif/esp-idf #8351: ESP-HTTPS clients and OTA failing when is_async == true and CONFIG_ESP_CONSOLE_UART_NONE=y

eritterlin, 2022-02-04:
----------------------------- Delete below -----------------------------
## Environment

- Module or chip used:  ESP32
- IDF version (run ``git describe --tags`` to find it):
    // v4.4
- Build System:        idf.py
- Compiler version xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: ESP-IDF VSCode environment
- Power Supply:         USB

## Problem Description
 This issue is easiest to reproduce with the OTA module. 

When the HTTP Client performing the ESP OTA is set to "is_async == true", and the following are set in the sdkconfig, the OTA fails. 
This can be repeated with other HTTP clients, and in some testing, it looks like data is being dropped somewhere. For example, when attempting to download 10k of data, my device would only download about 4 or 5k of that, and return ESP_OK. 

# CONFIG_CONSOLE_UART_DEFAULT is not set
# CONFIG_CONSOLE_UART_CUSTOM is not set
CONFIG_ESP_CONSOLE_UART_NONE=y
CONFIG_CONSOLE_UART_NUM=-1

Note that when the UART Console is defined, all the http clients work as expected.

### Expected Behavior
 The OTA would complete successfully 
### Actual Behavior
The OTA never completes
### Steps to reproduce

1. config sdkconfig with the above options
2. Run code attached below (pointing to your own OTA image, etc... 

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
esp_err_t do_firmware_upgrade(void)
{
   esp_http_client_config_t config;
   memset(&config, 0, sizeof(config));
   config.is_async = true;
   config.buffer_size = 2048;

   esp_err_t ret = esp_https_ota(&config);
   if (ret == ESP_OK)
   {
      ESP_LOGE(TAG, "ESP OTA Complete");
      esp_restart();
   }
   else
   {
      return ESP_FAIL;
   }
   return ESP_OK;
}

ESP-YJM, 2022-02-07:
@eritterlin When you disable the Console, no log output will print and how do you know that `For example, when attempting to download 10k of data, my device would only download about 4 or 5k of that, and return ESP_OK`. I follow your sdkconfig and set `is_async  = true`, it can OTA ok with master branch.

eritterlin, 2022-02-07:
@ESP-YJM  I've been writing out the number of bytes written and limited debug info to an external device. It seems random, but eventually it would appear that the HTTP Client callback stops receiving events to write data. Note that the operations complete successfully when is_async = false. 

Attached is my full SDK Config, I've reproduced this issue in 4.3 and 4.4. Maybe there is another interaction with my over-all sdkconfig? 

In the meantime I'll pull down the master branch and see if I can reproduce your results.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8016311/sdkconfig.txt)

ESP-YJM, 2022-02-08:
@eritterlin For HTTP OTA is not supported in v4.3 and v4.4 with `is_async  = true`. It can support on master branch. Do you use HTTP or HTTPS?

ESP-YJM, 2022-02-08:
@eritterlin When you enable the console output, this issue can't be reproduced? I need output log to trace the flow.

eritterlin, 2022-02-08:
@ESP-YJM  I am using HTTPS OTA, I am aware that HTTP is not supported in 4.3 and 4.4, and you are correct, everything works properly when I have the console enabled. It's the weirdest thing I've seen, I wouldn't have thought that two seemingly distinct components would interact like this.

eritterlin, 2022-02-08:
@ESP-YJM I would be overjoyed to provide logs, but alas, I don't think that's going to be possible in this case, is there anything else I could do?

ESP-YJM, 2022-02-10:
@eritterlin I think you can capture the wifi air packet to check the HTTPS flow.

ESP-YJM, 2022-02-10:
@eritterlin Could you please modify code as follow and test.
```
diff --git a/components/esp_http_client/esp_http_client.c b/components/esp_http_client/esp_http_client.c
index 68f4446..97a9991 100644
--- a/components/esp_http_client/esp_http_client.c
+++ b/components/esp_http_client/esp_http_client.c
@@ -1207,18 +1207,20 @@ static esp_err_t esp_http_client_connect(esp_http_client_handle_t client)
                 return ESP_ERR_HTTP_CONNECT;
             }
         } else {
-            int ret = esp_transport_connect_async(client->transport, client->connection_info.host, client->connection_info.port, client->timeout_ms);
-            if (ret == ASYNC_TRANS_CONNECT_FAIL) {
-                ESP_LOGE(TAG, "Connection failed");
-                if (strcasecmp(client->connection_info.scheme, "http") == 0) {
-                    ESP_LOGE(TAG, "Asynchronous mode doesn't work for HTTP based connection");
-                    return ESP_ERR_INVALID_ARG;
+            do {
+                err = esp_transport_connect_async(client->transport, client->connection_info.host, client->connection_info.port, client->timeout_ms);
+                if (err == ASYNC_TRANS_CONNECT_FAIL) {
+                    ESP_LOGE(TAG, "Connection failed");
+                    if (strcasecmp(client->connection_info.scheme, "http") == 0) {
+                        ESP_LOGE(TAG, "Asynchronous mode doesn't work for HTTP based connection");
+                        return ESP_ERR_INVALID_ARG;
+                    }
+                    return ESP_ERR_HTTP_CONNECT;
+                } else if (err == ASYNC_TRANS_CONNECTING) {
+                    ESP_LOGI(TAG, "Connection not yet established");
+                    //return ESP_ERR_HTTP_CONNECTING;
                 }
-                return ESP_ERR_HTTP_CONNECT;
-            } else if (ret == ASYNC_TRANS_CONNECTING) {
-                ESP_LOGD(TAG, "Connection not yet established");
-                return ESP_ERR_HTTP_CONNECTING;
-            }
+            } while(err == ASYNC_TRANS_CONNECTING);

```

eritterlin, 2022-02-15:
@ESP-YJM  Your patch did not solve the issue, I'm gonna try to get my hands on a sniffer and do some analysis there. I'll report back with my results, it could be a few days.

ESP-YJM, 2022-02-16:
@eritterlin Ok, besides, i want to know why you must need use async to OTA. Although you set `is_async = true`, the `esp_https_ota` still will block your process. I think you can use `is_async = false`.

closed 2022-04-11 by Alvin1Zhang