espressif/esp-idf #8352: Update PlatformIO to latest (4.4 as of writing)

rbray89, 2022-02-04:
https://registry.platformio.org/tools/platformio/framework-espidf should be update to the latest release of esp-idf (4.4 as of writing)

negativekelvin, 2022-02-04:
If you notice the owner of that package is platformio so I don't think espressif can update it

rbray89, 2022-02-04:
@negativekelvin I filed a ticket here as well, after looking for a long time to find a more appropriate place to file: https://github.com/platformio/platformio-registry/issues/7

Jason2866, 2022-02-06:
Afaik Submitting packages to platformio registry is moved from platformio to espressif.
(Same as for the toolchains). So requesting here seems the right place.
@pedrominatel  is this correct for idf releases?

igrr, 2022-02-06:
@Jason2866 I don't think so... @me-no-dev does have a way of uploading the toolchain packages (they show up in `espressif` namespace in platformio registry), however other packages (openocd-esp32, platform-espressif32) are under `platformio` namespace in platformio registry, so AFAIK we can't update them directly.

Jason2866, 2022-02-06:
@igrr There is no restriction in uploading. If you upload the package it will be usable under ownership `espressif`. Older packages are still under the ownership `platformio`.
Same as it is for the toolchains. Old ones are still under the ownership of platformio.
So @me-no-dev or anyone who is a member in the platformio team account can upload packages in platformio registry (as owner espressif).

igrr, 2022-02-07:
@Jason2866 Right, but then the resulting package would appear at https://registry.platformio.org/tools/espressif/framework-espidf and not https://registry.platformio.org/tools/platformio/framework-espidf, which I think is what the issue reporter is asking for?

Jason2866, 2022-02-07:
@igrr Since the newer toolchains are already with the ownership `espressif` t would be imho no problem to have all "other" from espressif in platformio registry.
For the platformio maintained framework it is just a change in the  [platform.json](https://github.com/platformio/platform-espressif32/blob/develop/platform.json) of the owner. I think Platformio crew would be grateful if the have less package pack work

igrr, 2022-04-05:
@Jason2866 sorry for the late response. I have confirmed with PlatformIO folks that Espressif can perform updates of tools and libraries in PIO registry. Updates of development frameworks is a different matter which requires a contract with PIO. We are in the process of sorting out the details.

VojtechBartoska, 2024-08-22:
I am closing this ticket as expired. Original discussion was already answered. Thanks for understanding.

closed 2024-08-22 by VojtechBartoska