espressif/esp-idf #8355: softap issue with custom IP

chegewara, 2022-02-06:
Hi,
what i discovered today is that softap is having problems with many basic accessibility when we change default static IP info.
I found that esp32 cant be pinged or accessed (at least when running socket server) when i 

Here is some code that i am using to do it:
```
        esp_netif_ip_info_t* ip_info; // some custom IP, gateway and netmask
	ESP_RETURN_ON_ERROR(esp_netif_dhcps_stop(esp_netif_ap), TAG, "failed to stop DHCP");
	ESP_RETURN_ON_ERROR(esp_netif_set_ip_info(esp_netif_ap, ip_info), TAG, "failed to setup IP");
	ESP_RETURN_ON_ERROR(esp_netif_dhcps_start(esp_netif_ap), TAG, "failed to start DHCP");
```
I was trying to call it before start wifi, after start wifi, after receiving `WIFI_EVENT_AP_START` and in all cases everything seems to works fine, which means the IP is setup properly, client device is getting correct network settings etc, except i cant ping esp32.

Now is what i find eventually to make it works:
- setup custom IP in softap mode,
- important: setup static IP setting on client device, and voila, everything is back to normal, esp32 can be pinged, socket server works, mdns works

board: esp32
esp-idf: master as of today
test app: custom

Thanks

AxelLin, 2022-02-06:
Just a suggestion: Please always use `git describe --tags` output instead of "master as of today".

chegewara, 2022-02-06:
I spent almost 2 hours to find when it works and when it does not. I believe espressif devs are smart enough to find the commit.

AxelLin, 2022-02-06:
> I believe espressif devs are smart enough to find the commit.

This is not true.
When the developers back from lunar new year holidays, they need to guess what the "today" mean and guess which commit you are using. (You could be at different timezone)
And there is no confusion if you use 'git describe --tags' as the issue template suggested.

negativekelvin, 2022-02-06:
>client device is getting correct network settings

>setup static IP setting on client device, and voila, everything is back to normal

These two are in conflict.  If all you do is change client ip settings, then it doesn't make sense.  What is the dhcp debug output from the client?  What is the client (esp32, mobile, pc)?  Wireshark?

chegewara, 2022-02-06:
How those 2 are in conflict?
case 1:
- esp32 set with custom IP settings
- client device with DHCP client (ubuntu linux, android 11)
- IP assigned by esp32 is correct
- i cant even ping esp32

case 2:
- esp32 set with custom IP settings
- client device with manually set IP (ubuntu linux, android 11)
- i can ping esp32, i can use socket server and mdns on esp32

If it makes no sense to you thats fine, because i am noob in the networking and thats why reporting it so lame. All i know i do have 2 (3) cases and the one which we are most interested in is not working.

PS 
on ubuntu this is only message i get after connecting to esp32
`ADDRCONF(NETDEV_CHANGE): wlp6s0: link becomes ready`

negativekelvin, 2022-02-07:
The IP, gateway, and netmask on the client are the exact same in both 1&2?

chegewara, 2022-02-07:
Looks like it was my bug with copy/paste.
Sorry for that.

closed 2022-02-07 by chegewara