espressif/esp-idf #8357: POWERGLITCH_EN and ERR_RST_ENABLE is missing in "X:\XXX\git\components\efuse\esp32s3\esp_efuse_table.csv"

ble0, 2022-02-06:
## Environment

- Development Kit:      ESP32-S3-DEVKITC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32­S3­WROOM­1
- IDF version (run ``git describe --tags`` to find it): v3.1-41-gb082b0e
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System:     Windows 10
- (Windows only) environment type: ESP Command Prompt.
- Using an IDE?: No
- Power Supply: USB

## Problem Description
In` Register 4.16. EFUSE_RD_REPEAT_DATA3_REG (0x003C), EFUSE_POWERGLITCH_EN and EFUSE_ERR_RST_ENABLE ESP32-S3 TRM (Pre-release v0.3)` are mentioned but it does not exist in C:\esp32\git\components\efuse\esp32s3\esp_efuse_table.csv.
1. So I add:
```
        SECURE_VERSION,                   EFUSE_BLK0,  142,   16,     Secure version for anti-rollback
        POWERGLITCH_EN,                   EFUSE_BLK0,  158,    1,     This bit be set to enable power glitch function
        ERR_RST_ENABLE,                   EFUSE_BLK0,  159,    1,     1: use BLOCK0 to check error record registers; 0: disable such check
```
2. Execute `C:\esp32\git\components\efuse>python efuse_table_gen.py esp32s3/esp_efuse_table.csv`
3. I checked the `C:\esp32\git\components\efuse\esp32s3\include\esp_efuse_table.h` and `C:\esp32\git\components\efuse\esp32s3\esp_efuse_table.c` which are already updated.
4. I compiled a sample project with the output below:
```
../main/hello_world_main.c: In function 'app_main':
../main/hello_world_main.c:174:31: error: 'ESP_EFUSE_POWERGLITCH_EN' undeclared (first use in this function); did you mean 'ESP_EFUSE_FLASH_ECC_EN'?
     esp_efuse_read_field_blob(ESP_EFUSE_POWERGLITCH_EN, &chip_ver, 1);
                               ^~~~~~~~~~~~~~~~~~~~~~~~
                               ESP_EFUSE_FLASH_ECC_EN
../main/hello_world_main.c:174:31: note: each undeclared identifier is reported only once for each function it appears in
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

### Expected Behavior
Code should compile

### Actual Behavior
Compile error

### Code to reproduce this issue
```
    uint32_t var;
    esp_efuse_read_field_blob(ESP_EFUSE_POWERGLITCH_EN, &var, 1);
    printf("    ESP_EFUSE_POWERGLITCH_EN: %u\n", var);
```

AxelLin, 2022-02-07:
> ## Environment
> * Development Kit:      ESP32-S3-DEVKITC
> * Kit version (for WroverKit/PicoKit/DevKitC): v1
> * Module or chip used:  ESP32­S3­WROOM­1
> * IDF version (run `git describe --tags` to find it): v3.1-41-gb082b0e

I think v3.1 does not support ESP32S3. The IDF version seems incorrect?

ble0, 2022-02-07:
But I followed the installation guides to compile using command line and did the setup of “esp32s3”. I manage to compile and run all “get-started” projects without any issues.

AxelLin, 2022-02-07:
https://github.com/espressif/esp-idf#esp-idf-release-and-soc-compatibility

ble0, 2022-02-07:
Hi, I downloaded the release/4.4 and able to compile it now. thanks

closed 2022-02-07 by ble0