espressif/esp-idf #8358: branch V4.4 [7c986a6] 2022.2.6, ESP32-S3, I2C, Task watchdog got triggered, when change optimization level to "debug without optimization(-O0)"

gaupen1186, 2022-02-07:
When I change optimization level to "debug without optimization(-O0)", the I2C code ERROR.
when I change it to default "debug -Og", it woks nice.

It may be related to the previous report: [IDFGH-6669) #8301](https://github.com/espressif/esp-idf/issues/8301)

ESP32-S3
IDF vesion branch v4.4,
commit： 7c986a64c76e569474c600aa154728c6fed5293f [7c986a6]
2022年2月6日 9:25:40

the code is here:
[da217_test.c.txt](https://github.com/espressif/esp-idf/files/8012252/da217_test.c.txt)

And the log is here:

I (0) cpu_start: App cpu up.
I (196) cpu_start: Pro cpu start user code
I (196) cpu_start: cpu freq: 160000000
I (197) cpu_start: Application information:
I (197) cpu_start: Project name:     roxs3
I (197) cpu_start: App version:      a3b06fa-dirty
I (197) cpu_start: Compile time:     Feb  7 2022 11:57:59
I (197) cpu_start: ELF file SHA256:  2b37353203406893...
I (198) cpu_start: ESP-IDF:          v4.4-114-g7c986a64c7
I (198) heap_init: Initializing. RAM available for dynamic allocation:
I (198) heap_init: At 3FC9D9B8 len 00042648 (265 KiB): D/IRAM
I (199) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (199) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (199) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (200) spi_flash: detected chip: generic
I (201) spi_flash: flash io: dio
W (201) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (203) sleep: Configure to isolate all GPIO pins in sleep state
I (204) sleep: Enable automatic switching of GPIO sleep configuration
I (205) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (206) gpio: GPIO[11]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:1 
I (206) DA217: [da217_chip_id_check] da217 chip id OK!
I (226) DA217: [da217_reset] OK!
I (226) DA217: [da217_set_range_res_power_bw] OK!
I (226) DA217: [da217_set_odr] OK!
I (226) DA217: [da217_set_fifo] OK!
I (226) DA217: [da217_int_config] OK!
I (226) DA217: [da217_set_active_int] OK!
E (5226) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (5226) task_wdt:  - IDLE (CPU 1)
E (5226) task_wdt: Tasks currently running:
E (5226) task_wdt: CPU 0: IDLE
E (5226) task_wdt: CPU 1: IDLE
E (5226) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x4200DB06:0x3FC9AB800x403771BE:0x3FC9ABD0 0x4201D129:0x3FCF5150 0x4200CED1:0x3FCF5170 0x42002681:0x3FCF5190 0x4037FB99:0x3FCF51C0
0x4200db06: task_wdt_isr at D:/ESP32/esp-idf/components/esp_system/task_wdt.c:183

0x403771be: _xt_lowint1 at D:/ESP32/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x4201d129: cpu_ll_waiti at D:/ESP32/esp-idf/components/hal/esp32s3/include/hal/cpu_ll.h:182

0x4200ced1: esp_pm_impl_waiti at D:/ESP32/esp-idf/components/esp_pm/pm_impl.c:837

0x42002681: esp_vApplicationIdleHook at D:/ESP32/esp-idf/components/esp_system/freertos_hooks.c:63

0x4037fb99: prvIdleTask at D:/ESP32/esp-idf/components/freertos/tasks.c:3974 (discriminator 1)

E (5226) task_wdt: Print CPU 1 backtrace

ginkgm, 2022-02-08:
Hi @gaupen1186 ,

This issue seems to be fixed on master:
da12db29042d80181e58cee8f1d5ff6bc808ff6c

Could you either try it on master, or pick this to v4.4? We will also backport this to v4.4.

gaupen1186, 2022-02-08:
> Hi @gaupen1186 ,
> 
> This issue seems to be fixed on master: [da12db2](https://github.com/espressif/esp-idf/commit/da12db29042d80181e58cee8f1d5ff6bc808ff6c)
> 
> Could you either try it on master, or pick this to v4.4? We will also backport this to v4.4.

gaupen1186, 2022-02-08:
Hi @ginkgm , I just tested it with master branch, and the issue persist.
My platform is S3, not C3. [da12db2](https://github.com/espressif/esp-idf/commit/da12db29042d80181e58cee8f1d5ff6bc808ff6c)  look like for C3 ( Merge branch 'bugfix/c3_i2c_timeout' into 'master' ).

Here is the log:

rst:0x7 (TG0WDT_SYS_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381a72
0x40381a72: spinlock_acquire at D:/ESP32/esp-idf/components/esp_hw_support/include/soc/spinlock.h:122
 (inlined by) xPortEnterCriticalTimeout at D:/ESP32/esp-idf/components/freertos/port/xtensa/port.c:288

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x16e4
load:0x403b6000,len:0xb60
load:0x403ba000,len:0x2f40
entry 0x403b6240
I (24) boot: ESP-IDF v5.0-dev-1478-g6895a662aa-dirty 2nd stage bootloader
I (24) boot: compile time 17:51:22
I (25) boot: chip revision: 0
I (28) boot.esp32s3: Boot SPI Speed : 80MHz
I (33) boot.esp32s3: SPI Mode       : DIO
I (37) boot.esp32s3: SPI Flash Size : 8MB
W (42) boot.esp32s3: PRO CPU has been reset by WDT.
W (48) boot.esp32s3: APP CPU has been reset by WDT.
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot:  3 storage          Unknown data     01 82 00110000 00100000
I (99) boot: End of partition table
...

mythbuster5, 2022-02-17:
https://github.com/espressif/esp-idf/issues/7812#issuecomment-998456730 could please have a try with this patch? This is fixed for C3, but I assume that it was  caused by the same reason

AxelLin, 2022-02-17:
> Hi @gaupen1186 ,
> 
> This issue seems to be fixed on master: [da12db2](https://github.com/espressif/esp-idf/commit/da12db29042d80181e58cee8f1d5ff6bc808ff6c)
> 
> Could you either try it on master, or pick this to v4.4? We will also backport this to v4.4.

@ginkgm @Alvin1Zhang 
Could you confirm if v4.3 branch also needs this fix?

gaupen1186, 2022-02-21:
> [#7812 (comment)](https://github.com/espressif/esp-idf/issues/7812#issuecomment-998456730) could please have a try with this patch? This is fixed for C3, but I assume that it was caused by the same reason

Sorry, I tried patch, but there were many conflicts and errors. I'm not familiar with git patch.
I'm using branch [latest dev 4.4](https://github.com/espressif/esp-idf/tree/release/v4.4)

AxelLin, 2022-04-08:
> [#7812 (comment)](https://github.com/espressif/esp-idf/issues/7812#issuecomment-998456730) could please have a try with this patch? This is fixed for C3, but I assume that it was caused by the same reason

@mythbuster5 
Above patch does not apply to v4.4, see https://github.com/espressif/esp-idf/issues/8358#issuecomment-1046733687.

AxelLin, 2022-04-08:
> Hi @gaupen1186 ,
> 
> This issue seems to be fixed on master: [da12db2](https://github.com/espressif/esp-idf/commit/da12db29042d80181e58cee8f1d5ff6bc808ff6c)
> 
> Could you either try it on master, or pick this to v4.4? We will also backport this to v4.4.

@ginkgm 
Just remind that v4.3 does not include this fix da12db2. (The patch needs slightly update to apply with v4.3)

AxelLin, 2022-04-25:
@o-marshmallow 
Would you mind to check if v4.3 branch also needs backport of da12db2? (assume you are still checking i2c issues)

o-marshmallow, 2022-04-28:
Hi @AxelLin,
I confirm that v4.3 needs this fix, and it will be fixed

AxelLin, 2022-05-23:
> Hi @AxelLin, I confirm that v4.3 needs this fix, and it will be fixed

Still don't find the fix in v4.3 branch. I'm wondering if the backport is missed again.

AxelLin, 2022-06-13:
> Hi @AxelLin, I confirm that v4.3 needs this fix, and it will be fixed

@o-marshmallow 
v4.3.3 does not include the fix.

AxelLin, 2022-12-25:
> @o-marshmallow Would you mind to check if v4.3 branch also needs backport of [da12db2](https://github.com/espressif/esp-idf/commit/da12db29042d80181e58cee8f1d5ff6bc808ff6c)? (assume you are still checking i2c issues)

v4.3 fix: b6bbb5e0a2722ea8e7c7ba035976ff99f4d3214f 19205b0696b17e7a04ba19c61ad738f77006461b

mythbuster5, 2024-11-06:
v4.4 is EOL, this issue will be closed.

closed 2024-11-06 by mythbuster5