espressif/esp-idf #8359: Build fails when using external mbedTLS

josesimoes, 2022-02-07:
## Environment

- IDF version: 4.4
- Build System: CMake

## Problem Description

Since the 4.4 it's extremely hard to use mbedTLS as an external component in a CMake build without convoluted changes in the build system.
When trying to build IDF as an library, and mbedTLS it's also part of the _other_ components that IDF is built along with, there are multiple overlapping or missing declarations/implementations. 
This is because of several changes and tweaks that were added to the sources of mbedTLS included in the IDF distribution. 
I would like to suggest that good practices are followed in IDF and that the components sources aren't modified. If something needs to be replaced, make it on a separate source files, with the changes required and, if needed, remove the "original" source file from the build. 
And always make those changes either private and in such a way that it doesn't completely "hijack" the source. This is because the component maybe used elsewhere in the global project, and not only in IDF.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

igrr, 2022-02-07:
Hi @josesimoes, could you please mention specific issues you are facing, as well as the version of external mbedTLS you are integrating?
We are trying to follow the best practices you suggested. Currently we maintain only [two patches](https://github.com/espressif/mbedtls/commits/mbedtls-2.28.0-idf) on top of mbedTLS upstream version, and with the right sdkconfig settings mbedTLS will still build without these patches. All other changes are done in mbedTLS port layer (in separate files, as you suggest).

josesimoes, 2022-02-07:
@igrr thanks for the prompt reply!
I'm using 2.28.0 which I believe it's the same as the current one in IDF 4.4...

Please take a look at the following build log. (just scroll to the end of it)
https://dev.azure.com/nanoframework/nf-interpreter/_build/results?buildId=28420&view=logs&j=9e65ef19-c40a-5dfd-be4a-e2e2659ab2c7&t=d0e3b688-51a5-5043-3693-28fc1ab5f14c

igrr, 2022-02-07:
From brief look it seems you are adding two different bignum.c files?

https://github.com/nanoframework/nf-interpreter/blob/45fcac755b0018546e0ea05503cf44c7bfda91cb/CMake/Modules/FindmbedTLS.cmake#L122-L124

https://github.com/nanoframework/nf-interpreter/blob/45fcac755b0018546e0ea05503cf44c7bfda91cb/CMake/Modules/FindmbedTLS.cmake#L242-L245

Seems like you need to keep only the latter?

josesimoes, 2022-02-08:
I the meantime I've tried to improve this, by replacing the previous approach (mimic mbedTLS CMake) and adding their CMake directly, so that it would build as a library.

I'm now in a worst position, because now the inclusion clashes with the inclusion of mbedTLS by IDF. There can't be 2 targets with the same name, which is what is happening right now.

As there is no way to control the inclusion of it by IDF, one is stuck with the one being distributed with IDF, without any option to include it as _another_ target when building IDF as a library...

Appreciate your take on this.

igrr, 2022-02-08:
> By looking at the IDF CMake I was under the impression that it was supposed to include both: the one in mbedtls source (library folder) and the 2nd one which is platform specific...

My bad! I have misread the linker error message. You are absolutely right!

Looking at the error again,
```
> ld.exe: esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a(bignum.c.obj): in function `mbedtls_mpi_mul_int':
D:/a/1/s/esp-idf/components/mbedtls/mbedtls/library/bignum.c:1694: multiple definition of `mbedtls_mpi_mul_int'; D:/a/1/s/nf-interpreter/build/targets/ESP32/ESP32\libNF_Network.a(mbedtls_implementations.c.obj):mbedtls_implementations.c:(.text+0x0): first defined here
```
it seems that one implementation of `mbedtls_mpi_mul_int` comes from `esp-idf/components/mbedtls/mbedtls/library/bignum.c` — which looks correct. Another one comes from `mbedtls_implementations.c`, part of NF_Network library. Unfortunately I don't know enough about nanoFramework to understand the relation of NF_Network to mbedTLS. Does NF_Network contain that the "other" copy of mbedTLS you were referring to? If not, any idea why NF_Network would be supplying implementations of mbedTLS functions?

> I the meantime I've tried to improve this, by replacing the previous approach (mimic mbedTLS CMake) and adding their CMake directly, so that it would build as a library.
> As there is no way to control the inclusion of it by IDF, one is stuck with the one being distributed with IDF, without any option to include it as another target when building IDF as a library...

I think when esp-idf/components/mbedtls/CMakeLists.txt calls `add_subdirectory` to include mbedTLS CMakeLists.txt file, mbedTLS library targets get created: mbedtls, mbedcrypto, mbedx509. One thing you could do is, if you need to link to those targets elsewhere, you can skip adding mbedTLS library second time and only call target_link_libraries to create a dependency on mbedTLS targets.

josesimoes, 2022-02-08:
@igrr OK we're getting somewhere. 😅 

A bit of context here: sometime ago, we (.NET nanoFramework) choose to include mbedTLS as our own component, instead of relying on the version distributed with IDF. Main reason was that it was usually quite behind, which is a security flaw from our perspective. With that move, we where able to update mbedTLS as needed, independently of IDF.

Now when working on the migration to IDF 4.4 these issues popped as you guys seem to have changed quite a bit on what concerns mbedTLS inclusion/dependencies.

Following your recomendation above, I'm reworking our build system to depend on mbedTLS library from IDF (which is not ideal for the reasons explained above).

mahavirj, 2022-02-11:
@josesimoes 

> Main reason was that it was usually quite behind, which is a security flaw from our perspective.

For which IDF release branch, you noticed this problem? As far as current status is concerned, we are supporting latest  mbedTLS revisions across our release branches:

|IDF Release Branch| mbedTLS Release| Comments|
|--|--|--|
| master | v2.28.0 | Soon to be upgraded to v3.1.0 |
| release/v4.4 | v2.28.0 | |
| release/v4.3 | v2.28.0 | |
| release/v4.2 and previous | v2.16.12 | |

josesimoes, 2022-02-11:
@mahavirj that has happened in past IDF releases. 
No doubts that all active branches are now "in sync" with mbedTLS version 😁

mahavirj, 2022-02-23:
@josesimoes If there are no more questions here, can we close the issue?

josesimoes, 2022-02-23:
@mahavirj my original problem remains: since 4.4, I can't use mbedTLS as an external dependency. I'm forced to use the one distributed with IDF, which is not ideal. It locks the project to wathever mbedTLS version that you're distributing with IDF, it makes it harder to control the build options for mbedTLS.

For the moment, I've changed our build system to depend on IDF mbedTLS, which is a step back from what we had before 4.4.
I would like to see it, again, been treated as a separate component that one can provide as an external component.

mahavirj, 2022-10-07:
@josesimoes Sorry, I didn't get back on this. We discussed this again, but honestly we do not see an easy to support this request. In-fact, while integrating mbedTLS-3.x, we found that there were many breaking changes and we had to heavily adapt IDF components for that. We would strongly recommend using mbedTLS version that is bundled in IDF.

FWIW, Current status of mbedTLS across IDF release branches can be found [here](https://github.com/espressif/mbedtls/wiki#mbed-tls-support-in-esp-idf).

closed 2022-10-10 by Alvin1Zhang