espressif/esp-idf #8360: Updating ESP-IDF version in Espressif IDF Visual Studio Code extension

domko99, 2022-02-07:
Hi. I would like to ask you are there any instructions how to update my ESP-IDF version in Visual Studio Code extension? I am using 4.3.1 but I would like to upgrade to 4.4 (needed for my project). I was trying everything, downloaded new version and configure Espressif extension in Visual Studio but nothing worked for me. Building project works only with old version. If there is everyone who did this update and could give me some simple advice how to done it I would be grateful. Thanks.

oachkatzalschwoaf, 2022-02-08:
My simple approach: just delete esp-idf and .espressif folder then restart VSC and open any esp-project. => espressif-plugin will start initial install wizard.

closed 2022-02-08 by domko99

brianignacio5, 2022-02-09:
What is the issue you are having @domko99 ? Could you provide more detail ?

domko99, 2022-02-09:
After the instalation of ESP-IDF 4.4 I have tried to make a new project but I was stucked with message restart this dialog window. So I did, restaer dialog and also VSC but still the same. I have seen here some similar issue with guidence to uninstall Espressif extension and install it from .vsix  file. I did and extension is working well. I have also tried to uninstall extension and install it from VSC extension but it didnt solve the problem. I think .vsix solve the problem or the problem was elsewhere.