espressif/esp-idf #8361: ability to route lwip debugs through ESP_LOGx macros

someburner, 2022-02-08:
When enabling LWIP debugs with `idf.py menuconifg`, these appear to (mostly?) get routed to `printf`. On the other hand pretty much everything else in the IDF is logged through ESP_LOGx macros. I would like to have a unified manner of logging where everything goes through the same macros, in order to support redirecting logs over UDP or elsewhere.

Doing something like this in `components/lwip/port/esp32/include/arch/cc.h` works, but could probably be made better with some macro magic to expand `x` and pass to the ESP_LOGx macros.

```c
#include "esp_log.h"

static const char * _LWIP_TAG = "lwip";
static int _ind = 0;
static char _pbuf[256];
static void _lwip_dbg_fn(const char * fmt, ...)
{
    va_list arglist;
    va_start(arglist, fmt);
    _ind += vsnprintf(_pbuf+_ind, sizeof(_pbuf) - _ind, fmt, arglist);
    va_end(arglist);
    if(_ind > 0) {
        if(_pbuf[_ind-1] == '\n') {
            // printf("newline\n");
            _pbuf[_ind - 1] = 0;
            ESP_LOGI(_LWIP_TAG, "%s", _pbuf);
            _ind = 0;
        }
    }
}

// #define LWIP_PLATFORM_DIAG(x)   do {printf x;} while(0)
#define LWIP_PLATFORM_DIAG(x)   do {_lwip_dbg_fn x;} while(0)
```

I'm sure there are some pitfalls in doing this, particularly if routing logs over UDP, but I think such a feature would be easy enough to make optional and would result in consistent log output that could be `grep`d or redirected more easily. Thoughts?

negativekelvin, 2022-02-10:
>some macro magic

something like ...

```
#define LOG_LOCAL_LEVEL ESP_LOG_DEBUG
#include "esp_log.h"

#define LOG_FUNC(format, ...) ESP_LOGD("lwip", format, ##__VA_ARGS__ ) 

#define LWIP_PLATFORM_DIAG(x) LOG_FUNC x
```

xueyunfei998, 2022-02-11:
hi @negativekelvin @someburner 

This is a very good suggestion, and this change will be added to idf later.

xueyunfei998, 2022-03-15:
hi @someburner 
You can submit a PR, thanks

someburner, 2022-03-15:
@negativekelvin

The only issue with doing it this way:

```
#define LOG_FUNC(format, ...) ESP_LOG_LEVEL(ESP_LOG_DEBUG, "lwip", format, ##__VA_ARGS__ )
#define LWIP_PLATFORM_DIAG(x) LOG_FUNC x
```

is that all LWIP debugs have a `\n` at the end. So the output doesn't look as clean (as ESP_LOG appends its own `\n`).

@xueyunfei998

Do you think something like my original solution would be accepted? The only alternative I can see would involve adding new macros to `esp_log.h` that define a separate macro that does not append its own `\n` at the end.

xueyunfei998, 2022-03-29:
hi @negativekelvin 

I think it's better to use the macro method, but there will be one more /n later.

During the test, I found that too much /n didn't seem to make any difference.

someburner, 2022-04-17:
@xueyunfei998 

https://github.com/espressif/esp-idf/pull/8785

closed 2022-05-15 by espressif-bot