espressif/esp-idf #8362: 1 level deep repo clone causes idf_tools to go bonkers

KaeLL, 2022-02-08:
- IDF version: master | 6895a662aa 

## Problem Description

Title. `describe`ing a shallow repo expectedly fails, and `idf.py`-related tools used to occasionally mention the error, when installing and exporting IDF. Recently, it started to nag really hard about it with the message
```
fatal: No names found, cannot describe anything.
WARNING: Git describe was unsuccessful: b''
```
even when invoking autocomplete of `idf.py` commands, which is really annoying.

### Expected Behavior

The previous one, without the nagging. Bring it back.

### Actual Behavior

```
$ idf.py <double TAB> fatal: No names found, cannot describe anything.
WARNING: Git describe was unsuccessful: b''
fatal: No names found, cannot describe anything.
WARNING: Git describe was unsuccessful: b''
```

### Steps to reproduce

Clone the repo with `--depth=1`, install and export should make the warning show up by about 4 times, invoke idf.py commands or idf.py and autocomplete, and have fun.

niraami, 2022-02-09:
+1 This also happens in our internal fork

closed 2022-02-23 by espressif-bot

dobairoland, 2022-02-23:
@KaeLL Thank you for reporting this. It should be fixed now.