espressif/esp-idf #8364: esp-2021r2-patch2-8.4.0 not working if user folder has a space in it

apiesse, 2022-02-09:
## Environment
- Development Kit:  none
- Module or chip used: ESP32
- IDF version: v4.4
- Build System: CMake
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System: Windows
- (Windows only) environment type: idf.py
- Using an IDE?: Yes, Eclipse 2021-06 with esp-idf plugin

## Problem Description
I've just updated a project from esp-idf-v4.3.2 to esp-idf-v4.4 and I have the folloqing error message:
```
xtensa-esp32-elf-gcc.exe: error: C:\Users\Marco: No such file or directory
xtensa-esp32-elf-gcc.exe: error: Mariotti\AppData\Roaming/.ccache/tmp/cmd_args.bHD3G8: No such file or directory
xtensa-esp32-elf-gcc.exe: fatal error: no input files
compilation terminated.
```
As you can guess, the user directory is "C:\Users\Marco Mariotti\", but somhow the space is no longer escaped.
The error was not present in esp-idf-v4.3.2 (using esp-2021r2-8.4.0 toolchain).

igrr, 2022-02-09:
Hi @apiesse, thanks for reporting this. It's interesting to hear that having a directory with space in its path worked for you in older versions of ESP-IDF. This is explicitly not supported and there were several places in the build system where spaces were not handled correctly.
https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/get-started/index.html#step-1-install-prerequisites

We have fixed these issues in `master` recently and the future v5.0 release will support spaces in paths.

Could you please check the following:
* when you used v4.3.2, did ESP-IDF directory have path with spaces?
* did your project have path with spaces?

apiesse, 2022-02-09:
> 

Thanks for the reply @igrr .
That surprised me as well give the IDE Tools are installed in C:\espressif and the project is located in C:\svn\project\firmware\trunk\app (same with esp-idf-v4.3.2) as the documentation suggested.

igrr, 2022-02-09:
I see, looks like it might be a ccache issue. Are you building using `idf.py build` from command line or from an IDE?
If it's the former, could you do `set IDF_CCACHE_ENABLE=0` in the command prompt, then `idf.py fullclean` and finally `idf.py build` again.

apiesse, 2022-02-09:
This is working both from command line and Eclipse!
Thanks you, hope to see this soon in some relaase before 5.0!

igrr, 2022-02-09:
Thanks for confirming that disabling the cache helps.

I'm afraid we can't add support for spaces in paths to existing v4.x release because it introduces certain breaking changes to project CMakeLists.txt files. So by fixing spaces in paths, we unfortunately also have a chance of breaking some projects. Because of that, we decided to include this change only in v5.0.

However the issue with ccache seems to be unrelated, and I think it can be fixed by upgrading ccache to a more recent version. We'll try to reproduce the issue and upgrade ccache version in the existing v4.x releases.

In the meantime, you can re-enable ccache (`set IDF_CCACHE_ENABLE=1`) and then configure ccache to use different cache directory — instead of somewhere in your home directory, point it e.g. to C:\espressif\ccache. This can be done using `CCACHE_DIR` environment variable. More on this in [ccache manual](https://ccache.dev/manual/4.3.html). This way you will keep having faster builds, and will have everything working until we manage to upgrade ccache.

acebearxp, 2022-02-09:
I think I meet a  similar situation.
When I execute `idf.py gdbgui`, there is a red error line shown as below
[image]

obviously the **\t** in `.espressif\tools\xtensa...` was represented as a "TAB"

The red line scared me, but the gdbgui still works fine, I think  it is just wrong in the error message, perhaps the internal function is not wrong

xiaguangbo, 2022-04-16:
@igrr I also encountered a similar problem, I put 'esp-idf' and '.espressif' in 'C:\Program Files...', and then 'kconfiglib.KconfigError: D: /.../project/esp/project/template-app/build/kconfigs_projbuild.in:1: 'C:/Program' not found (in 'source "C:/Program"').', I use esp-idf v4.4

binary1230, 2022-08-27:
I kept running into this 

(damn you windows for using a space in my home directory name and making it really hard to change, why...) 

Here's a couple workarounds from my travels:

1. probable best thing (new)
The latest ccache release fixes this issue, I have submitted a PR for upgrading this in ESP-IDF

```
Ccache 4.6.2- Release date: 2022-08-22
Corrected handling of space in paths when using response file on Windows. [contributed by Sergey Semushin]
```
The upstream ccache fix for this is probably this PR: https://github.com/ccache/ccache/pull/1124

2. Workaround: Set the CCACHE_DIR somewhere that doesn't have spaces in the filename, using this powershell command (or equivalent in cmd prompt).  In here, $env:TMP uses 8.3 filename format to properly escape the space, so, this happens to work:
```powershell
$env:CCACHE_DIR=$env:TMP; idf.py build
```

3. Sledgehammer: 
```powershell
$env:IDF_CCACHE_ENABLE=0; idf.py build
```

Hope that helps.

Bonus: It might be cool to get a hint from idf.py that the user's home directory contains a space in the filename, and warn about potential issues with that. It was hard to track down that this was a ccache / path spaces bug.  In the future, other stuff could break too (spaces in home directory names is evil), so this breadcrumb could help devs reach the solution quicker.

Thanks!
-Dom

dobairoland, 2022-08-29:
Thanks @binary1230 for sharing your findings and creating a PR for this. Space support for ccache is really good news. We handle your PR soon.

closed 2022-09-02 by espressif-bot