espressif/esp-idf #8365: Support for 32MB version of ESP32S3 Module is missing

ganhailin, 2022-02-09:
## Environment

- Development Kit:     ESP32-S3-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): V1
- Module or chip used:  ESP32-S3-WROOM-2 (N32R8V)
- IDF version (run ``git describe --tags`` to find it):
    v5.0-dev-1429
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32s3-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Linux
- Using an IDE?: Yes, Clion with IDF_TARGET=esp32s3 set and export.sh as env script
- Power Supply:         USB

## Problem Description

1.options in menuconfig does not contains 32Mbyte
2.esptool is unable to flash upper 16Mbit of flash (either with --no-stub or not),
&ensp;    erasing or writing above 0x1000000 with --no-stub will warp to 0x00000000 then bootloader will overwride
&ensp;    writing with stub will resault a error ,showing  "opi_flash.c 363" at uart port.
&ensp;    try to patch stub with ER4B cmd in the flash datasheet 
&ensp;[https://www.macronix.com/Lists/Datasheet/Attachments/7693/MX25UM25645G,%201.8V,%20256Mb,%20v1.1.pdf](url)
&ensp;and set addr_len to 32bit will do the same thing as --no-stub,
&ensp; it seems like the high 8bit addr is never sent out or sent zero
3.partition size can't be bigger than 16M
4.bootloader is not correctly parsing 32M flash or switch flash to opi_dtr mode
5.spi-flash api works fine, including erasing writing,even mount fatfs ,I don't under stand why stub can't behave as normal.
6.mmap to Data MMU can't be more than 16M
&ensp;    Cache_Get_DROM_MMU_End() always returned 0x00000400,so map can't be more than 0x00000400/4×64K=16M,
&ensp;    hardcoded Cache_Get_DROM_MMU_End() to Cache_Get_DROM_MMU_End()+128*4 works for me(not 256 to save last 8M for PSRAM)
## Expected Behavior
    all above works as it should be
### Actual Behavior
    see Problem Description
### Steps to reproduce

1. just run menuconfig
2. esptool.py  -c esp32s3 -p /dev/ttyUSB0 -b 2000000 --before=default_reset  --after=hard_reset write_flash --no-compress --flash_mode dout --flash_freq 80m --flash_size 32MB 0x1000000 something.bin 
   &ensp;then you will find header be overrided
3. add an partittion larger than 0x1000000 in partittion.csv and build
4. just boot up and see the uart output shows you have a 2MB Flash 
5. with parttion in latter 16M of flash and format &mount as fatfs ,or just perform a partition read/write/erase,all work as it should be
6. try to map an 16Mbyte partition and will return NO_MEM because get_mmu_region never gave you more than 256 mmu entry

## Debug Logs
I already patch a lot at my idf clone ...so it looks likes below
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0108,len:0x21b8
load:0x403b6000,len:0xd50
load:0x403ba000,len:0x3498
entry 0x403b6298
I (36) boot: ESP-IDF v5.0-dev-1429-ga10d9abba0-dirty 2nd stage bootloader
I (36) boot: compile time 05:01:58
D (36) bootloader_flash: non-XMC chip detected by SFDP Read (87), skip.
D (43) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (49) boot: chip revision: 0
I (53) boot.esp32s3: magic e9
I (57) boot.esp32s3: segments 03
I (61) boot.esp32s3: spi_mode 03
I (65) boot.esp32s3: spi_speed 0f
I (69) boot.esp32s3: spi_size 05
I (72) boot.esp32s3: Boot SPI Speed : 80MHz
I (77) boot.esp32s3: SPI Mode       : SLOW READ
[... 196 lines cut ...]
    uint32_t Cache_Get_DROM_MMU_End(void);
    printf("Cache_Get_DROM_MMU_End() returned:0x%08x\n",Cache_Get_DROM_MMU_End());
    esp_err_t ret2 = esp_partition_mmap(partition, 0, partition->size, SPI_FLASH_MMAP_DATA, &map_ptr, &map_handle);
    // Map the partition to data memory
    ESP_LOGE(TAG,"mmap return :%d",ret2);
    ESP_ERROR_CHECK(ret2);
    ESP_LOGI(TAG, "Mapped partition to data memory address %p", map_ptr);

    // Read back the written verification data using the mapped memory pointer
//    char read_data[sizeof(store_data)];
//    memcpy(read_data, map_ptr, sizeof(read_data));
//    ESP_LOGI(TAG, "Read sample data from partition using mapped memory: %s", (char*) read_data);

//    assert(strcmp(store_data, read_data) == 0);
    int countint_1=0;
    for(int i=0;i<16*1024*1024;i+=4){
        if((i%1024)==0)printf("reading at %p\r",(void*)i);
        int bytetowrite=(countint_1)*(countint_1)^0x5aa5a55a;
        countint_1++;
        if(((int*)map_ptr)[i/4]!=bytetowrite)
            break;
    }
    printf("\ndone\n");
```
## Other items if possible

ginkgm, 2022-02-09:
Hi @ganhailin ,
I think the module you got is still in NPI status, we are still completing the support for those modules.
Sorry for inconvenience.

igrr, 2022-02-21:
Issue 1 (lack of menuconfig options) has now been resolved by merging https://github.com/espressif/esp-idf/pull/7688.

TGiles1998, 2022-04-05:
Hi, Unfortunately I can't seem to flash the 32MB either. Im using the lastest esp-idf-v4.2.3. It just resorts back to 2MB.

 (0) cpu_start: ApESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0108,len:0x1648
load:0x403b6000,len:0xb7c
load:0x403ba000,len:0x2f58
entry 0x403b6248
I (33) boot: ESP-IDF v4.4 2nd stage bootloader
I (33) boot: compile time 21:04:11
I (33) boot: chip revision: 0
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : SLOW READ
I (44) boot.esp32s3: SPI Flash Size : 2MB
I (49) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (88) boot: End of partition table
I (92) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=07c40h ( 31808) map
I (108) esp_image: segment 1: paddr=00017c68 vaddr=3fc90dc0 size=02940h ( 10560) load
I (112) esp_image: segment 2: paddr=0001a5b0 vaddr=40374000 size=05a68h ( 23144) load
I (124) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=17190h ( 94608) map
I (148) esp_image: segment 4: paddr=000371b8 vaddr=40379a68 size=07350h ( 29520) load
I (157) esp_image: segment 5: paddr=0003e510 vaddr=50000000 size=00010h (    16) load
I (162) boot: Loaded app from partition at offset 0x10000
I (162) boot: Disabling RNG early entropy source...

igrr, 2022-04-05:
@TGiles1998  I'm sorry, there must be some mistake. You say you are using esp-idf v4.2.3, but this SDK version has no support for esp32-s3 chip at all. Are you sure that is the version you are using? (The bootloader log says `ESP-IDF v4.4 2nd stage bootloader`)

Assuming you are using IDF v4.4, I think the issue is that it doesn't have support for flash sizes >16MB. This support has been added in master branch, we will backport it to release/v4.4 after the upcoming bugfix release (v4.4.1). So it will be released in v4.4.2.

TGiles1998, 2022-04-05:
Apologies I took the latest release v4.2.3 from the release section posted 25 days ago. I'll try the newer one thanks!

igrr, 2022-04-05:
No problem! Please note that Github simply shows releases in the chronological order. As mentioned in the release notes of v4.2.3 (2nd paragraph):

> ESP-IDF v4.2.3 is a bugfix update for ESP-IDF v4.2. At the time of writing, ESP-IDF v4.4 is the latest stable release.

TGiles1998, 2022-05-22:
Hi @igrr @Alvin1Zhang ,

Sorry didn't realise this was still open, I've been able to access 32MB of RAM but have yet to find a solution to access 32MB of PSRAM?

Any help would be amazing thank you!

Thomas

thyclouds, 2022-06-07:
Hi @igrr!

We have followed work by @ESP32DE very closely and were able to get the Winbond chip recognized.

[image]

However trying to flash and cross the 24-bit boundry, above the 16MB. We then tried to create separate partitions above 16MB and place files there and read:

[image]

But then it failed to read it:
[image]

Could youpls advice on how to access flash above the 16MB?

AndrewCapon, 2023-05-07:
I know it's nearly been a year but is there a way of working with larger flash sizes now?

igrr, 2023-05-07:
@AndrewCapon yes, there are a few flash models > 16MB which are supported, listed in the documentation.

If you are running into a problem, I recommend opening a new issue and providing information such as IDF version, flash model, as well as the program you are trying to run.

AndrewCapon, 2023-05-08:
Hi @igrr 

Thanks for the info, I will double check here.

Andy

nichonaugle, 2023-12-10:
@igrr 
Here is the link: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_flash/spi_flash_optional_feature.html

higaski, 2024-03-17:
What's the current state of >16MB flash support?

As far as I know they ROM bootloader still does not ([and probably never will?](https://github.com/espressif/esptool/issues/859#issuecomment-1469623705)) support >16MB flash. For that very reason the esptool --no-stub option does not work above 16MB either. This has the unpleasant side effect that the [esp-serial-flasher](https://github.com/espressif/esp-serial-flasher/issues/92) library does not support >16MB at the moment.

Sorry to be so blunt, but that's pretty sad considering you've been selling those devices for like what... 3 years now?

kylefmohr, 2024-03-18:
> Sorry to be so blunt, but that's pretty sad considering you've been selling those devices for like what... 3 years now?

@higaski This is also my understanding of the situation. The 32MB version is [still being sold in Espressif's official Amazon store](https://www.amazon.com/Espressif-ESP32-S3-DevKitC-1-N32R8V-Development-Board/dp/B09R4GSDJM). In their [Official AliExpress store, the 32MB still shows up, but appears to be out of stock](https://www.aliexpress.us/item/3256803793464226.html). Kinda seems like false advertising!

igrr, 2024-03-18:
Hi folks,
I think all the problems describes in the original issue report at the top have been fixed for a while now. There is also a list of supported flash chips linked in one of the comments above.

@higaski As for the support for larger flash sizes in the esp-serial-flasher, this is something that could be implemented by adding flasher stub support in that project.

@kylefmohr If you need to order a module with 32MB flash but can't find it in stock, please contact Espressif sales team (via the form on the website) and they will try to help you.

higaski, 2024-03-18:
> @higaski As for the support for larger flash sizes in the esp-serial-flasher, this is something that could be implemented by adding flasher stub support in that project.
> 

Then please either appoint someone to that task or offer me some assistance in implementing it.

igrr, 2024-03-18:
@higaski Okay, I'll reply in that esp-serial-flasher issue.

higaski, 2024-03-23:
Before I open another issue... Any chance that there is a known bug that app partitions can't be placed above 16MB?

Flashing the following partition table for example
```
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
phy_init, data, phy,     ,        4K,
storage,  data, spiffs,  ,        16320K,
nvs,      data, nvs,     ,        1M,
factory,  app,  factory, ,        4M,
```
results in "image has invalid magic byte" errors.

Swapping the order
```
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
phy_init, data, phy,     ,        4K,
factory,  app,  factory, ,        4M,
storage,  data, spiffs,  ,        16320K,
nvs,      data, nvs,     ,        1M,
```
there is no issue.

ZenanLin1999, 2024-04-06:
I have the same problem when trying to upload an APP over 16MB to a ESP32-S3-N32R8 board. It's really strange to have an issue reboot over and over again. So, I googled and found this open issue. Really disappointed. Hopefully, this issue can be closed soon!

mythbuster5, 2025-01-20:
We don'y have official 32MB flash module, but over 16MB module is supported on idf, I will close this issue. If you have more information to update, please reopen it. Thanks!

closed 2025-01-20 by mythbuster5

igrr, 2025-01-20:
ESP32-S3-WROOM-2-N32R8V is marked as EoL, but there is indeed ESP32-S3-WROOM-2-N32R16V which is in production.

Icarus113, 2025-07-09:
> Just for reference for anyone reading this. This restriction is documented here:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_flash/spi_flash_optional_feature.html#restrictions

FYI this doc has been updated, programming guide will be synced soon.

https://github.com/espressif/esp-idf/blob/master/docs/en/api-reference/peripherals/spi_flash/spi_flash_optional_feature.rst#restrictions