espressif/esp-idf #8368: Dropped A2DP Packets when sending stream to I2S interface for SPDIF (BT_APPL: Pkt dropped)

joba-1, 2022-02-09:
## Environment

- Development Kit:      [none] (ESP32 MH ET Live MiniKit)
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-1509-g5893797bf0
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

Scenario: using I2S to send a bluetooth A2DP stream (clean sine wave) as SPDIF to pin 27
Problem: The stream works for ~1,5 minutes, then starts to have a glitch every ~4-5s. 

The problem is not the added SPDIF encoding, the data coming from the esp callback is already wrong.

### Expected Behavior
The sine wave sent (or any other sound) gets transferred without continuity problems without time limit.

### Actual Behavior
The sine wave is perfect for ~1,5 minutes, then there are short intervals of ~30 samples not matching the sine wave.
After that the sine wave is good for ~4-5s, but not necessarily in phase to before the glitch.
Then the problem repeats until reset of the chip.

### Steps to reproduce

* Install latest ESP32 IDF (or any older - I also tried 4.4 and 3.5.5)
* get code with git clone git://github.com/amedes/esp_a2dp_sink_spdif.git
* run idf.py menuconfig to set output pin to 27 (optional if seeing the warning messages is enough)
* run idf.py flash monitor 
* connect phone via bluetooth to the ESP32 and send music (or better use a sine wave to see the glitch in the wave form)
* let it play for 1,5-2 min and start hearing soft clicks about every 4-5s
* watch the monitor to see "W (102105) BT_APPL: Pkt dropped" for each click.
* If you connect a red led to pin 27 and send the light into an optical SPDIF-In port of a soundcard you can record the wave directly with audacity.

// If possible, attach a picture of your setup/wiring here.

[image]

[image]

### Code to reproduce this issue
Full project (not mine): https://github.com/amedes/esp_a2dp_sink_spdif

The whole code is basically the esp_a2dp_sink example with added encoding of the spdif format and other I2S config for the increased bit frequency (to cover the SPDIF overhead). 

This is the function that initializes the I2S interface:

```cpp

// Some relevant defines for below function

#define I2S_NUM			(0)

#define I2S_BITS_PER_SAMPLE	(32)
#define I2S_CHANNELS		2
#define BMC_BITS_PER_SAMPLE	64
#define BMC_BITS_FACTOR		(BMC_BITS_PER_SAMPLE / I2S_BITS_PER_SAMPLE)
#define SPDIF_BLOCK_SAMPLES	192
#define SPDIF_BUF_DIV		2	// double buffering
#define DMA_BUF_COUNT		2
#define DMA_BUF_LEN		(SPDIF_BLOCK_SAMPLES * BMC_BITS_PER_SAMPLE / I2S_BITS_PER_SAMPLE / SPDIF_BUF_DIV)
#define I2S_BUG_MAGIC		(26 * 1000 * 1000)	// magic number for avoiding I2S bug
#define SPDIF_BLOCK_SIZE	(SPDIF_BLOCK_SAMPLES * (BMC_BITS_PER_SAMPLE/8) * I2S_CHANNELS)
#define SPDIF_BUF_SIZE		(SPDIF_BLOCK_SIZE / SPDIF_BUF_DIV)
#define SPDIF_BUF_ARRAY_SIZE	(SPDIF_BUF_SIZE / sizeof(uint32_t))

...
[... 244 lines cut ...]
I (113485) BT_AV: Audio packet count 5600
I (114965) BT_AV: Audio packet count 5700
I (116445) BT_AV: Audio packet count 5800
I (117855) BT_AV: Audio packet count 5900
I (117855) RCCT: AVRC event notification: 5
I (117855) BT_AV: Play position changed: 20203-ms
W (118135) BT_APPL: Pkt dropped

I (119355) BT_AV: Audio packet count 6000
W (120735) BT_APPL: Pkt dropped

I (120755) BT_AV: Audio packet count 6100
...
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8035644/sdkconfig.txt)

- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
[esp_a2dp_sink_spdif.zip](https://github.com/espressif/esp-idf/files/8035652/esp_a2dp_sink_spdif.zip)

- [ ] coredump (This provides stacks of tasks.)
none

joba-1, 2022-02-09:
If I send slightly less data than the callback delivers, the drops are gone.

BetterJincheng, 2022-03-02:
@joba-1 I played music without any pkt dropping with your code and config. Any suggestions? Thanks!

BetterJincheng, 2022-03-02:
@joba-1 And I did not understand how your 'rate control' works...

BetterJincheng, 2022-03-02:
@joba-1 
```
size_t write_ringbuf(const uint8_t *data, size_t size)
{
    // rate control
    UBaseType_t items;
    vRingbufferGetInfo(s_ringbuf_i2s, NULL, NULL, NULL, NULL, &items);
    if (items < RINGBUF_SIZE * 3 / 8) {
        xRingbufferSend(s_ringbuf_i2s, (void *)data, AUDIO_SAMPLE_SIZE, (portTickType)portMAX_DELAY);
    } else if (items > RINGBUF_SIZE * 5 / 8) {
        size -= AUDIO_SAMPLE_SIZE;
    }

    BaseType_t done = xRingbufferSend(s_ringbuf_i2s, (void *)data, size, (portTickType)portMAX_DELAY);
    if(done){
        return size;
    } else {
        return 0;
    }
}
```

joba-1, 2022-03-02:
Ah, I see now what you mean. I wasn't aware of this code.

This rate limit code does the following:

* If the ring buffer is rather empty, then send the first audio sample of the callback twice
* If the ring buffer is rather full, then discard the last audio sample of the callback

It assumes, the rate bt sends data and i2s consumes data is out of sync for not more than one sample per callback.
I'll add some logs and try to drop/add more samples if the buffer is even closer to full/empty and hope this is still not audible.

Looks promising, thanks for the hint.

joba-1, 2022-03-02:
> @joba-1 I played music without any pkt dropping with your code and config. Any suggestions? Thanks!

I can only assume that your bt sender is a tiny bit slower than mine or my esp is running a bit slower than yours

joba-1, 2022-03-03:
dropped even more frames to a maximum of 100 out of 2560 bytes, then drops are gone.

BetterJincheng, 2022-03-17:
@joba-1 "Pkt dropped" happens when sink_rx_queue is about full. The root cause is that the Bluetooth Stack is busy to handle 
the incoming audio data. 

Now some information for your reference.

The task priority of `bt_app_a2d_data_cb` is 19, the priority of task handling sink_rx_queue is 20, and, your  I2S task priority is 22 .

A possible case is that, your I2S is always busy, `bt_app_a2d_data_cb` has fewer chance to occupy the CPU. So packet dropped can happened.

Lower down the priority of I2S task to 18, and try again.

Turn to us anytime.
Thanks

joba-1, 2022-03-17:
thanks for the update. Will try the prio change.

AxelLin, 2022-03-17:
> Lower down the priority of I2S task to 18, and try again.
> 
> Turn to us anytime. Thanks

@BetterJincheng 
FYI, there is a bug regarding task priority if using esp32 dual-core.
https://github.com/espressif/esp-idf/issues/7580

BetterJincheng, 2022-03-17:
@joba-1 Sometimes, there are misunderstandings about the use of `data_cb`. The really reasonable thing to do is， copy the data and let the `data_cb` returns, handle the data in another task.

joba-1, 2022-03-17:
I can use any guidance, thanks for that.

Currently I (or better: amedes) do it like in the original espressif a2dp to i2s example: the cb puts data into a circular buffer and another task fetches it from there.
This other task converts the samples to spdif format (which doubles the sample size) and writes that out with double speed i2s to a data pin (that drives the toslink led).

My main problem seems to be not priority, but that a2dp bit rate (which is controlled solely by the external bt device) and the spdif i2s bit rate need to be precisely synchronous but are not. 
The current code adds or drops at most 1 sample per cb (~1/1000th), but I require to drop much more (>10 samples per cb). That way the circular buffer fills up and then starts blocking the cb, no matter what prio it has.
Of course I cannot statically drop e.g. 20 frames each cb. If the number is too high I would drain the buffer sooner or later and get ugly sound artifacts as well.

Is it possible to reduce the packet size I get from the bt cb? Else I try to split it up in an intermediate level.

joba-1, 2022-03-18:
I tried increasing (and decreasing) the task prios. Did not help.
Then I rewrote the [rate limiting code](https://github.com/joba-1/esp_a2dp_sink_spdif/blob/c7d33f29a52e7da59f4baee206bcc0fad7681f7e/main/bt_app_core.c#L192) to use a floating average buffer usage instead of the current usage (fluctuated too much) and allowed dropping up to 3 samples. That did it. 

Thanks for your support, your hint on the rate limiting code did it!

closed 2022-03-18 by joba-1