espressif/esp-idf #8369: UART not firing timeout interrupt

philtao-dev, 2022-02-09:
I use the ESP32-S2-WROVER module on a custom PCB.  Development using Platformio / ESP-IDF framework.

I use UART0 for Modbus communication - events do not get triggered after receiving Modbus a command (in my case usually read command with 8 bytes).
An event is only triggered when when the FIFO RX buffer has reached the "full threshold" (120 bytes).  
My understanding is that after receiving the Modbus command, there should be RX timeout event triggered, but that does not happen.

I made further tests using the **uart-echo_example**, where I added **uart_intr_config()** and **uart_enable_rx_intr()** with **.rxfifo_full_thresh = 20** and **.rx_timeout_thresh = 2**.

The RX FIFO buffer full threshold now creates the interrupt after receiving 20 bytes, but still no RX timeout interrupt.  Am I doing something wrong?
```
#define UART_NUM        UART_NUM_0
#define ECHO_TEST_TXD  (GPIO_NUM_43)
#define ECHO_TEST_RXD  (GPIO_NUM_44)
#define ECHO_TEST_RTS  (UART_PIN_NO_CHANGE)
#define ECHO_TEST_CTS  (UART_PIN_NO_CHANGE)
#define BUF_SIZE (1024)

uart_config_t uart_config = {
    .baud_rate = 19200,
    .data_bits = UART_DATA_8_BITS,
    .parity    = UART_PARITY_DISABLE,
    .stop_bits = UART_STOP_BITS_1,
    .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
    .source_clk = UART_SCLK_APB,
};

void app_main(void)
{
  vTaskDelay(5000 / portTICK_PERIOD_MS);
  ESP_ERROR_CHECK(uart_driver_install(UART_NUM, BUF_SIZE * 2, 0, 0, NULL, 0));
  ESP_ERROR_CHECK(uart_param_config(UART_NUM, &uart_config));
  ESP_ERROR_CHECK(uart_set_pin(UART_NUM, ECHO_TEST_TXD, ECHO_TEST_RXD, ECHO_TEST_RTS, ECHO_TEST_CTS));

  uart_intr_config_t uart_intr = {
      .intr_enable_mask = UART_INTR_RXFIFO_TOUT | UART_INTR_RXFIFO_FULL,
      .rxfifo_full_thresh = 20,
      .rx_timeout_thresh = 2,
  };
  ESP_ERROR_CHECK(uart_intr_config(UART_NUM, &uart_intr));
  ESP_ERROR_CHECK(uart_enable_rx_intr(UART_NUM));

  uint8_t *data = (uint8_t *) malloc(BUF_SIZE);

  while (1) {
    int length = 0;
    ESP_ERROR_CHECK(uart_get_buffered_data_len(UART_NUM, (size_t*)&length));
    if (length) {
      ESP_LOGI(TAG, "received %d bytes", length);
      length = uart_read_bytes(UART_NUM, data, length, 20 / portTICK_RATE_MS);      
    }
    vTaskDelay(20 / portTICK_PERIOD_MS);
  }
}
```
I send a number of bytes to the UART, but the only output I get is when the RX buffer is full:
```
I (53420) UART test: received 20 bytes
I (61380) UART test: received 20 bytes
```

philtao-dev, 2022-02-12:
I would like to report on the UART register state, but I do not know how to do it.
Can someone tell me how to read the **UART_RX_TOUT_EN** bit (31) in the **UART_CONF1_REG** register (0x0024)?

philtao-dev, 2022-02-13:
I did a bit more investigation by printing the content of the UART registers before and after configuration of the interrupt (increased .rx_timeout_thresh to 20):
```
I (5100) UART test: UART_CONF1_REG(0) value: 0x8000C078
I (5100) UART test: UART_MEM_CONF_REG(0) value: 0x640012
I (5100) UART test: After interrupt config
I (5100) UART test: UART_CONF1_REG(0) value: 0x8000C014
I (5100) UART test: UART_MEM_CONF_REG(0) value: 0xC80012
```
I can see that UART_RXFIFO_FULL_THRHD is reduced form 120 to 20.
That confirms the value  **UART_RX_TOUT_EN = 1**
After setting **.rx_timeout_thresh = 20**, the value of **UART_RX_TOUT_THRSHD = 200** (_why a factor 10?_)

The ESP32-S2 technical reference says:

> UART_RX_TOUT_THRHD This register is used to configure the threshold time that receiver takes to receive one byte. The UART_RXFIFO_TOUT_INT interrupt will be triggered when the receiver takes more time to receive one byte with UART RX_TOUT_EN set to 1. (R/W)

> UART_RX_TOUT_EN This is the enable bit for UART receiver’s timeout function. (R/W)

So it seems that the required conditions I could find are met for the RX timeout function to be enabled.  Anything else I missed?
I have seen other threads with similar issue, but no explanation of why and how to fix it.
Is that me doing something wrong or is it a bug?  Please help, so I can move on!

philtao-dev, 2022-02-14:
Not even an acknowledgement that anyone at least read and understood what I exposed!
If I did not expose my problem correctly or did not properly follow all the standards, please someone let me know and I will complete / correct.
I feel very lonely!

ginkgm, 2022-02-16:
Hi @philtao-dev ,

Thanks for reporting and investigate into this issue. The register printing will be very helpful. And the value before and after configuration helps us to know the default value and how the function works.

We need some time to look into this issue though..

Michael

alisitsyn, 2022-02-16:
Hi philtao-dev,

How is the console output configured in your project? By default it goes to UART0. Do you use the console port for modbus communication?

> After setting .rx_timeout_thresh = 20, the value of UART_RX_TOUT_THRSHD = 200 (why a factor 10?)

The definition of TOUT threshold in the register is different for ESP32 and ESP32S2 targets. 
The 1 in UART_MEM_CONF_REG.UART_RX_TOUT_THRHD  register corresponds to the 1 bit time.
the actual value in the register being set to (symb_len * tout) bit,  where `tout` is defined as a number of bytes on the current baud rate.

philtao-dev, 2022-02-17:
Thank you @ginkgm for looking into that issue.
I have the following code to print UART_CONF1_REG(0) and UART_MEM_CONF_REG(0) - let me know if you need more registers.
```cpp
void printReg() 
{
  int * ptr;
  ptr = (int *)UART_CONF1_REG(0);
  ESP_LOGI(TAG, "UART_CONF1_REG(0) value: 0x%X", (unsigned int)*ptr);

  ptr = (int*)UART_MEM_CONF_REG(0);
  ESP_LOGI(TAG, "UART_MEM_CONF_REG(0) value: 0x%X", (unsigned int)*ptr);
}
```
and I added the following in app_main() to print default value, and values after configuration (I have added the uart_set_rx_timeout() as suggested by @alisitsyn)
 ```cpp
 printReg();

  ESP_LOGI(TAG, "Configure and enable UART interrupt - rxfifo_full_thresh = 20 / rx_timeout_thresh = 20");
  uart_intr_config_t uart_intr = {
      .intr_enable_mask = UART_INTR_RXFIFO_TOUT | UART_INTR_RXFIFO_FULL,
      .rxfifo_full_thresh = 20,
      .rx_timeout_thresh = 20  
  };
  ESP_ERROR_CHECK(uart_intr_config(UART_NUM, &uart_intr));
  ESP_ERROR_CHECK(uart_enable_rx_intr(UART_NUM));
  printReg();

  ESP_LOGI(TAG, "Set uart_set_rx_timeout = 10");
  ESP_ERROR_CHECK(uart_set_rx_timeout(UART_NUM, 10));
  printReg();
```
Here is the output:
```
V (96) memspi: raw_chip_idV (5100) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (5100) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (5100) intr_alloc: Connected src 37 to int 12 (cpu 0)
I (5100) UART test: UART_CONF1_REG(0) value: 0x8000C078
I (5100) UART test: UART_MEM_CONF_REG(0) value: 0x640012
I (5100) UART test: Configure and enable UART interrupt - rxfifo_full_thresh = 20 / rx_timeout_thresh = 20
I (5100) UART test: UART_CONF1_REG(0) value: 0x8000C014
I (5100) UART test: UART_MEM_CONF_REG(0) value: 0xC80012
I (5100) UART test: Set uart_set_rx_timeout = 10
I (5100) UART test: UART_CONF1_REG(0) value: 0x8000C014
I (5100) UART test: UART_MEM_CONF_REG(0) value: 0x640012
```

philtao-dev, 2022-02-17:
Hi @alisitsyn,
The console is sent to the native USB CDC.  That seems to work well except I need to add a 5 sec time delay at the beginning of the program to give for the PC to detect the USB port and give me time to open the terminal.  How can i get the log data as you showed in your post using USB CDC console? Here is all that I get:
```
Executing task in folder 220204-084359-espidf-peripherals-uart: C:\Users\Utilisateur\.platformio\penv\Scripts\platformio.exe device monitor <

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, 
printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on COM17  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
V (96) memspi: raw_chip_idV (5100) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (5100) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (5100) intr_alloc: Connected src 37 to int 12 (cpu 0)
I (5100) UART test: UART_CONF1_REG(0) value: 0x8000C078
I (5100) UART test: UART_MEM_CONF_REG(0) value: 0x640012
I (5100) UART test: Configure and enable UART interrupt - rxfifo_full_thresh = 20 / rx_timeout_thresh = 20
I (5100) UART test: UART_CONF1_REG(0) value: 0x8000C014
I (5100) UART test: UART_MEM_CONF_REG(0) value: 0xC80012
I (5100) UART test: Set uart_set_rx_timeout = 10
I (5100) UART test: UART_CONF1_REG(0) value: 0x8000C014
I (5100) UART test: UART_MEM_CONF_REG(0) value: 0x640012
```
UART0 will be used for a Modbus slave, and UART1 will be used for a Modbus master.
At the moment I am not doing anything Modbus related.  I am just trying to have the UART work using the above sample code and sending data with PuTTY.

Thanks for making it clear about the difference between ESP32 and ESP32S2 (bit vs. byte)

As you can see in previous message I have added code to set timeout for TOUT interrupt using uart_set_rx_timeout()

Let me know what additional information and test you need.

alisitsyn, 2022-02-17:
Hi @philtao-dev ,
> The console is sent to the native USB CDC.

I was trying to reproduce the issue but it was unsuccessful. This time I was using the native USB console via `ESP System Settings -> Channel for console output (USB CDC)  --->` option. Below is the confirmation:
[image]
``` log
idf.py -p /dev/ttyACM0 monitor
Executing action: monitor
Running idf_monitor in directory /home/esp-man/esp32/IDFGH-6739
Executing "/home/esp-man/.espressif/python_env/idf5.0_py3.8_env/bin/python /home/esp-man/esp32/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- --target esp32s2 /home/esp-man/esp32/IDFGH-6739/build/uart_echo.elf -m '/home/esp-man/.espressif/python_env/idf5.0_py3.8_env/bin/python' '/home/esp-man/esp32/esp-idf/tools/idf.py' '-p' '/dev/ttyACM0'"...
--- idf_monitor on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
device reports readiness to read but returned no data (device disconnected or multiple access on port?)
Waiting for the device to reconnect.
I (34) boot: ESP-IDF v5.0-dev-1599-gb66cc63c41-dirty 2nd stage bootloader
I (35) boot: compile time 12:20:33
I (35) boot: chip revision: 0
I (35) boot.esp32s2: SPI Speed      : 80MHz
I (35) boot.esp32s2: SPI Mode       : DIO
I (35) boot.esp32s2: SPI Flash Size : 2MB
I (35) boot: Enabling RNG early entropy source...
I (36) boot: Partition Table:
I (36) boot: ## Label            Usage          Type ST Offset   Length
I (37) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (38) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (39) boot:  2 factory          factory app      00 00 00010000 00100000
I (40) boot: End of partition table
I (41) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=0803ch ( 32828) map
I (49) esp_image: segment 1: paddr=00018064 vaddr=3ffbdaa0 size=01d10h (  7440) load
I (51) esp_image: segment 2: paddr=00019d7c vaddr=40022000 size=0629ch ( 25244) load
I (58) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=16ebch ( 93884) map
I (77) esp_image: segment 4: paddr=00036ee4 vaddr=4002829c size=057f8h ( 22520) load
I (83) esp_image: segment 5: paddr=0003c6e4 vaddr=50000000 size=00010h (    16) load
I (90) boot: Loaded app from partition at offset 0x10000
I (91) boot: Disabling RNG early entropy source...
I (111) spi_flash: detected chip: generic
I (111) spi_flash: flash io: dio
I (112) cpu_start: Starting scheduler on PRO CPU.
I (16992) UART_TEST: received 18 bytes
I (16992) DATA: 31 31 31 31 31 31 31 31 31 31 31 31 31 31 31 31 
I (17002) DATA: 31 31 
I (21342) UART_TEST: received 20 bytes
I (21342) DATA: 32 32 32 32 32 32 32 32 32 32 32 32 32 32 32 32 
I (21352) DATA: 32 32 32 32 
I (21372) UART_TEST: received 6 bytes
I (21372) DATA: 32 32 32 32 32 32 
I (48592) UART_TEST: received 20 bytes
I (48592) DATA: 33 33 33 33 33 33 33 33 33 33 33 33 33 33 33 33 
I (48602) DATA: 33 33 33 33 
I (48622) UART_TEST: received 5 bytes
I (48622) DATA: 33 33 33 33 33
```
The first event is buffer full = 20 bytes: 
`I (48592) UART_TEST: received 20 bytes
I (48592) DATA: 33 33 33 33 33 33 33 33 33 33 33 33 33 33 33 33 
I (48602) DATA: 33 33 33 33`
, the second event is timeout set to 2 bytes ~ 22 bits idle period.
`I (48622) UART_TEST: received 5 bytes
I (48622) DATA: 33 33 33 33 33`

> The console is sent to the native USB CDC. That seems to work well except I need to add a 5 sec time delay at the beginning of the program to give for the PC to detect the USB port and give me time [...]

1. idf.py build
2. idf.py dfu
3. idf.py dfu flash
4. reset target using the button
5. send the string using cutecomm terminal without cr/lf
6. see the log above

[image]

philtao-dev, 2022-02-18:
Thanks @alisitsyn for all the efforts you put into that issue,
Great to see that all is working fine.  I just need to find out what is different in my setup / configuration. 
I use Platformio in a Windows environment.
For step "5. Reset target using the button", I briefly connect EN (pin 41) to GND, but that disconnects the native USB, so I loose the monitor console.  So I added a line with "**esp_reset()**" after the 5 sec delay in order to get the reboot log:
```
I (36) boot: ESP-IDF 4.4.0 2nd stage bootloader
I (36) boot: compile time 17:38:49
I (36) boot: chip revision: 0
I (37) qio_mode: Enabling default flash chip QIO
I (37) boot.esp32s2: SPI Speed      : 80MHz
I (37) boot.esp32s2: SPI Mode       : QIO
I (38) boot.esp32s2: SPI Flash Size : 4MB
I (38) boot: Enabling RNG early entropy source...
I (38) boot: Partition Table:
I (38) boot: ## Label            Usage          Type ST Offset   Length
I (40) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (41) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (42) boot:  2 factory          factory app      00 00 00010000 00100000
I (44) boot:  3 storage          Unknown data     01 82 00110000 00200000
I (45) boot: End of partition table
I (45) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=075fch ( 30204) map 
I (50) esp_image: segment 1: paddr=00017624 vaddr=3ffbceb0 size=02f18h ( 12056) load
I (52) esp_image: segment 2: paddr=0001a544 vaddr=40022000 size=05ad4h ( 23252) load
I (58) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=15af4h ( 88820) map 
I (72) esp_image: segment 4: paddr=00035b1c vaddr=40027ad4 size=053dch ( 21468) load
I (76) esp_image: segment 5: paddr=0003af00 vaddr=50000000 size=00010h (    16) load
I (83) boot: Loaded app from partition at offset 0x10000
I (84) boot: Disabling RNG early entropy source...
D (103) FLASH_HAL: extra_dummy: 0
V (103) memspi: raw_chip_id: 164020

V (104) memspi: chip_id: 204016

V (106) memspi: raw_chip_id: 164020

V (107) memspi: chip_id: 204016

D (108) spi_flash: trying chip: issi
D (108) spi_flash: trying chip: gd
D (108) spi_flash: trying chip: mxic
D (108) spi_flash: trying chip: winbond
D (108) spi_flash: trying chip: boya
D (109) spi_flash: trying chip: generic
I (109) spi_flash: detected chip: generic
I (109) spi_flash: flash io: qio
D (109) chip_generic: set_io_mode: status before 0x200
V (109) chip_generic: set_io_mode: status update 0x200
D (109) cpu_start: calling init function: 0x40080e7c
V (110) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (111) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE02
D (112) intr_alloc: Connected src 73 to int 10 (cpu 0)
I (112) esp_core_dump_uart: Init core dump to UART
V (112) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (114) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x40E
D (115) intr_alloc: Connected src 28 to int 3 (cpu 0)
I (115) cpu_start: Starting scheduler on PRO CPU.
D (115) heap_init: New heap initialised at 0x3fffc000
V (115) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (115) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (115) intr_alloc: Connected src 17 to int 9 (cpu 0)
```
Here is **Platformio.ini** - just in case there are some useful clues in there:
```
[env:esp32-s2-saola-1]
platform = espressif32
framework = espidf
board = esp32-s2-saola-1
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
build_type = debug
board_build.partitions = tao_bms_partitions.csv
```
and **sdkconfig.esp32-s2-saola-1**
[sdkconfig_220218.txt](https://github.com/espressif/esp-idf/files/8094342/sdkconfig_220218.txt)

The output I get running the test code and waiting a long time between characters has no effect:
[image]

alisitsyn, 2022-02-18:
@philtao-dev ,

> For step "5. Reset target using the button", I briefly connect EN (pin 41) to GND, but that disconnects the native USB, so I loose the monitor console. 

After the disconnection of the native USB, the chip will restart and USB device come up again then you can send your data.
If you can not do this we can continue with the console configured to UART1 and example UART connected to UART0 and repeat the results showed above.

> Log I (36) boot: ESP-IDF 4.4.0 2nd stage bootloader

I don't see any issues in your register bank dump and the TOUT feature should work.
In order to find a reason for issue we need to use the same official tools. I would ask you to clone and setup official release v4.4 of ESP-IDF because I can not guarantie that your commit installed as part of PlatformIO is correct. 
[setup ESP-IDF](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html)
I use the official v4.4, commit 5ff09f0a06. Once installed, please use `git describe` command in the esp-idf folder and send the output here. Then please compile and start the project I attached below with my sdkconfig file.
The output should be like v4.4-253-g5ff09f0a06.

[IDFGH-6739.zip.log](https://github.com/espressif/esp-idf/files/8095321/IDFGH-6739.zip.log)

philtao-dev, 2022-02-19:
@alisitsyn 
I have completely uninstalled esp-idf, VScode, python, esp-idf tools and anything related
Then I cloned the official esp-idf v4.4 (link you provided), and reinstalled VScode, ESP-IDF pre-requisite and extension (not using Platformio this time).
I then created a new projet where I copied your code and your sdkconfig, but still does not work!

output of git describe: `v5.0-dev-1599-gb66cc63c41`
Why is it now v5.0 and is followed by `-dirty` in the log?

Here is the log:
```
I (52) boot: ESP-IDF v5.0-dev-1599-gb66cc63c41-dirty 2nd stage bootloader
I (53) boot: compile time 12:16:35
I (53) boot: chip revision: 0
I (54) boot.esp32s2: SPI Speed      : 80MHz
I (54) boot.esp32s2: SPI Mode       : DIO
I (54) boot.esp32s2: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (55) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (56) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (58) boot:  2 factory          factory app      00 00 00010000 00100000
I (59) boot: End of partition table
I (59) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=0806ch ( 32876) map 
I (66) esp_image: segment 1: paddr=00018094 vaddr=3ffbdaa0 size=01d10h (  7440) load
I (68) esp_image: segment 2: paddr=00019dac vaddr=40022000 size=0626ch ( 25196) load
I (74) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=16ebch ( 93884) map 
I (94) esp_image: segment 4: paddr=00036ee4 vaddr=4002826c size=05828h ( 22568) load
I (101) esp_image: segment 5: paddr=0003c714 vaddr=50000000 size=00010h (    16) load
I (107) boot: Loaded app from partition at offset 0x10000
I (108) boot: Disabling RNG early entropy source...
I (128) spi_flash: detected chip: generic
I (129) spi_flash: flash io: dio
I (129) cpu_start: Starting scheduler on PRO CPU.
I (41850) UART_TEST: received 20 bytes
I (41850) DATA: 4e 4e 4e 4e 4e 4e 4e 4e 4e 4e 4e 4e 4e 4e 4e 4e 
I (41860) DATA: 4e 4e 4e 4e 
I (52280) UART_TEST: received 20 bytes
I (52280) DATA: 4e 13 13 13 13 13 13 13 13 13 13 13 13 13 13 13 
I (52290) DATA: 13 13 13 13 
```
Do you think it could have something to do with hardware as it is a custom PCB with a RS485 transceiver with automatic direction control connected to UART0?  I do not think so considering that this is the only issue I have and the characters are correctly received on UART0 and printed on the log.

alisitsyn, 2022-02-20:
@philtao-dev ,

Thank you for your result. I will try to reproduce this on my side with new clonned v4.4 on Windows 10.

>Why is it now v5.0 and is followed by -dirty in the log?

It is better to checkout the [commit](https://github.com/espressif/esp-idf/commit/5ff09f0a06627e1f5bb526517ea0f7a3edaeff63). However, I can repeat my result with v5.0 as well.

> Do you think it could have something to do with hardware as it is a custom PCB with a RS485 transceiver with automatic direction control connected to UART0?

I don't think this is the reason If you use the tranceiver with the automatic direction control like below:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#circuit-c-auto-switching-transmitter-receiver

alisitsyn, 2022-02-20:
Just to confirm Is this correct that if you send the < 20 bytes and you do not see the "UART_TEST: received N bytes"?

Another think: In some cases the TOUT function in the driver requires the call to [uart_set_always_rx_timeout](https://github.com/espressif/esp-idf/blob/master/components/freemodbus/port/portserial_m.c#L249) because the TOUT feature needs to leave at least one byte in the FIFO to work correctly.

philtao-dev, 2022-02-20:
The RS485 transceiver I use is the [MAX13487EESA+](https://datasheets.maximintegrated.com/en/ds/MAX13487E-MAX13488E.pdf) with auto direction control.

I confirm that if I send less than 20 bytes then stop, I do not see the "UART_TEST: received N bytes" output.

After adding `uart_set_always_rx_timeout(UART_NUM, true);`  the RX FIFO FULL interrupt happens after 19 bytes:
```
I (127) cpu_start: Starting scheduler on PRO CPU.
I (50568) UART_TEST: received 19 bytes
I (50568) DATA: 4f 27 27 27 27 27 27 27 27 27 27 27 27 27 27 27
I (50578) DATA: 27 27 27
I (77998) UART_TEST: received 19 bytes
I (77998) DATA: 27 26 26 26 26 26 26 26 26 26 26 26 26 26 26 26
I (78008) DATA: 26 26 26
```
Zip of the test project directory:   [uart_echo.zip](https://drive.google.com/file/d/1DYAsRal_uEUzEExG9VV4Q6eU9GRiBACn/view?usp=sharing)

alisitsyn, 2022-02-21:
@philtao-dev ,

Thank you for clarification. I would like to ask you about one more thing to record the RX, TX and A, B signals using logic analyzer or oscilloscope in your environment.

Thank you.

alisitsyn, 2022-02-21:
@philtao-dev ,

I was trying to reproduce the issue on my windows machine with `ESP-IDF v4.4-253-g5ff09f0a06` but it also works and can trigger timeout.
```
device reports readiness to read but returned no data (device disconnected or multiple access on port?)
Waiting for the device to reconnect.
I (34) boot: ESP-IDF v4.4-253-g5ff09f0a06-dirty 2nd stage bootloader
I (35) boot: compile time 14:03:07
I (35) boot: chip revision: 0
I (35) boot.esp32s2: SPI Speed      : 80MHz
I (35) boot.esp32s2: SPI Mode       : DIO
I (36) boot.esp32s2: SPI Flash Size : 2MB
I (36) boot: Enabling RNG early entropy source...
I (37) boot: Partition Table:
I (37) boot: ## Label            Usage          Type ST Offset   Length
I (38) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (39) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (40) boot:  2 factory          factory app      00 00 00010000 00100000
I (41) boot: End of partition table
I (42) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=080f4h ( 33012) map
I (49) esp_image: segment 1: paddr=0001811c vaddr=3ffbde50 size=01cb0h (  7344) load
I (51) esp_image: segment 2: paddr=00019dd4 vaddr=40022000 size=06244h ( 25156) load
I (57) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=16d8ch ( 93580) map
I (76) esp_image: segment 4: paddr=00036db4 vaddr=40028244 size=05c04h ( 23556) load
I (82) esp_image: segment 5: paddr=0003c9c0 vaddr=50000000 size=00010h (    16) load
I (89) boot: Loaded app from partition at offset 0x10000
I (90) boot: Disabling RNG early entropy source...
I (109) spi_flash: detected chip: generic
I (109) spi_flash: flash io: dio
W (109) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (110) cpu_start: Starting scheduler on PRO CPU.
I (10111) UART_TEST: UART_CONF1_REG(0) value: 0x8000C078
I (10111) UART_TEST: UART_MEM_CONF_REG(0) value: 0x640012
I (10111) UART_TEST: Configure and enable UART interrupt - rxfifo_full_thresh = 20 / rx_timeout_thresh = 2
I (10111) UART_TEST: UART_CONF1_REG(0) value: 0x8000C014
I (10111) UART_TEST: UART_MEM_CONF_REG(0) value: 0x140012
I (10111) UART_TEST: Set uart_set_rx_timeout = 10
I (10111) UART_TEST: UART_CONF1_REG(0) value: 0x8000C014
I (10111) UART_TEST: UART_MEM_CONF_REG(0) value: 0x640012
W (16541) INT: TOUT: 1
I (16551) UART_TEST: received 4 bytes
I (16551) DATA: 32 32 32 32 
W (22191) INT: TOUT: 1
I (22211) UART_TEST: received 8 bytes
I (22211) DATA: 64 73 61 66 64 73 61 66 
W (70771) INT: TOUT: 1
I (70791) UART_TEST: received 9 bytes
I (70791) DATA: 73 64 66 73 64 67 73 64 67 
W (73351) INT: TOUT: 1
I (73371) UART_TEST: received 9 bytes
I (73371) DATA: 73 64 66 73 64 67 73 64 67 
W (74691) INT: TOUT: 1
I (74711) UART_TEST: received 4 bytes
I (74711) DATA: 73 64 66 73 
W (75841) INT: TOUT: 1
I (75851) UART_TEST: received 4 bytes
I (75851) DATA: 73 64 66 73 
W (76921) INT: TOUT: 1
I (76931) UART_TEST: received 6 bytes
I (76931) DATA: 73 64 66 61 64 66 
W (78221) INT: TOUT: 1
I (78231) UART_TEST: received 5 bytes
I (78231) DATA: 64 73 66 64 73 
W (79251) INT: TOUT: 1
I (79271) UART_TEST: received 9 bytes
I (79271) DATA: 73 64 66 73 64 67 73 64 67 
W (80491) INT: TOUT: 1
I (80511) UART_TEST: received 9 bytes
I (80511) DATA: 73 64 66 73 64 67 73 64 67 
W (84821) INT: TOUT: 1
[... 94 lines cut ...]
I (42) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=0806ch ( 32876) map
I (49) esp_image: segment 1: paddr=00018094 vaddr=3ffbdaa0 size=01d10h (  7440) load
I (50) esp_image: segment 2: paddr=00019dac vaddr=40022000 size=0626ch ( 25196) load
I (57) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=16f0ch ( 93964) map
I (76) esp_image: segment 4: paddr=00036f34 vaddr=4002826c size=05828h ( 22568) load
I (82) esp_image: segment 5: paddr=0003c764 vaddr=50000000 size=00010h (    16) load
I (88) boot: Loaded app from partition at offset 0x10000
I (89) boot: Disabling RNG early entropy source...
I (109) spi_flash: detected chip: generic
I (109) spi_flash: flash io: dio
I (110) cpu_start: Starting scheduler on PRO CPU.
I (39230) UART_TEST: received 8 bytes
I (39230) DATA: 64 67 66 64 73 67 66 73 
I (48050) UART_TEST: received 11 bytes
I (48050) DATA: 54 65 73 74 20 73 74 72 69 6e 67 
I (51870) UART_TEST: received 4 bytes
I (51870) DATA: 31 32 33 34 
I (55030) UART_TEST: received 4 bytes
I (55030) DATA: 33 34 35 36
```
This also works just fine on my side and is able to generate the TOUT after receiving the data (see above).
The difference is: I use just USB-TTL adapter connected directly to RXD, TXD.
Please check the level shifting with your adapter then check and share the communication diagram.

Thanks.

philtao-dev, 2022-02-21:
@alisitsyn 
Traces sending character "a" (0x61) from PuTTY
RS485 A & B:
[image]
 
RXD0 (CH1) & TXD0 (CH2):
[image]

philtao-dev, 2022-02-22:
I have tested a new PCB I just assembled --> same result (interrupt only after 20 bytes)
The circuit on the PCB is straight forward from the RS485 connector --> MAX13487--> Level shifter --> RXD0 / TXD0

We know the ESP32S2 receives the bytes on RXD0 (RXD0 trace, and also printing of the 20 bytes)
I have just noticed that the printed bytes are not what I send:  I send character "a" `0x61`, and it print `0x4F`
On both sides the baud rate is the same (19200), with 8 data bits, 1 stop bit and no parity.

alisitsyn, 2022-02-22:
@philtao-dev,
Thanks for your report. These things happen often. It is good that the reason for the issue is finally resolved. Good luck with your project!

closed 2022-02-23 by Alvin1Zhang