espressif/esp-idf #8371: BUG: Master branch does not compile in C++

b1ackviking, 2022-02-10:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-S2
- IDF version: v5.0-dev-1519-g5810ed1d04
- Build System: idf.py
- Compiler version: xtensa-esp32s2-elf-g++ (crosstool-NG esp-2021r2) 8.4.0
- Operating System: Linux
- Using an IDE?: No

## Problem Description

[This commit](https://github.com/espressif/esp-idf/commit/c5fd79547ac9b7bae06fa660e9f814d18d3390b7) has broken the C++ build.

The following is not valid in C++ https://github.com/espressif/esp-idf/commit/c5fd79547ac9b7bae06fa660e9f814d18d3390b7#diff-08ec6194ee68acf46a8bfb6250fb32f14c5d3bc7c2d113f7b0a19fe070561d66R36-R42

And the following two snippets bring back the problem addressed here #5897 and here #6692 and solved with this commit https://github.com/jrepp/esp-idf/commit/611c092353b48e5f7fda0e340fbeb5d9d0212c8f

https://github.com/espressif/esp-idf/commit/c5fd79547ac9b7bae06fa660e9f814d18d3390b7#diff-c1b348028ea444fde69bc0cc1afdc340d7d6b0ecbcbcc569685a5b71ed051792R396-R400

https://github.com/espressif/esp-idf/commit/c5fd79547ac9b7bae06fa660e9f814d18d3390b7#diff-6a05639314f67df0bae4a9c0534d54898a2f79c93a7f00cd1b43621520815547R529-R533

### Expected Behavior

Build succeeds.

### Actual Behavior

Build fails with the following error:

```
FAILED: esp-idf/redfs/CMakeFiles/__idf_redfs.dir/redfs.cpp.obj
/home/b1ack/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++ -DHUB_REVISION=\"E4\" -DINCLUDE_xTaskGetCurrentTaskHandle=1 -I/home/b1ack/projects/abra/esp-firmware/build/config -I/home/b1ack/projects/abra/esp-firmware/components/redfs/include -I/home/b1ack/projects/abra/esp-firmware/components/redfs/port/include -I/home/b1ack/projects/abra/esp-firmware/components/redfs/reliance-edge/include -I/home/b1ack/projects/abra/esp-firmware/components/redfs/reliance-edge/os/freertos/include -I/home/b1ack/projects/abra/esp-firmware/components/redfs/reliance-edge/core/include -I/home/b1ack/projects/esp-idf/components/newlib/platform_include -I/home/b1ack/projects/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/b1ack/projects/esp-idf/components/freertos/esp_additions/include/freertos -I/home/b1ack/projects/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/b1ack/projects/esp-idf/components/freertos/esp_additions/include -I/home/b1ack/projects/esp-idf/components/esp_hw_support/include -I/home/b1ack/projects/esp-idf/components/esp_hw_support/include/soc -I/home/b1ack/projects/esp-idf/components/esp_hw_support/include/soc/esp32s2 -I/home/b1ack/projects/esp-idf/components/esp_hw_support/port/esp32s2/. -I/home/b1ack/projects/esp-idf/components/esp_hw_support/port/esp32s2/private_include -I/home/b1ack/projects/esp-idf/components/heap/include -I/home/b1ack/projects/esp-idf/components/log/include -I/home/b1ack/projects/esp-idf/components/lwip/include/apps -I/home/b1ack/projects/esp-idf/components/lwip/include/apps/sntp -I/home/b1ack/projects/esp-idf/components/lwip/lwip/src/include -I/home/b1ack/projects/esp-idf/components/lwip/port/esp32/include -I/home/b1ack/projects/esp-idf/components/lwip/port/esp32/include/arch -I/home/b1ack/projects/esp-idf/components/soc/include -I/home/b1ack/projects/esp-idf/components/soc/esp32s2/. -I/home/b1ack/projects/esp-idf/components/soc/esp32s2/include -I/home/b1ack/projects/esp-idf/components/hal/esp32s2/include -I/home/b1ack/projects/esp-idf/components/hal/include -I/home/b1ack/projects/esp-idf/components/hal/platform_port/include -I/home/b1ack/projects/esp-idf/components/esp_rom/include -I/home/b1ack/projects/esp-idf/components/esp_rom/include/esp32s2 -I/home/b1ack/projects/esp-idf/components/esp_rom/esp32s2 -I/home/b1ack/projects/esp-idf/components/esp_common/include -I/home/b1ack/projects/esp-idf/components/esp_system/include -I/home/b1ack/projects/esp-idf/components/esp_system/port/soc -I/home/b1ack/projects/esp-idf/components/esp_system/port/include/private -I/home/b1ack/projects/esp-idf/components/xtensa/include -I/home/b1ack/projects/esp-idf/components/xtensa/esp32s2/include -I/home/b1ack/projects/esp-idf/components/driver/include -I/home/b1ack/projects/esp-idf/components/driver/esp32s2/include -I/home/b1ack/projects/esp-idf/components/driver/deprecated -I/home/b1ack/projects/esp-idf/components/esp_pm/include -I/home/b1ack/projects/esp-idf/components/esp_ringbuf/include -I/home/b1ack/projects/esp-idf/components/efuse/include -I/home/b1ack/projects/esp-idf/components/efuse/esp32s2/include -I/home/b1ack/projects/esp-idf/components/vfs/include -I/home/b1ack/projects/esp-idf/components/esp_wifi/include -I/home/b1ack/projects/esp-idf/components/esp_event/include -I/home/b1ack/projects/esp-idf/components/esp_netif/include -I/home/b1ack/projects/esp-idf/components/esp_eth/include -I/home/b1ack/projects/esp-idf/components/tcpip_adapter/include -I/home/b1ack/projects/esp-idf/components/esp_phy/include -I/home/b1ack/projects/esp-idf/components/esp_phy/esp32s2/include -I/home/b1ack/projects/esp-idf/components/esp_timer/include -I/home/b1ack/projects/esp-idf/components/spi_flash/include -mlongcalls  -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fmacro-prefix-map=/home/b1ack/projects/abra/esp-firmware=. -fmacro-prefix-map=/home/b1ack/projects/esp-idf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu++11 -fexceptions -fno-rtti -D_GNU_SOURCE -DIDF_VER=\"v5.0-dev-1519-g5810ed1d04\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -Werror -std=gnu++2a -MD -MT esp-idf/redfs/CMakeFiles/__idf_redfs.dir/redfs.cpp.obj -MF esp-idf/redfs/CMakeFiles/__idf_redfs.dir/redfs.cpp.obj.d -o esp-idf/redfs/CMakeFiles/__idf_redfs.dir/redfs.cpp.obj -c /home/b1ack/projects/abra/esp-firmware/components/redfs/redfs.cpp
In file included from /home/b1ack/projects/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:77,
                 from /home/b1ack/projects/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/portable.h:58,
                 from /home/b1ack/projects/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:70,
                 from /home/b1ack/projects/esp-idf/components/vfs/include/esp_vfs.h:15,
                 from /home/b1ack/projects/abra/esp-firmware/components/redfs/redfs.cpp:6:
/home/b1ack/projects/esp-idf/components/esp_common/include/esp_macro.h:37:32: error: __VA_OPT__ can only appear in the expansion of a C++2a variadic macro [-Werror]
 #define CHOOSE_MACRO_VA_ARG(_0 __VA_OPT__(,) __VA_ARGS__, MACRO, ...) MACRO
                                ^
/home/b1ack/projects/esp-idf/components/esp_common/include/esp_macro.h:37:32: error: macro parameters must be comma-separated
 #define CHOOSE_MACRO_VA_ARG(_0 __VA_OPT__(,) __VA_ARGS__, MACRO, ...) MACRO
                                ^~~~~~~~~~
/home/b1ack/projects/esp-idf/components/esp_common/include/esp_macro.h:37:46: error: __VA_ARGS__ can only appear in the expansion of a C++11 variadic macro [-Werror]
 #define CHOOSE_MACRO_VA_ARG(_0 __VA_OPT__(,) __VA_ARGS__, MACRO, ...) MACRO
                                              ^
In file included from /home/b1ack/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/xtensa-esp32s2-elf/sys-include/stdlib.h:19,
                 from /home/b1ack/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/xtensa-esp32s2-elf/include/c++/8.4.0/cstdlib:75,
                 from /home/b1ack/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/xtensa-esp32s2-elf/include/c++/8.4.0/bits/stl_algo.h:59,
                 from /home/b1ack/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/xtensa-esp32s2-elf/include/c++/8.4.0/algorithm:62,
                 from /home/b1ack/projects/abra/esp-firmware/components/redfs/redfs.cpp:3:
/home/b1ack/projects/esp-idf/components/esp_common/include/esp_macro.h:41:36: error: '_0' was not declared in this scope
 _Static_assert(CHOOSE_MACRO_VA_ARG(_0, x, 0, 1) == 0, "CHOOSE_MACRO_VA_ARG() result does not match for 0 arguments");
                                    ^~
/home/b1ack/projects/esp-idf/components/esp_common/include/esp_macro.h:41:36: note: suggested alternative: 'B0'
/home/b1ack/projects/esp-idf/components/esp_common/include/esp_macro.h:41:40: error: 'x' was not declared in this scope
[... cut ...] GFV6S14r0A2rNUqjIDHWNFBNVKvVGqugWp1RMDoC6BBauNSUbjVqegMamE0Zj0rFgrQdZc6xapJmJNmdo0gpbeBep9X6gNsEnlSmg1KYxUaY31lTXhJN2SG3xvcum6tixYTQ0GKVUgDjc0hI4AW7TkSOAVriU57zZheBFq8358tVaEk1s%2BbWkAywCBJFbOQJt9azViVjdK69elLmlNkI%2Bu5HTnVqdIFiSCSRXXec05Fzg2pWzJdRjcPT/rA3AGDYcUN4abMpssfMRzMXnPWzc9mrz%2BawvFq0xEqLMSgsxaSfFxLjXUv5PSxUII%2BB6x6HvcIXLEh8sPtuc%2Bg9qmTnlaYJV49bqOA1em3VhrrZLyHBawZ0607jPEHhfqX%2Ba37P9ei5mobrn4juc83m0JHqy1RNsJWjNdiQsTYi3dgLAP4eeerpGauGh/C%2BMkFwRIz4zBBO7oEWZz5Ic6eh3DrNCOpueqp8Ft6GRHFAA)

b1ackviking, 2022-02-10:
All interested, I am going to submit a PR with a fix shortly.

0xjakob, 2022-02-10:
@b1ackviking Thanks for the problem description, we'll also take a look.

closed 2022-03-09 by espressif-bot