espressif/esp-idf #8373: ESP32S2 UART1 is not waking up from automatic light sleep.

nilesh-dryad, 2022-02-10:
## Environment

- Development Kit:      ESP32-S2-Kaluga-1
- Kit version (for WroverKit/PicoKit/DevKitC): V1.3
- Module or chip used:  ESP32-S2-Wrover
- IDF version: v4.3.1 (Stable)
- Build System:   idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:        external 5V-2A

## Problem Description

I am trying to use UART1 in automatic light sleep. I am using IO17 as UART1 TX and IO18 as UART1 RX. I am using UART_REF_TICK as clock to avoid garbage data. Transmission works fine without any issue. But on reception, some of the data is garbage. It is expected in light sleep but just few bytes and I am getting garbage data in between frames.

I am sending following data as ASCII string from my PC to ESP32S2.

`UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUSRING: 1,190,020005037B227478706B223A7B22696D6D65223A66616C73652C22746D7374223A343533363538332C2266726571223A3836392E3532352C2272666368223A302C22706F7765223A32372C226D6F6475223A224C4F5241222C2264617472223A22534631324257313235222C22636F6472223A22342F35222C2269706F6C223A747275652C2273697A65223A31362C226E637263223A747275652C2264617461223A2259483049434359414D5141456E58377031384C7764673D3D227D7D`

I added UUUUUUU series in the beginning to wake up the UART. U means 0x55. So it has enough positive edge to wake up the UART. But I get following.

`UUUUUUUUUUUUUUUUUUUUU�%�9� 1,190,020005037B227478706B223A7B22696D6D65223A66616C73652C22746D7374223A343533363538332C2266726571223A3836392E3532352C2272666368223A302C22706F7765223A32372C226D6F6475223A224C4F5241222C2264617472223A22534631324257313235222C22636F6472223A22342F35222C2269706F6C223A747275652C2273697A65223A31362C226E637263223A747275652C2264617461223A2259483049434359414D5141456E58377031384C7764673D3D227D7D`

Note that above string, I get in 120 bytes chunk so I merged it for showing here. Let me know if you need more detail or any other file.

### Expected Behavior
It should work with tickless idle and automatic light sleep and got the correct data everytime.

### Actual Behavior
It is not getting correct data.

### Steps to reproduce
I have attached app_main.c, uart.c and sdkconfig file. Please rename it to respective format. You can run those to easily reproduce it.
[uart.txt](https://github.com/espressif/esp-idf/files/8041492/uart.txt)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8041495/sdkconfig.txt)
[app_main.txt](https://github.com/espressif/esp-idf/files/8041496/app_main.txt)

negativekelvin, 2022-02-10:
You send an endless 0,1 stream, it will decode U's but it doesn't actually sync to a start bit until after an idle.

nilesh-dryad, 2022-02-10:
Yes you are right. I changed the sequence to other random number and now it receives the bytes I want. But I found another thing that it actually does not wake on UART RX, it wakes up on GPIO wakeup which I configured. I disabled UART wakeup from my application, still ESP32s2 wakes up on UART reception. So my question again is the same, it ESP32S2 wakes up on GPIO, then why is it not receiving all the bytes? Because on GPIO wakeup, 0101 sequence doesn't matter at all.

nilesh-dryad, 2022-02-10:
My bad. You are right. I am too involved in this so I forgot this. AFAIK It takes around 500us for UART to get ready.

negativekelvin, 2022-02-15:
If the uart becomes ready in the middle of a byte it will not sync with correct start bit and the frame will be corrupted.

nilesh-dryad, 2022-02-16:
I would simplify my question for better understanding for me.
How many bytes I might loose if I use 9600 baudrate and I receive something in light sleep mode.

nilesh-dryad, 2022-02-18:
I change the baudrate to 38400 and this is what I am getting. I am sending two extra bytes in the begining. Green Line is RX line and Blue line is the sleep status of ESP32S2. When Blue line is low, ESP32S2 is in light sleep and when it is high ESP32S2 is out of light sleep. I can afford to loose upto 4 bytes but no matter what baudrate I keep, ESP always wakes up after 1.2ms.

[image]

Look at this second image. Why ESP32S2 again goes to sleep immediately after wake up.

[image]

negativekelvin, 2022-02-18:
In general yes because of these limitations that you have to work around.

esp-lis, 2022-03-24:
> Yes you are right. I changed the sequence to other random number and now it receives the bytes I want. But I found another thing that it actually does not wake on UART RX, it wakes up on GPIO wakeup which I configured. I disabled UART wakeup from my application, still ESP32s2 wakes up on UART reception. So my question again is the same, If ESP32S2 wakes up on GPIO, then why is it not receiving all the bytes, (1-2 bytes loss is ok)? Because on GPIO wakeup, 0101 sequence doesn't matter at all.

In the latest esp-idf version (c1b3049), an example of the light sleep uart wakeup has been added. The path is as follows:
$IDF_PATH/examples/system/light_sleep/
for reference.

Also, to make sure that the UART is receiving correctly, I think the following steps should be required:
1. Enable UART wake-up and UART interrupt;
2. Give a WAKEUP type semaphore to task in UART ISR;
3. After task take a WAKEUP type semaphore, execute esp_pm_lock_acquire(ESP_PM_CPU_FREQ_MAX or ESP_PM_APB_FREQ_MAX) to prevent the system from entering sleep again;
4. Receive data from UART;
5. esp_pm_lock_release() to allow system enter sleep.

jack0c, 2022-05-17:
@nilesh-dryad any update?

nilesh-dryad, 2022-05-17:
@jack0c Nope it didn't wake up from UART. But I used some workaround to make it work. I used another GPIO connected with UART RX and used it as a GPIO interrupt and from that ISR I used power management lock to not let ESP go into light sleep for configurable time so that I can receive most of the UART frame. It worked for me. But It didn't work with PPP. I had to use AT commands for MQTT, HTTPS, TCP and UDP. It worked quite well for me. We can close this issue.

jack0c, 2022-06-15:
I will close the issue. Feel free to reopen necessary.

closed 2022-06-15 by jack0c