espressif/esp-idf #8376: Porting the example on Arduino framework

simogaspa84, 2022-02-10:
Hi guys.. 

I would like to try the Ethernet example controlling ENC28j60

https://github.com/espressif/esp-idf/tree/master/examples/ethernet/enc28j60 

in the arduino framework instead of the espressif framework

Is it possible... ?

Which steps are necessary ? 

Thanks a lot

me-no-dev, 2022-02-17:
It should be doable. Here are the steps:
1. Grab [main.c](https://github.com/espressif/esp-idf/blob/master/examples/ethernet/enc28j60/main/enc28j60_example_main.c) and paste it's contents into the `.ino` file of your sketch. Rename `app_main` to `setup` and add an empty `loop`
2. Copy all `.h` and `.c` files from the [component](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/enc28j60/components/eth_enc28j60) into your sketch folder.
3. Adjust any pin settings in the files, compile and upload

This should work, but I have not tried the example myself.

AndriiFilippov, 2022-05-13:
Hi @simogaspa84  ! 
Yes, it is possible. 
Please, use this guide to install the Arduino-ESP32 support: 
https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

1. Grab [main.c](https://github.com/espressif/esp-idf/blob/master/examples/ethernet/enc28j60/main/enc28j60_example_main.c) and paste it's contents into the .ino file of your sketch. Rename app_main to setup and add an empty loop
2. Copy all .h and .c files from the [component](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/enc28j60/components/eth_enc28j60) into your sketch folder.
3. Adjust any pin settings in the files, compile and upload

This ENC28J60 example works with ESP-IDF v4.4 + Arduino latest version 1.8.19.  
[ESP32_ENC28J60.zip](https://github.com/espressif/esp-idf/files/8687023/ESP32_ENC28J60.zip)

closed 2022-05-31 by Alvin1Zhang