espressif/esp-idf #8380: Is there any demo for esp32s3 display a Dynamic image with touch input device li9341

Kingwulin, 2022-02-11:
@suda-morris Hi,I had tried,but is not work ok,would you please give me a hand ,Thanks

suda-morris, 2022-02-11:
ili9341 has a similar register layout to st7789 if you're using an SPI interface. So please try the  `esp_lcd_new_panel_st7789`. We have several SPI interfaced LCD examples in esp-idf. e.g. `gc9a01`, `tjpgd`.

suda-morris, 2022-03-07:
Hi @ivanroberto It's not ready at the moment, as the LCD touch driver is more like a BSP driver instead of SOC driver. But due to the fact that most LCD application needs a touch input event, for convenience, we'd like to make such a driver in esp-idf or in a separate component package.

ivanroberto, 2022-03-07:
Hi @suda-morris, thanks for your reply. It would be great if there is some kind of integration of the input event. We are looking forward to this.
Thanks

tore-espressif, 2022-05-13:
Hello @ivanroberto , thank you for you patience.

Few days ago, we released first version of [esp_lcd_touch](https://components.espressif.com/component/espressif/esp_lcd_touch) altogether with drivers for some commonly used [touch controllers](https://components.espressif.com/search/esp_lcd_touch).

These components are distributed via [IDF Component Manager](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-component-manager.html).

Currently, we have only one example using it (more will follow soon) on our [ESP-BOX](https://www.espressif.com/en/news/ESP32-S3-BOX) development board.
The example can be found here: https://github.com/espressif/esp-bsp/tree/master/examples/touchscreen_colorwheel

The example uses ESP-BOX Board Support Package. If you haven't used IDF Component Manager in your project yet, you might find following info useful:

This is the example's (incomplete) dependency tree:
```
└─ touchscreen_colorwheel main
    └── esp-box
           └── lvgl/lvgl
           └── esp_lcd_touch_tt21100
                  └── esp_lcd_touch
... 
```
Description of used SW components:

- `lvgl` [Light and Versatile Graphics Library](https://github.com/lvgl/lvgl)
- `esp_lcd_touch` Base touch component (analogy to `esp_lcd`)
- `esp_lcd_touch_tt21100` Implementation of TT21100 touch controller present on ESP-BOX
- `esp-box` Board support package for ESP-BOX; contains 'glue' between `esp_lcd`, `esp_lcd_touch` and `LVGL`

If you have any questions about `esp_lcd_touch` please raise them in [esp-bsp repository](https://github.com/espressif/esp-bsp), thanks.

---
_Note_: All the touch components are version 0.9.0, which indicates that the API may change and that we are open to suggestions to make this API better. For example: in version 1.0.0 we plan to allow using the same controller with different interfaces (SPI/I2C), see: https://github.com/espressif/esp-bsp/issues/28

closed 2022-05-13 by igrr

ivanroberto, 2022-05-15:
HI dear @tore-espressif, thanks a lot! 
We appreciated all the work and support!

But, we are worried about the fact that esp_lcd_touch use I2C port without taking care of thread-safe issues.
I know that you probably do not take care of i2c driver, but if you can help to internalize this feature inside espressif's team, it would be nice!

Thanks a lot

ivanroberto, 2022-05-16:
Hi @tore-espressif, I just realized that espressif already has a thread-safe i2c driver called [i2c_bus](https://github.com/espressif/esp-box/tree/master/components/i2c_bus) used into esp-box repo.
Maybe if you change esp_lcd_touch_xxxxx components to use i2c_bus as an option on kconfig file (for those that still wanna use the direct i2c driver).
And publish [i2c_bus](https://github.com/espressif/esp-box/tree/master/components/i2c_bus) in IDF Component Manager.

Thanks

espzav, 2022-05-17:
Hi @ivanroberto, the [I2C driver](https://github.com/espressif/esp-idf/blob/master/components/driver/i2c.c) in esp-idf should be thread-safe. There is a semaphore in function `i2c_master_cmd_begin`.  

In new version of the esp_lcd_touch is used I2C from esp_lcd. More is here: https://github.com/espressif/esp-bsp/tree/feature/esp_lcd_touch_support_multiple_interfaces/components It will be merged, after it will be released IDF 4.4.2

tore-espressif, 2022-05-17:
> the [I2C driver](https://github.com/espressif/esp-idf/blob/master/components/driver/i2c.c) in esp-idf should be thread-safe

We will provide more info and update the documentation soon