espressif/esp-idf #8381: I modify the demo esp-idf/examples/peripherals/lcd/rgb_panel according the demo /esp-idf/examples/peripherals/lcd/i80_controller and the chart is always drift, How can I sovle it

Kingwulin, 2022-02-11:
I modify the demo esp-idf/examples/peripherals/lcd/rgb_panel according the demo /esp-idf/examples/peripherals/lcd/i80_controller
and the chart is always drift, How can I sovle it 
[image]

suda-morris, 2022-02-11:
If this is related to your last post in https://github.com/espressif/esp-idf/issues/8380 Please continue the conversation there, instead of opening a new one. Thanks. 

If this is a new issue you met, could you clarify it a little bit?  I'm not sure what kind of LCD you're using, and how you initialize your driver. Any code to reproduce?

Kingwulin, 2022-02-11:
两个是不同的问题
这个问题是这样的：
1、如果用原本 [demo ](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/rgb_panel) 是可以正常显示的
2、如果我仿照 [i80_controller](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/i80_controller) 这个例子添加 logo 显示，是现实坐标图回一直漂移，但是显示图片不断刷新输出也是一样会漂移
3、我所使用的屏幕是[这个](https://m.tb.cn/h.fPKESjy?sm=6bc1f2?tk=RWqN2c6lxzC)

Kingwulin, 2022-02-11:
你好
两个是不同的问题
这个问题是这样的：
1、如果用原本 [demo ](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/rgb_panel)是可以正常显示的
2、如果我仿照 [i80_controller](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/i80_controller) [这个](https://m.tb.cn/h.fPKESjy?sm=6bc1f2?tk=RWqN2c6lxzC)例子添加 logo 显示，显示的实坐标图回一直漂移，如果是只显示图片不断刷新输出也是一样会漂移
3、我所使用的屏幕是[这个](https://m.tb.cn/h.fPKESjy?sm=6bc1f2?tk=RWqN2c6lxzC)
[image]
[image]

suda-morris, 2022-02-11:
seems like this is a timing issue, you can try setting proper parameters in `panel_config.timings` according to your LCD spec.

Kingwulin, 2022-02-11:
I change the panel_config.timings according the datasheet below ,but the question is still exit
[2021923171633414.pdf](https://github.com/espressif/esp-idf/files/8046357/2021923171633414.pdf)
[image]

Kingwulin, 2022-02-11:
[image]
also I set the dclk freq to 9M/s

Kingwulin, 2022-02-11:
would you please share me anohter rgbpanel demo that display a dynamic image

suda-morris, 2022-03-04:
We found some hardware bug in the LCD_CAM peripheral, and we're working on a workaround on it. Will let you know when the fix is available on the master branch. Thanks.

closed 2022-03-18 by espressif-bot