espressif/esp-idf #8384: gen_esp32part.py unable to execute, getting UnicodeDecodeError: 'utf-8' codec can't decode byte

jemo07, 2022-02-11:
Hello, 

I'm executing the following command, to resolve an issue with partition table: 
python /esp/esp-idf/components/partition_table/gen_esp32part.py ptable.img ptable.cvs
The result of that command is: 
Traceback (most recent call last):
  File "/esp/esp-idf/components/partition_table/gen_esp32part.py", line 578, in <module>
    main()
  File "/esp/esp-idf/components/partition_table/gen_esp32part.py", line 526, in main
    table, input_is_binary = PartitionTable.from_file(args.input)
  File "/esp/esp-idf/components/partition_table/gen_esp32part.py", line 138, in from_file
    data = data.decode()
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xed in position 2: invalid continuation byte

Here is a hex capture of the ptable.img file: 
xxd ptable.img | head 
00000000: 0200 ed02 dc4d 017c f1c0 0000 21c1 ff31  .....M.|....!..1
00000010: c1ff c020 00d9 02c0 2000 3902 2195 fe0c  ... .... .9.!...
00000020: 1332 4200 21e4 ff32 0e00 226c 4bcd 02dc  .2B.!..2.."lK...
00000030: 1321 b9ff 3802 21b6 ffc0 2000 3902 010e  .!..8.!... .9...
00000040: f1c0 0000 2d0c 451e 0908 310c 12c8 21d8  ....-.E...1...!.
00000050: 11e8 0112 c110 0df0 0000 0000 12c1 f0c9  ................
00000060: 2109 31cd 0216 4207 4202 030c 0227 146c  !.1...B.B....'.l
00000070: 320c 0427 1366 3034 8042 a00f 3724 5d01  2..'.f04.B..7$].
00000080: 5ef1 c000 0021 e4ef 3d0c 22d2 0142 a006  ^....!..=."..B..
00000090: 22c2 3001 a9f2 c000 0021 75fe 0c13 3242  ".0......!u...2B
I can add the img file as needed.

jemo07, 2022-02-12:
Hello team, I've been trying to figure out what could be causing the issue to I added the following code to gen_esp32part.py above line 137 been referenced on the error: 

```
     print('>>>>>>',data)#deleteme
        data = data.decode()
```
Here is the output of the code of the `ptable.img`  when I execute `gen_esp32part.py ptable.img ptable.cvs`

`>>>>>> b'\x02\x00\xed\x02\xdcM\x01|\xf1\xc0\x00\x00!\xc1\xff1\xc1\xff\xc0 \x00\xd9\x02\xc0 \x009\x02!\x95\xfe\x0c\x132B\x00!\xe4\xff2\x0e\x00"lK\xcd\x02\xdc\x13!\xb9\xff8\x02!\xb6\xff\xc0 \x009\x02\x01\x0e\xf1\xc0\x00\x00-\x0cE\x1e\t\x081\x0c\x12\xc8!\xd8\x11\xe8\x01\x12\xc1\x10\r\xf0\x00\x00\x00\x00\x12\xc1\xf0\xc9!\t1\xcd\x02\x16B\x07B\x02\x03\x0c\x02\'\x14l2\x0c\x04\'\x13f04\x80B\xa0\x0f7$]\x01^\xf1\xc0\x00\x00!\xe4\xef=\x0c"\xd2\x01B\xa0\x06"\xc20\x01\xa9\xf2\xc0\x00\x00!u\xfe\x0c\x132B\x00E\xf1\xff!\xf6\xf8"\x02\x00f\x12"\xc5n\xfe\xcd\x02"\xa0\xfd ,\x10f\x12\x05\x85\x1a\xff\xc5\x10\xff\xc2\xcc\xfe\xc0\xc0t\xf6,\x05\x85(\x03\x05\xd4\x02\x01\xea\xf0\xc0\x00\x00\x0c\x12F\x00\x00\x00\x00\x081\xc8!\x12\xc1\x10\r\xf0\x00\x00D\x8b\xfe?d*\x10@\x12\xc1\xf0\xc9!\t1\xcd\x02\xcc\xc2!\xfb\xff\x01\xc7\xef\xc0\x00\x00-\x0c\xc6\x07\x00\x01;\xf1\xc0\x00\x00\x01\xf7\xff\xc0\x00\x00=\x02\x0cd-\x0c\x01\x87\xf2\xc0\x00\x00\x01\xd5\xf0\xc0\x00\x00\x0c\x12\x081\xc8!\x12\xc1\x10\r\xf0\x00\x00l\x00\xf2?\x12\xc1\xf0\xc9!\xd9\x11\t1\xdd\x02\x01-\xf1\xc0\x00\x00!\xb3\xef"\xd2\x02\xc2\x02\n\xcc,\x06\x0e\x00\x00A\xf7\xff\x0c\x1c\xc0 \x00(\x04\xc0" \xc0 \x00)\x04\xc0 \x008\x04\x0c" # \xc0 \x00)\x04\xc0 \x008\x04\x0cB # \xc0 \x00)\x04=\r"\xa0\x00\x05C\x1c\x01\xba\xf0\xc0\x00\x00\x081-\x0c\xd8\x11\xc8!\x12\xc1\x10\r\xf0\x00\x12\xc1\xf0\t1\xd9\x11\xe9\x01 \xd0t\xc2a\x02\x01\x11\xf1\xc0\x00\x00\x05^\xfe\xed\x02!\xaf\xf8"\x02\x00\x8c\x12&\x1e\n\x01\xab\xf0\xc0\x00\x00\x0c\x02\xc6(\x00\xc1\x90\xef"\x0c\n\x16-\x05f\x12\x0c\x01\xa6\xf0\xc0\x00\x00\x0c\x12\x06#\x00\x00\x00E\x07\xffE\x97\x1c1\xd0\xff"\xaf\xfe\xc0 \x00B#\x00 $\x10\xc0 \x00)\x03\xc0 \x00H\x03|\xd2 $\x10\xc0 \x00)\x03\xc0 \x00H\x03|\xb2 $\x10\xc0 \x00"c\x00\xe2L\n\xc5\x98\x1c\x86\x10\x00\x00\x16\xc2\xfaE\x93\x1cE\xae\x1c1\xbe\xff\xc0 \x00(\x03\xe0\xe2 \xc0 \x00\xe9\x03\xc0 \x00H\x03\x0c" $ \xc0 \x00)\x03\xc0 \x00H\x03\x0cB $ \xc0 \x00)\x031\x93\xef-\r\x055\x1c\xd2L\nE\x91\x1c\xc6\xd9\xff\x081\xc8!\xd8\x11\xe8\x01\x12\xc1\x10\r\xf0\x00\x00\x00\x98\xef\xfe?1\xff\xff)\x03\r\xf0\x00\x12\xc1\xf0\xd9\x11 \xd0t\xc9!\t1\xcd\x03\x01\xd5\xf0\xc0\x00\x00\xf6-\x01\xcc\x9c\x01r\xf0\xc0\x00\x00\x0c\x02\xc6\x15\x00-\r\x85\xcc\xfd\x8c\xb28"9\x0c82(B9\x1cF\x0e\x00\xec\r!<\xf7"\x02\x00\x8c\x82\x0c\x02)\x0c)\x1c\x86\t\x00\x00!d\xf0829\x0c8B(R9\x1cF\x05\x00!,\xf7"\x02\x00V\xc2\xfd!]\xf08\x029\x0c8\x12("9\x1c),\x01\\\xf0\xc0\x00\x00\x0c\x12\x081\xc8!\xd8\x11\x12\xc1\x10\r\xf0\x00\x12\xc1\xe0\tq\xc9a\x01\xb5\xf0\xc0\x00\x00\x05G\xfe\xb6"\x0b!:\xef"\xd2\x02"\x02\nf\x12\n\x01O\xf0\xc0\x00\x00\x0c\x02\xc6\t\x00\x0c\x12\x85\xc3\xfd\xc1\x15\xf7\x8c\xf2"\x0c\x00\xcc\xa2\x0c\x12=\x01\x85\xf3\xff-\x01\x85\x93A\x0c\x12"L\x00\x01D\xf0\xc0\x00\x00\x0c\x12\x08q\xc8a\x12\xc1 \r\xf0\x00\x12\xc1\xf0\xd9\x11 \xd0t\xc9!\t1\xcd\x03\x01\x9c\xf0\xc0\x00\x00\xf6-\x01\xcc\x9c\x019\xf0\xc0\x00\x00\x0c\x02\xc6\x14\x00-\rE\xbe\xfd\xcc\xbd1\x07\xf72\x03\x00\x8c\xf3\x86\xf8\xff\x00\x001\xfc\xf62\x03\x00\x9cCF\xf5\xff\x00H\x0c1,\xf0I3H\x1cICH,IS\x86\x03\x00H\x0c1(\xf0I\x03H\x1cI\x13H,I#\x8cr=\x0cKL\x8b\\\x05\xbbF\x01$\xf0\xc0\x00\x00\x0c\x12\x081\xc8!\xd8\x11\x12\xc1\x10\r\xf0\x00\x12\xc1\xf0\xc9!\xd9\x11\t1  t0\xd0t\x0c\x0c\xf6"\x19\x05\xb7\xfd\xc7\x12\x13\xc7\x9d\x06\xc5\xe1F\x06\x02\x00\x00\x0c\x13\x10\x11 \xc5\xd2F\x0c\x1c\x081-\x0c\xd8\x11\xc8!\x12\xc1\x10\r\xf0\x00\x12\xc1\xf0\xe9\x01\t1\xc9!\xd9\x11  t\xed\x03\x0c\x14\xf6"\x02B\xa0\x00\x0c\x03\x0c\x1d\xcd\x03\xe0\xcd\x83@\xcc \xc0\xc0t7\x9c\x19\x05\xb2\xfd=\x02\x9c2-\x0e2\xc3\x14\x1c\x04\x01\xb2\xf1\xc0\x00\x00\xcd\r\x86\x00\x00\x00\xcd\x03\x081-\x0c\xd8\x11\xc8!\xe8\x01\x12\xc1\x10\r\xf0\x00\x00\x00\x12\xc1\xe0\xe9A\xf91\tq\xc9a\xd9Q  t\xed\x040\xf0t\x0c\x15\xf6"\x01\x0c\x05\x0c\x04\x0c\x1d\xcd\x04\xe0\xcd\x83P\xcc \xc0\xc0tG\x9c\x1dE\xac\xfd\x9c\x92+?\xc03\x11:2K3-\x0e\x1c\x04\x01\x9a\xf1\xc0\x00\x00\xcd\r\x86\x00\x00\x00\xcd\x04\x08q-\x0c\xd8Q\xc8a\xe8A\xf81\x12\xc1 \r\xf0\x00\x12\xc1\xf0\xc9!\xd9\x11\x02a\x03 \xd0t0\xc3 \xf6-P\x16\xd3\x04\x01<\xf0\xc0\x00\x00-\rE\xa7\xfdM\x02\x8c\xd2"\x02\x83\xdc\xd2"$!\xdc\x82"\x04\x88\xdc21\xd3\xef-\x0c\xccM2\xc3\x1e\x86\x03\x002\xc3\x18\x06\x02\x00\x00\x002\xa0\x83-\x0c:4\x0cd\x01}\xf1\xc0\x00\x00\x01\xcc\xef\xc0\x00\x00\x0c\x12\x86\x00\x00\x00\x0c\x02\x081\xc8!\xd8\x11\x12\xc1\x10\r\xf0\x00\x00\x00\x80\xf2\xfe?\x84\xf2\xfe?\x12\xc1\xd0\x02a\x0b\xc2a\n\x01 \xf0\xc0\x00\x00"\xa0\x012\xc1\x14\xc5\xf6\xff\xc1\xa5\xee",\xb8&\x02\x0b"\xdc\x022\x02\xe4"\xa0\xff\'\x93,!\x03\xfc=\x01\x1cD\x85\x96\xfd!\xf0\xff\x0c\x03,D\x01\x17\xf0\xc0\x00\x00!\xee\xffB\x01\x17R\x01\x18b\x01\x19=\x01E\xf0\x1d\x0c\xa2"l\xb8\x01\xac\xef\xc0\x00\x00\x08\xb1\x0c\x12\xc8\xa1\x12\xc10\r\xf0\x00\x80\x93\xfe?|\x93\xfe?\xe5\xf2\xfe?\xfc\xe9\xff\xff\x9c\x0b\x00\x00\xeb\x0b\x00\x00\xa0\xf4\xfe?\xc0*\x10@\xac*\x10@\x12\xc1\xd0\t\xb1\xc9\xa19\x11\xcd\x02\xd9\x91\xe9\x81\xf2a\x07\x01\xfa\xef\xc0\x00\x00E\x18\xfe\x16\x0c\x0c\xf6"\x02\x86.\x00\xd1~\xee"\xdd\x02"\x02\nf\x12\x03\xc6*\x00\x00\x0cbE\xd0\xfcf\x02\x18!\xe8\xff=\x0cB\xa0l\x01>\xf1\xc0\x00\x00!\xe5\xff8\x112B\x00\x06\xd2\x00\xe1\xc5\xff=\x0c-\x0eB\xa0 \x01\xe8\xef\xc0\x00\x002\xcc 9\x01\xf2,\x19V\xd2\x05!\x97\xeeB\xa0@\x01\xe2\xef\xc0\x00\x00V\xe2\x042\xdd [... cut ...]
The ptable was created with the following command: `esptool.py --port /dev/ttyUSB0 -b 115200 read_flash 0x9000 0xc00 ptable.img`
The image loaded on the ESP8266 checks out with the following: 
```
esptool.py --chip esp8266 image_info flash_4M.bin 
esptool.py v3.3-dev
Image version: 1
Entry point: 4010057c
3 segments

Segment 1: len 0x00a20 load 0x40100000 file_offs 0x00000008 [IRAM]
Segment 2: len 0x002fc load 0x3ffe8000 file_offs 0x00000a30 [DRAM]
Segment 3: len 0x002a4 load 0x3ffe82fc file_offs 0x00000d34 [DRAM]
```

Please advice...

jemo07, 2022-02-12:
Also, the same issue occurs when I attempt to run the tool in the getting the partition size: 
```
parttool.py --port "/dev/ttyUSB0" get_partition_info --partition-boot-default --info size
Running ~/.espressif/python_env/idf5.0_py3.8_env/bin/python /esp/esp-idf/components/esptool_py/esptool/esptool.py --port /dev/ttyUSB0 read_flash 32768 3072 /tmp/tmp3x19sok2...
esptool.py v3.3-dev
Serial port /dev/ttyUSB0
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 60:01:94:37:d3:d9
Stub is already running. No upload is necessary.
3072 (100 %)
3072 (100 %)
Read 3072 bytes at 0x8000 in 0.3 seconds (77.8 kbit/s)...
Hard resetting via RTS pin...
Traceback (most recent call last):
  File "/home/jemo/esp/esp-idf/components/partition_table/parttool.py", line 365, in <module>
    main()
  File "/home/jemo/esp/esp-idf/components/partition_table/parttool.py", line 334, in main
    target = ParttoolTarget(**target_args)
  File "/home/jemo/esp/esp-idf/components/partition_table/parttool.py", line 116, in __init__
    partition_table = gen.PartitionTable.from_binary(f.read())
  File "/home/jemo/esp/esp-idf/components/partition_table/gen_esp32part.py", line 281, in from_binary
    result.append(PartitionDefinition.from_binary(data))
  File "/home/jemo/esp/esp-idf/components/partition_table/gen_esp32part.py", line 433, in from_binary
    res.name = res.name.decode()
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xf9 in position 2: invalid start byte
```

radimkarnis, 2022-02-14:
Hello @jemo07,

it seems like you are decoding an ESP8266 image. ESP8266 is not supported in ESP-IDF.

Anyways, can you please provide the img file? I will try to see if we can easily make the tool work!

Radim

jemo07, 2022-02-14:
Thank you Radim Ah, I see, I did not realize this and thank you for taking the time to look into it. Just wondering if there is a simpler solution, given the output of estool.py: 
```
esptool.py --chip esp8266 image_info flash_4M.bin 
esptool.py v3.3-dev
Image version: 1
Entry point: 4010057c
3 segments

Segment 1: len 0x00a20 load 0x40100000 file_offs 0x00000008 [IRAM]
Segment 2: len 0x002fc load 0x3ffe8000 file_offs 0x00000a30 [DRAM]
Segment 3: len 0x002a4 load 0x3ffe82fc file_offs 0x00000d34 [DRAM]
```

Can't I leverage this to read from a segment? How to I find the offset of the flash to do that? That would help me understand how to write the different segments into a single image, boot+app1+app2. 

Cheers,

dobairoland, 2022-02-16:
Hi @jemo07. I've compared the ESP-IDF version of the script with https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/components/partition_table/gen_esp32part.py. There are some difference between them not counting the syntax changes. Could you please try that one as well?

Also why would you want to download the partition table in binary format and transform back to CSV? Don't you have already the original version of the partition table in CSV format from which the binary was generated? (I'm not an 8266 SDK user so please pardon me my questions)

dobairoland, 2022-02-16:
What could help us to reproduce the issue is sharing the original partition table in CSV format as well as in binary format.

Alvin1Zhang, 2022-03-04:
@jemo07 Thanks for reporting, would you please help share if any updates for the issue? Thanks.

FernandoGarcia, 2022-04-01:
Hi!

I'm getting this error with ESP32-Wrover IE.

I'm running this commands:

```
esptool --port /dev/ttyUSB0 -b 115200 read_flash 0x8000 0xc00 ptable.img

python /home/fernandogarcia/.arduino15/packages/esp32/hardware/esp32/2.0.2/tools/gen_esp32part.py ptable.img
```
The result is:

```
Traceback (most recent call last):
  File "/home/fernandogarcia/.arduino15/packages/esp32/hardware/esp32/2.0.2/tools/gen_esp32part.py", line 578, in <module>
    main()
  File "/home/fernandogarcia/.arduino15/packages/esp32/hardware/esp32/2.0.2/tools/gen_esp32part.py", line 526, in main
    table, input_is_binary = PartitionTable.from_file(args.input)
  File "/home/fernandogarcia/.arduino15/packages/esp32/hardware/esp32/2.0.2/tools/gen_esp32part.py", line 138, in from_file
    data = data.decode()
UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 0: ordinal not in range(128)
```
Any suggestion to fix it?

Best regards.

jemo07, 2022-04-29:
> 

Hello @Alvin1Zhang No progress, (Covid and travel got on the way, apologies) I was not able to extract the partition table even with the ES8266 version. Been trying many things. I was hoping to try other options but I have not had any progressed. I'm happy to share the bin files, it is a commercial product that I don't own the rights to, so I need some other way of sharing that is not public.

closed 2023-06-13 by Alvin1Zhang