espressif/esp-idf #8385: Bluetooth AVRCP browsing support

sgolob, 2022-02-11:
**Is your feature request related to a problem? Please describe.**

Currently, song navigation is limited to next/previous commands.

**Describe the solution you'd like**

When using ESP as A2DP sink / AVRCP controller, I would like to navigate and view media content on my phone using AVRCP browsing channel as described in Bluetooth AVRCP specification 1.4 and newer.

wmy-espressif, 2022-02-21:
Hi @sgolob  thanks for the question. AVRCP browsing channel is listed in our current development plan. AVRCP browsing channel uses L2CAP Enhanced Retransmission Mode(ERTM in short), L2CAP ERTM is supposed to be supported in ESP-IDF 5.0, and we will consider to support of AVRCP browsing channel after that.

Craft4Cube, 2024-08-20:
Any updates on this? I would also be interested in using the AVRCP Browsing Functionality on an ESP32

wmy-espressif, 2024-08-27:
Hi @Craft4Cube, according to our plan, AVRCP browsing function(for AVRCP Controller role) can be delivered by the end of this year.

Currently we are developing another feature: AVRCP cover art, and it is expected to be delivered a couple of weeks later, before esp-idf v5.4 is released.