espressif/esp-idf #8386: MSC cannot continue work after reinsert the USB flash disk

starr3333, 2022-02-11:
## Environment

- Development Kit:   ESP32-S2-DevKitC-1
- Module or chip used:  ESP32-­S2­-SOLO-­U
- IDF version : idf-v4.4
- Build System:  idf.py
- Compiler version : toolchain-xtensa-esp32s2 @ 8.4.0+2021r1
- Operating System:  Windows 10
- Using an IDE?: Platformio
- Power Supply: USB

## Problem Description
I download the esp-idf official example of MSC, and modified this program so that it can read a file in a loop.

When I try the MSC example, I found that the current MSC example program doesn't seem to support hot-plugging. When the USB flash disk was removed and then reconnected, the usb_host and msc_host library will not resume to correct state.

### Expected Behavior
In the official MSC example, if the USB flash disk was removed, we can not uninstall the usb_host, because usb_host_uninstall() in the program will return a 0x103 error.

if we commit the usb_host_uninstall() , and reinstall the msc_host, I found that the xQueueReceive will not return anything, Therefore, the program will not be able to determine the status of the USB flash disk.

### Steps to reproduce
I have attached app_main.c, You can run those to easily reproduce it.
[app_main.txt](https://github.com/espressif/esp-idf/files/8048734/app_main.txt)

I made a few modify from the official program. I used the connectState variable to detect the status of the USB flash disk. Once it is detected that the flash disk is disconnected, the program will stop reading the text and reinstall the msc_host and usb_host.

The original official program is:
https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/host/msc/main/msc_example_main.c

## Debug Logs
(Before this, the device can work normally, now we reinsert the USB flash disk)
[  6005] file_operations(): Read from file: 'helloword123
[  6009]  USBH: Device 1 gone
[  6015] msc_event_cb(): MSC device disconnected
[  6570] app_main(): Done
[  6571] wait_for_msc_device(): Waiting for USB stick to be connected
[  6571] wait_for_msc_device(): Waiting for connect...
[  6578] wait_for_msc_device(): Waiting for connect...
(no more output)

starr3333, 2022-02-15:
thank you for your reply,
It's a good compromise to solve this problem!

Looking forward to supporting hot-plugging in the future :)

closed 2022-02-15 by starr3333

esp-saurabhbansal, 2023-04-14:
Hello,

Just for your information, We have recently released USB Device MSC driver. Please check if it is useful to you. We highly appreciate the feedback.

The driver makes ESP based device recognizable by USB-hosts as a USB Mass Storage Device.
It either allows the embedded application to access the partition or Host PC accesses the partition over USB MSC. They can't be allowed to access the partition at the same time.

The driver supports storage media of two types:
1. SPI Flash
2. SD MMC Card

Driver can be found [here](https://github.com/espressif/idf-extra-components/blob/master/usb/esp_tinyusb/tusb_msc_storage.c).
Sample application can be found [here](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_msc).

Thanks.