espressif/esp-idf #8389: unable to clone the master and stable repos

rhr407, 2022-02-11:
~/esp$ git clone -b v4.4 --recursive https://github.com/espressif/esp-idf.git
Cloning into 'esp-idf'...
remote: Enumerating objects: 307129, done.
remote: Counting objects: 100% (190/190), done.
remote: Compressing objects: 100% (92/92), done.
error: RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed

o-marshmallow, 2022-02-15:
Hi @rhr407 ,

Can you try cloning with SSH instead?
```
git clone -b v4.4 --recursive git@github.com:espressif/esp-idf.git
```

dobairoland, 2022-02-15:
Another workaround is to download a ZIP from the [releases](https://github.com/espressif/esp-idf/releases) page.

I'd like to point out that this is not an issue with ESP-IDF. The error is related to your Internet connection (instability can cause this), git and or Github. Often repeating the operation helps.

closed 2022-02-15 by dobairoland