espressif/esp-idf #8390: Provide early hook for hardware init

septatrix, 2022-02-11:
**Is your feature request related to a problem? Please describe.**

To save power on small devices it is common to only power the device when a button is pressed activating a mosfet or similar. Afterwards the device can pull-down the mosfet itself to stay active after the button is released. Even with logging and image verification disabled, no RTC calibration cycles etc the earliest possible point is after ~6,691,112 clock cycles (~42ms@160MHz) on an ESP32. Similar times can be observed for an ESP32-C3. For some button presses this is too slow.
Also some of the measures to reduce boot time might be undesirable.

**Describe the solution you'd like**

I would like a hook somewhere in [`call_start_cpu0`](https://github.com/espressif/esp-idf/blob/e8cba8df281c8bcc3ad281dd55772a5709c093d7/components/esp_system/port/cpu_start.c#L258). This could be archived using e.g. a weak (aliased) function (like `esp_perip_clk_init` or `start_cpu0`) or some sections like it is the case for `__attribute__((constructor))` and `ESP_SYSTEM_INIT_FN`.
This hook would preferably be placed as soon as possible. Much of the system could still be uninitialized as long as the GPIO peripherals work. As such this might even be possible before `.bss` is cleared though having the hook a bit later might make it more useful for other use cases.

**Describe alternatives you've considered**

Overwriting `start_cpu0_default` by declaring `start_cpu0` was tried though the gained reaction time in comparison to `__attribute__((constructor))` or `app_main` was miniscule as the constructors are already almost the first thing the function does and the freertos overhead is rather small. Another bummer was that the former is static such that it was not possible to wrap it but it would instead have to be completely copied only to add a few lines.

Solving this in hardware was also considered but did not deem satisfactory.

septatrix, 2022-02-11:
> However I'd recommend doing this in bootloader instead to minimize the delay and also to minimize the variance of the delay (e.g. in case bootloader has rollback or security features enabled.) There is an example on adding custom hooks to the 2nd stage bootloader: https://github.com/espressif/esp-idf/tree/e8cba8df281c8bcc3ad281dd55772a5709c093d7/examples/custom_bootloader/bootloader_hooks

Thanks for the quick reply. Yes after revisiting I also thought about the bootloader and am currently experimenting with it. The examples say that many things do not work before the bootloader (and some probably also still not until `call_start_cpu0` ran...is it still possible to use the GPIO driver/component or would I have to write to the raw registers instead?

> One thing you could try to do is wrap call_start_cpu0 using `-Wl,--wrap=call_start_cpu0` linker argument. In the `__wrap_call_start_cpu0` wrapper you can set up the GPIOs and then call `__real_call_start_cpu0`. Make sure you don't call any functions and don't rely on static data to be zero-initialized. Also you need to place the wrapper function into IRAM using IRAM_ATTR. (Example: [CMakeLists.txt](https://github.com/espressif/esp-idf/blob/0ebf93c01e826b3311a7cae5bdd85c4c1a68db31/components/esp_rom/CMakeLists.txt#L200), [wrapper](https://github.com/espressif/esp-idf/blob/0ebf93c01e826b3311a7cae5bdd85c4c1a68db31/components/esp_rom/patches/esp_rom_longjmp.S#L40-L71) — in this case it's in assembly, but the idea is the same.)

Yes I guess that could work but I would like to not tinker with the build system - although it looks quite clean with CMake. And given that the bootloader hooks seem like a cleaner solution I will tinker with them for now.

igrr, 2022-02-11:
No, i'm afraid GPIO driver would be unusable in both cases (bootloader and pre-call_start_cpu0). You'll have to write to GPIO registers directly. Please check the technical reference manual for details, it should only be a couple of registers.

> but I would like to not tinker with the build system

Note that you can add such wrappers in your own component CMakeLists file, for example in the `main` component. You don't have to modify ESP-IDF to achieve this. The solution with wrappers is a lot more fragile than the solution with bootloader hooks, though.

There is another option in case you are using a bare chip in your design (not a module). You could then use VDDSDIO pin as the signal to control the MOSFET. VDDSDIO starts supplying the flash chip when the ESP32 powers up, and loses power when ESP32 enters deep sleep mode. Note, VDDSDIO will also go low in light sleep mode, but you can inhibit this using `esp_sleep_pd_config(ESP_PD_DOMAIN_VDDSDIO, ESP_PD_OPTION_ON)` before entering light sleep.

You could also use one of the unused pins with weak pull up status, such as GPIO15 on ESP32.

septatrix, 2022-02-11:
> No, i'm afraid GPIO driver would be unusable in both cases (bootloader and pre-call_start_cpu0). You'll have to write to GPIO registers directly. Please check the technical reference manual for details, it should only be a couple of registers.

Yes, for the C3 that should be `0x803` to `0x805`.

> Note that you can add such wrappers in your own component CMakeLists file, for example in the `main` component. You don't have to modify ESP-IDF to achieve this. The solution with wrappers is a lot [...]

I agree that the bootloader solution is more robust. Sadly I have started this project with PlatformIO which does not seem to compile the bootloader. However as it is rather small I think it should not be too cumbersome to switch to "native" esp-idf.

> There is another option in case you are using a bare chip in your design (not a module). You could then use VDDSDIO pin as the signal to control the MOSFET. VDDSDIO starts supplying the flash chip [...]

As the device will only send a single ESP-NOW packet this might be perfect. However the C3 does not seem to have that pin, correct?

> You could also use one of the unused pins with weak pull up status, such as GPIO15 on ESP32.

I would have to check the design to see if that would be strong enough or maybe too weak.

negativekelvin, 2022-02-12:
C3 has VDD_SPI.  Also you can use rom gpio functions.

septatrix, 2022-02-12:
> Also you can use rom gpio functions.

By which you probably mean to ones here: [`components/esp_rom/include/esp32c3/rom/gpio.h`](https://github.com/espressif/esp-idf/blob/20847eeb96c486629a678d6ffbe73670013fb771/components/esp_rom/include/esp32c3/rom/gpio.h)? It says that they are part of the private API which I would like to avoid but I guess that is acceptable in comparison to writing to raw registers.

> C3 has VDD_SPI.

That also seems great if the behaviour is the same. The manual only says that it is configured as input-disabled on reset though it does not state that it is an output or which level it has (which I would assume to mean that it is in high-impedance). Or is it configured in the normal bootloader? Using a GPIO however should also work as the default pull-down on that line is very weak in comparison to the one in the ESP32-C3 (1MΩ vs 45kΩ). Currently the GPIO3 is used which is one of the few without a pull-up on reset. think a pure hardware solution might actually work then: What do you think @makermoekoe (as this might be useful for your fabulous Picoclick-C3)?

makermoekoe, 2022-02-14:
This idea is amazing.
Next to the fact that we don't have to struggle with toggling a GPIO in early bootloader state this solution will probably work with either any ESP32 chip as well as different IDE's. I have to give it a try on a spare Picoclick and measure the delay between button press and VDD_SPI. Will keep you updated.

septatrix, 2022-02-17:
I have no bare chip at hand and all of my dev boards have embedded flash so the VDD_SPI is not accessible. However based on the documentation I would assume that you have to manually configure the pin as a normal GPIO part and then turn that off

boarchuz, 2022-02-17:
@makermoekoe Do you have an external pulldown resistor on the LDO enable?
I wouldn't expect VDD_SPI to be actively discharged and pulled low; assume hi-z.

makermoekoe, 2022-02-18:
> I have no bare chip at hand and all of my dev boards have embedded flash so the VDD_SPI is not accessible. However based on the documentation I would assume that you have to manually configure the pin as a normal GPIO part and then turn that off.
> 
> There is also the EFUSE_VDD_SPI_AS_GPIO fuse though I am not sure if that is required to set

Yesterday I've learned how to make use of these EFUSES and set the VDD_SPI_AS_GPIO EFUSE bit to 1, so that it can be used as a normal GPIO. Unfortunately, this new GPIO is now not turning itself high at boot, so the internal flash memory can not be read (I'm using the ESP32-C3FN4 with 4Mb embedded flash). I have to exchange the ESP32 chip because it probably cannot be revived anymore 😄 

> @makermoekoe Do you have an external pulldown resistor on the LDO enable?
> I wouldn't expect VDD_SPI to be actively discharged and pulled low; assume hi-z.

On the LDO EN itself - yes, but there is a diode in between (see picture). Maybe I have to add another one on the VDD_SPI side? Should give it a try.

[image]

septatrix, 2022-02-18:
Have you tried using one of the GPIOs which default to a pull-up on reset instead of the VDD_SPI yet?

makermoekoe, 2022-02-21:
> Have you tried using one of the GPIOs which default to a pull-up on reset instead of the VDD_SPI yet?

Nope not yet, but what I found out is that when I remove R2 in the shown schematics above (replaced with 0 ohm) it actually works with VDD_SPI.

ESP-Marius, 2023-11-17:
I think we'll consider this question answered for now, as mentioned earlier we do provide hooks for doing early init in the bootloader, as well as the option to wrap functions using the linker.

Providing another hook would of course be possible, but the difficulty is that every user might have different demands (how early, which system features need to be available).

But please feel free to keep adding comments/votes to this issue if you think there is a specific point in the boot-up process that needs a hook feature. If we see that there is a big demand for a hook at a specific point then we will of course reconsider and potentially re-open the issue.

closed 2023-11-17 by ESP-Marius