espressif/esp-idf #8391: CONFIG_COMPILER_STACK_CHECK_MODE_ALL causes stack overflow detection in task activity task

bschick, 2022-02-12:
## Environment

- Module or chip used:  ESP32-C3-MINI-1-N4
- ESP-IDF v4.4-rc1
- Build System:        idf.py
- Operating System:    macOS
- Using an IDE?: No
- Power Supply:    USB

## Problem Description

I set the `CONFIG_COMPILER_STACK_CHECK_MODE_ALL=y` and to a clean build I hit the stack overflow dump. I have tried to increase the ISR stack size all the way to `CONFIG_FREERTOS_ISR_STACKSIZE=6144` and the dump still occurs. My assumption is the isr logic in /components/freertos/port/riscv/portasm.S is doing something to trigger the stack smash detection since a 6k stack size is huge for the one small ISR in my code that does little more than call `xQueueSendFromISR`. 

```
***ERROR*** A stack overflow in task activity task has been detected.
Stack dump detected
Core  0 register dump:
MEPC    : 0x403808dc  RA      : 0x4038949c  SP      : 0x3fc98310  GP      : 0x3fc93c00
0x403808dc: panic_abort at /Users/bschick/esp/esp-idf/components/esp_system/panic.c:402

0x4038949c: __ubsan_include at /Users/bschick/esp/esp-idf/components/esp_system/ubsan.c:293

TP      : 0x3fc9153c  T0      : 0x4005890e  T1      : 0x0000000f  T2      : 0x00000001
S0/FP   : 0x00000003  S1      : 0x80000003  A0      : 0x3fc98350  A1      : 0x3c0f1970
A2      : 0x0000000b  A3      : 0x3fc98383  A4      : 0x00000001  A5      : 0x3fc9d000
A6      : 0x00000000  A7      : 0x00000000  S2      : 0x00001881  S3      : 0x00000001
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000001  T6      : 0x00000000
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000

Backtrace:

panic_abort (details=details@entry=0x3fc98350 <xIsrStack+6000> "***ERROR*** A stack overflow in task activity task has been detected.") at /Users/bschick/esp/esp-idf/components/esp_system/panic.c:402
402	    *((volatile int *) 0) = 0; // NOLINT(clang-analyzer-core.NullDereference) should be an invalid operation on targets
#0  panic_abort (details=details@entry=0x3fc98350 <xIsrStack+6000> "***ERROR*** A stack overflow in task activity task has been detected.") at /Users/bschick/esp/esp-idf/components/esp_system/panic.c:402
#1  0x4038949c in esp_system_abort (details=details@entry=0x3fc98350 <xIsrStack+6000> "***ERROR*** A stack overflow in task activity task has been detected.") at /Users/bschick/esp/esp-idf/components/esp_system/esp_system.c:121
#2  0x4038d214 in vApplicationStackOverflowHook (xTask=<optimized out>, pcTaskName=0x3fcb5d48 "") at /Users/bschick/esp/esp-idf/components/freertos/port/riscv/port.c:380
#3  0x4038bc6c in vTaskSwitchContext () at /Users/bschick/esp/esp-idf/components/hal/esp32c3/include/hal/cpu_ll.h:43
#4  0x4038953c in isr_skip_decrement () at /Users/bschick/esp/esp-idf/components/freertos/port/riscv/portasm.S:95
Backtrace stopped: frame did not save the PC
ELF file SHA256: f76f37ef438ea659
```

### Expected Behavior

Not stack overflow triggered

Dazza0, 2022-02-14:
@bschick Probably because the stack overflow isn't happening on the ISR stack, but on the "activity task" stack. Try increasing the stack size of the "activity" task./

bschick, 2022-02-15:
Ooops, sorry about the noise. I had assumed activity task was part of IDF but it was in other shared code in my project.

closed 2022-02-15 by bschick