espressif/esp-idf #8394: No Bluetooth SPP connection possible with IDF 4.3.2

euphi, 2022-02-12:
## Environment

- Development Kit:      [ESP32-Devkit-V1]
- Module or chip used:  ESP32-WROOM-32|
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         platformio
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     Linux
- Using an IDE?: not relevant
- Power Supply:         USB

### platformio

Platformio Configuration (`platformio.ini`):

```
[platformio]
src_dir = main

[env:esp32dev]
platform = espressif32
framework = espidf
board = esp32doit-devkit-v1
```

Platformio Environment:

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (3.5.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40302.0 (4.3.2) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - tool-ninja 1.7.1 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 8.4.0+2021r2-patch2

```

## Problem Description

The "initiator" example can't connect to classic Bluetooth device.

There is already an issue for Arduino ESP32: https://github.com/espressif/arduino-esp32/issues/6061

To investigate for that issue and exclude the IDF as root cause, I installed the IDF and tried the Initiator example.

It seems that the IDF has the same issue, as I can't connect to the device with the SPP Initiator example.

To rule out name resolving issue, I changed the code that it connect to a pre-configured adress, and then it is still not possible to complete connection.

I also asked a question at https://esp32.com/viewtopic.php?f=19&t=26042 - but there is no answer yet.

The log shows that a connection is initiated but then no serial connection is possible;

`W (36287) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 1 closed: Peer connection failed (res: 16)`

Detailed Log:

```
32mI (3217) BT_BTM: BTM_InqDbRead: bd addr [201301180226]
 
[... 483 lines cut ...]
    switch(event){
    case ESP_BT_GAP_DISC_RES_EVT:
        ESP_LOGI(SPP_TAG, "ESP_BT_GAP_DISC_RES_EVT");
        esp_log_buffer_hex(SPP_TAG, param->disc_res.bda, ESP_BD_ADDR_LEN);
        /* Find the target peer device name in the EIR data */
//        for (int i = 0; i < param->disc_res.num_prop; i++){
//            if (param->disc_res.prop[i].type == ESP_BT_GAP_DEV_PROP_EIR
//                && get_name_from_eir(param->disc_res.prop[i].val, peer_bdname, &peer_bdname_len)){
                esp_log_buffer_char(SPP_TAG, peer_bdname, peer_bdname_len);
                ESP_LOGI(SPP_TAG,"Dev-Name %s", peer_bdname);
//              if (strlen(remote_device_name) == peer_bdname_len
//                    && strncmp(peer_bdname, remote_device_name, peer_bdname_len) == 0) {
//                    memcpy(peer_bd_addr, param->disc_res.bda, ESP_BD_ADDR_LEN);
                    /* Have found the target peer device, cancel the previous GAP discover procedure. And go on
                     * dsicovering the SPP service on the peer device */
                    esp_bt_gap_cancel_discovery();
                    esp_spp_start_discovery(peer_bd_addr);
//                }
//            }
//        }
        break;

[...]

```

euphi, 2022-02-12:
I tried to disable SSP in the IDF Config (unset CONFIG_BT_SSP_ENABLED) - **this seem to solve the issue.**

(_Sorry, I also noticed that I already posted the config with disabled SSP in my initial issue description. The log is from a configuration with SSP enabled. The config is the one that works now for me_)

So, the problem is with SSP and some (old) bluetooth chip (that I don't know - and I can't access it (mix of epoxy, shrink tube, hot glue..), but I guess it is similar or the same chip as used in HC-05 or HC-06 modules.

BetterJincheng, 2022-02-16:
@euphi I tried between two ESP32 modules and found nothing wrong about `initiator`. If you want to connect ESP32 to other device, you should make sure that the two have a common way of pairing.
Turn to us if any questions. Thanks!

euphi, 2022-02-16:
With SSP disabled in sdkconfig, pairing works fine.

However, with SSP enabled in config, pairing (or connecting to already paired device) does NOT work - even if I don't set SSP related configuration/parameters.  (When using the Initiator example) - or no call to `enableSSP()` when using Arduino-ESP32s `BluetoothSerial`.

So, I think SSP has a negative impact on pairing/connecting on some BT device, even when SSP is not used.

ferbar, 2022-02-27:
I want to establish a SPP connection from an ESP32 to a linux-pc with usb bt-dongle or raspberry pi zero w (both bluez). I'm having the same troubles as described by @euphi:

I can reproduce the error in bt_spp_initiator when _Secure Simple Pairing_ is enabled. Changing the spp_sec mode to NONE didn't change anything.

`static const esp_spp_sec_t sec_mask = ESP_SPP_SEC_NONE;
`

Logs:
```
I (2931) BT_RFCOMM: RFCOMM_CreateConnection(): scn:30, dlci:60, is_server:0 mtu:990, p_mcb:0x0, p_port:0x3ffc6bb4
I (2941) BT_L2CAP: L2CA_ErtmConnectReq()  PSM: 0x0003  BDA: 0002720376fc  p_ertm_info: 0x0 allowed:0x0 preferred:0
E (2951) BT_BTM: btm_sec_l2cap_access_req: (initiator) remote features unknown!!sec_flags:0x88

I (2961) BT_L2CAP: L2CAP - L2CA_conn_req(psm: 0x0003) returned CID: 0x0041
I (2971) BT_RFCOMM: PORT_SetEventCallback() handle:2
I (2971) BT_RFCOMM: PORT_SetEventMask() handle:2 mask:0x30005
I (2981) BT_RFCOMM: PORT_GetState() handle:2
I (2981) BT_RFCOMM: PORT_SetState() handle:2
I (2991) BT_RFCOMM: PORT_SetState() handle:2 FC_TYPE:0xc
I (2991) SPP_INITIATOR_DEMO: ESP_SPP_CL_INIT_EVT
I (7111) BT_L2CAP: L2CAP - Calling Disconnect_Ind_Cb(), CID: 0x0041  No Conf Needed
```
Looks ais if it's from here, however the code didn't change recently.
 _remote features unknown!!_ :
https://github.com/espressif/esp-idf/blob/master/components/bt/host/bluedroid/stack/btm/btm_sec.c#L2280

The program running on linux is the spp_echo_server: https://gist.github.com/ukBaz/217875c83c2535d22a16ba38fc8f2a91

bt_spp_initiator was additionaly modified to connect directly to a mac and not send too much data.

removing the call to esp_bt_gap_set_security_param didn't change anything.