espressif/esp-idf #8397: Bounds checking is needed in esp_crt_bundle_set/esp_crt_bundle_init

jhnlmn, 2022-02-13:
https://github.com/espressif/esp-idf/blob/20847eeb96c486629a678d6ffbe73670013fb771/components/mbedtls/esp_crt_bundle/esp_crt_bundle.c#L174

When I receive certificate bundles they sometimes arrive corrupted and then esp_crt_bundle_init crashed.
It should be easy for you to add some minimal bounds check validation to esp_crt_bundle_init to prevent crashes on invalid data.
Otherwise I have to re-implement all logic of esp_crt_bundle_init in my own code to ensure that my FW does not crash.

jhnlmn, 2022-02-28:
Hi, sorry, but you cannot just make a fix for one specific corrupted certificate. A good embedded code should have complete bounds checks again all possible corrupted input (unless prohibitively expensive). I am receieving certificate bundle over network and any byte can be corrupted and FW should not crash because of that. This is the code that I had to write to validate bundle before calling esp_crt_bundle_set. As you can see, I had to duplicate a lot of code from esp_crt_bundle_init, but with bound checks.

void my_crt_bundle_set_safe(const uint8_t *x509_bundle, int bundle_size) {

        int BUNDLE_HEADER_OFFSET {2};
        int CRT_HEADER_OFFSET {4};
        bool valid {true};
        if(bundle_size < BUNDLE_HEADER_OFFSET + CRT_HEADER_OFFSET) {
            valid = false;
        }
        else {
            unsigned num_certs = (x509_bundle[0] << 8) | x509_bundle[1];
            int offset = BUNDLE_HEADER_OFFSET;
            for (int i = 0; i < num_certs; i++) {
                if(offset + CRT_HEADER_OFFSET > bundle_size) {
                    valid = false;
                    break;
                }
                const uint8_t *cur_crt = x509_bundle + offset;
                size_t name_len = cur_crt[0] << 8 | cur_crt[1];
                size_t key_len = cur_crt[2] << 8 | cur_crt[3];
                if(offset + name_len + key_len + CRT_HEADER_OFFSET > bundle_size) {
                    valid = false;
                    break;
                }
                offset += CRT_HEADER_OFFSET + name_len + key_len;
            }
        }
        if(!valid) {
            ESP_LOGE(TAG, "bad cert");
            return false;
        }
        esp_crt_bundle_set(x509_bundle);

I guess you have to add a new API function like
void esp_crt_bundle_set_safe(const uint8_t *x509_bundle, int bundle_size)

Thank you

AdityaHPatwardhan, 2022-03-28:
Hi @jhnlmn, Sorry for the delayed reply. 

I had a few concerns with this.
When you say you obtain the crt_bundle through network. Do you know the exact size of the certificate bundle ?
If the cert bundle is corrupted then I am not sure if bound checking itself will be sufficient to validate the certificate.

jhnlmn, 2022-03-29:
@ AdityaHPatwardhan

"bound checking itself will be sufficient to validate the certificate."
I did not ask to validate certificate. I only asked to prevent my firmware from crashing, like it does now inside esp_crt_bundle_init
Crashing is absolutely unacceptable.
If you do not believe me, you can read online about defensive programming and bounds checking. In few words, no matter how corrupt is input, software should not crash, this is especially true for embedded firmware.

"Also can you explain what do you mean by receiving the crt_bundle over the network ? Do you perform OTA or something similar ?"
In order to perform OTA you first need to somehow send URL and certificate of OTA server to devices. Your examples have certificate bundle embedded inside FW image, but this is not very flexible since URL and certificate may change in the future. Therefore, we have a way to provision devices with URL and certificates at run time.

AdityaHPatwardhan, 2022-03-29:
Can you see if this fix resolves your issue of crashing.
[esp_crt_bundle_bound_check.patch.zip](https://github.com/espressif/esp-idf/files/8368413/esp_crt_bundle_bound_check.patch.zip)

AdityaHPatwardhan, 2022-03-29:
@jhnlmn Thank you for your detailed explanation. I now understand the issue completely. 
I agree the API should not crash. The API in its current state is indeed vulnerable to the issues that you mentioned. I understand the bound checks that you have suggested would be necessary for the API.
Considering we are near a new major release ( IDF-5.0)  I think it would be appropriate to break this API and update it with the necessary bound checking.

jhnlmn, 2022-03-30:
OK, thank you. I am OK for now with my own implementation of my_crt_bundle_set_safe.
In the future you can just add a new API with bounds checking,
while keeping the existing one as-is for compatibility.
In general, it is a good idea to add Size to all APIs which receive a Pointer - then you will not need to change APIs.

closed 2022-04-01 by espressif-bot