espressif/esp-idf #8400: 【ESP32-C3/Scan duplicate enable 】Scan duplicate enable feature invalid

xiaolongba, 2022-02-13:
## Environment

- Development Kit:      [ESP32-C3-DevkitM-1 v1.0]
- Kit version (for WroverKit/PicoKit/DevKitC): [master branch]
- Module or chip used:  [ESP32-C3]
- IDF version (run ``git describe --tags`` to find it):
    // v5.0-dev-1519-g5810ed1d04
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // riscv32-esp-elf-gcc.exe (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [Plain Command Prompt|PowerShell].
- Using an IDE?: [Vscode]
- Power Supply:         [USB|]

## Problem Description
I am developing the BLE application based on esp32-c3, but I found that the duplicate advertising reports are still exist even if using the **`BLE_SCAN_DUPLICATE_ENABLE`** feature and the related configuration as below:
[image]

### Expected Behavior

the duplicate advertising reports shall be ignored

### Actual Behavior

the duplicate advertising reports are still exist even if using the **`BLE_SCAN_DUPLICATE_ENABLE`** feature
### Steps to reproduce

1. configure the scan parameters as below:
    ```c
    static esp_ble_scan_params_t gs_m_ble_scan_params = {
      .scan_type          = BLE_SCAN_TYPE_ACTIVE,
      .own_addr_type      = BLE_ADDR_TYPE_PUBLIC,
      .scan_duplicate     = BLE_SCAN_DUPLICATE_ENABLE,
      .scan_filter_policy = BLE_SCAN_FILTER_ALLOW_ALL,
      .scan_interval      = SCAN_INTERVAL,
      .scan_window        = SCAN_WINDOWS,
    };
    ```
2. start scanning

### Code to reproduce this issue
NULL

## Debug Logs

```
I (458) Scanning-for-undirected-adv_demo: mac address:
I (458) Scanning-for-undirected-adv_demo: e0 b6 55 be 1a b5 
I (458) Scanning-for-undirected-adv_demo: searched Adv Data Len 19, Scan Response Len 0, RSSI:-60dBm
I (468) Scanning-for-undirected-adv_demo: adv data:
I (468) Scanning-for-undirected-adv_demo: 02 01 06 0f ff 8f 03 0a 10 ad 05 01 3b 1e fc 1d 
I (478) Scanning-for-undirected-adv_demo: 41 b0 ec 
I (488) Scanning-for-undirected-adv_demo: mac address:
I (488) Scanning-for-undirected-adv_demo: 46 88 f7 5d 07 2c 
I (498) Scanning-for-undirected-adv_demo: searched Adv Data Len 19, Scan Response Len 0, RSSI:-75dBm
I (508) Scanning-for-undirected-adv_demo: adv data:
I (508) Scanning-for-undirected-adv_demo: 02 01 1a 02 0a 07 0c ff 4c 00 10 07 24 1b cd df 
I (518) Scanning-for-undirected-adv_demo: bd fc 28 
I (528) Scanning-for-undirected-adv_demo: mac address:
I (528) Scanning-for-undirected-adv_demo: 3a 43 b9 a8 f6 cb 
I (538) Scanning-for-undirected-adv_demo: searched Adv Data Len 31, Scan Response Len 0, RSSI:-52dBm
I (548) Scanning-for-undirected-adv_demo: adv data:
I (558) Scanning-for-undirected-adv_demo: 1e ff 06 00 01 09 20 02 f9 07 f1 61 a3 6b ae b0 
I (568) Scanning-for-undirected-adv_demo: 86 8c cd cd db 0f a6 2a ea 84 76 54 87 c8 a5 
I (568) Scanning-for-undirected-adv_demo: mac address:
I (578) Scanning-for-undirected-adv_demo: 7c df a1 66 aa 49 
I (588) Scanning-for-undirected-adv_demo: searched Adv Data Len 18, Scan Response Len 11, RSSI:-24dBm
I (598) Scanning-for-undirected-adv_demo: adv data:
I (598) Scanning-for-undirected-adv_demo: 02 01 06 0b 09 e6 88 91 e6 98 af e8 b0 81 3f 02 
[... 369 lines cut ...]
I (77228) Scanning-for-undirected-adv_demo: searched Adv Data Len 22, Scan Response Len 0, RSSI:-75dBm
I (77238) Scanning-for-undirected-adv_demo: adv data:
I (77238) Scanning-for-undirected-adv_demo: 15 2a 06 38 90 1a cd 43 e2 c5 85 28 96 e2 0e a5 
I (77248) Scanning-for-undirected-adv_demo: b0 0b b7 f0 01 f5 
I (77298) Scanning-for-undirected-adv_demo: mac address:
I (77298) Scanning-for-undirected-adv_demo: 7a 9b fa 7b 47 55 
I (77298) Scanning-for-undirected-adv_demo: searched Adv Data Len 17, Scan Response Len 0, RSSI:-91dBm
I (77308) Scanning-for-undirected-adv_demo: adv data:
I (77318) Scanning-for-undirected-adv_demo: 02 01 1a 02 0a 0c 0a ff 4c 00 10 05 51 1c aa 21 
I (77318) Scanning-for-undirected-adv_demo: 82 
I (77328) Scanning-for-undirected-adv_demo: mac address:
I (77338) Scanning-for-undirected-adv_demo: 78 da 07 c3 71 6f 
I (77338) Scanning-for-undirected-adv_demo: searched Adv Data Len 25, Scan Response Len 0, RSSI:-75dBm
I (77348) Scanning-for-undirected-adv_demo: adv data:
I (77358) Scanning-for-undirected-adv_demo: 18 2a 06 cc d6 a6 5e 01 37 94 36 5e b3 5f 31 8d 
I (77368) Scanning-for-undirected-adv_demo: 78 96 ad 41 d5 3d 7e 3e a0 
```
as you can see the log above, several devices have been reported several times. whether it is linked to **`the Maximum number of devices in scan duplicate filter`** also? if the number of devices reach to this threshold value, the scan duplicate filter will be refreshed and then the duplicate advertising reports will be appear again after a period of time?

## Other items if possible
NULL

so my question is as below:

1. is it possible to ignore the duplicate advertising reports compeletely

xiaolongba, 2022-02-16:
Hello Espressif experts,

3 days have passed, any updates?

xiaolongba, 2022-02-25:
Hello guys,

could you pls give me your hand on this case? i think it doesn't take much your time.

thanks.

xiaolongba, 2022-03-13:
Hello guys,

almost one month has passed. could you pls give me your hand on this case? or could you tell me where I can get the support if I cannot get the support from here? I think it doesn't take much of your time.

thanks.

jack0c, 2022-03-16:
@xiaolongba Sorry for the delay. @xiewenxiang Please PTAL.

xiewenxiang, 2022-03-16:
@xiaolongba 

In fact, it is already in effect, two cases will be refreshed

1. The cached scan rst list is full.
2. The timeout has expired.

If either condition is met, the scan event is refreshed and reported again.

xiaolongba, 2022-03-16:
@jack0c @xiewenxiang Bro.

Big thanks. based on your description, I still have some doubts:

1. why the duplicated reports are still filled in the scan rst list? in my opinion, the duplicated reports shall be discarded, the scan rst list only fills in the different types of reports, else the cached scan rst list will be full soon even if the number of cached scan rst list is configured to a large number.
2. timeout you mentioned is referred as scan timeout?

thanks again and sorry for inconvenience.

AxelLin, 2022-03-16:
Check if increasing BT_CTRL_SCAN_DUPL_CACHE_SIZE helps.

xiewenxiang, 2022-03-16:
@xiaolongba 

question 1:

There are three ways of filtering：`Address`;  `Adv Data`; `Address + Adv data`, It depends on the configuration items in Menuconfig `component config -> bluetooth -> bluetooth controller -> BLE Scan Duplicate Options -> Scan Duplicate Type`

question 2:

It is a timer for internal maintenance.

xiaolongba, 2022-03-16:
> Check if increasing BT_CTRL_SCAN_DUPL_CACHE_SIZE helps.

@AxelLin yes, it is in effect when increasing the BT_CTRL_SCAN_DUPL_CACHE_SIZE , but the scan rst list still is easy to full. there are only around 10 BLE devices on my side. also, I check the report data, some of them are the same. so scan rst list has cached the same report data already. In my understanding, the scan rst list should compare to the report data it received, it should be discarded when the report data is already inside. but in fact, the scan rst list fills in any report data it received.  else I think the scan rst list is not meaningful if the advertiser reports its data every 20ms as it will be full soon.

> question 1:
> 
> There are three ways of filtering：`Address`; `Adv Data`; `Address + Adv data`, It depends on the configuration items in Menuconfig `component config -> bluetooth -> bluetooth controller -> BLE Scan Duplicate Options -> Scan Duplicate Type`

@xiewenxiang yes, as I mentioned at the beginning, I configure the filtering rules as below:
[image]

> question 2:
> 
> It is a timer for internal maintenance.

can I get the timer value? or modify it? or how can I know the timer has expired?

again, I think the scan rst list should compare to any report data and decide if it shall be filled in. but the truth is that any report data scan rst list received are all filled in, it should be unreasonable.  the purpose of scan rst list is to filter out the duplicated reporting data, but now the same reporting data are present on the same scan rst list.

xiaolongba, 2022-03-16:
appended more details, the reason why I think the duplicated reports have been presented on the scan rst list as the I got several same reports on the REPORT_EVENT. so the scan rst list is refreshed and it may be caused by the scan rst list is full already. but the reason why the scan rst list is full, in my understanding, it shall fill in any reports once it is received as there are only 10 devices on my side.

by the way, is it possible to print out the scan rst list and check if there are the same reports inside? or could you tell me the reports how to fill in the scan rst list? if the scan rst list compare the reports with the reports inside the scan rst list and then decide if it can be filled in?

zhp0406, 2023-09-12:
@xiaolongba 
Hi,Has the issue been resolved?

xiaolongba, 2023-09-14:
after that,i haven't tested yet since the official guys have not gave me the further information. but you can have a try with the latest sdk. if it is possible, pls tell me the results. 

thanks in advance

zhp0406, 2023-09-14:
@xiaolongba 
My test didn't encounter any issues. If you're still experiencing problems on your end, please report them to me. It's important to note that under the BT_CTRL_SCAN_DUPL_TYPE_DATA_DEVICE type, it is considered the same PDU only when the address, address type, adv data, and PDU type are all the same; otherwise, it will be reported multiple times.

[image]

zhp0406, 2023-09-14:
@xiaolongba 
[image]

Indastri, 2023-12-14:
Since there is no response from the user closing this issue. If there will be another problem feel free to reopen this issue or create a new one. Cheers.

closed 2023-12-14 by Indastri